esphome/issues #4452: ESP_ERR_NOT_SUPPORTED with esp_camera

Martinvdm, 2023-04-23:
### The problem

Getting an error setting up ESP32Camera with esp-wrover-kit
[18:08:18][E][esp32_camera:123]:   Setup Failed: ESP_ERR_NOT_SUPPORTED

It has worked, so it seems to be a update change somehow i think.

config:
```
esphome:
  name: the_name
  platform: ESP32
  board: esp-wrover-kit # esp32dev

logger:
api:
ota:
captive_portal:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  use_address: 10.10.11.23

switch:
  - platform: restart
    name: ${display_name} Restart

# Sensors
sensor:
   - platform: wifi_signal
     name: ${display_name} WiFi Signal
     
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  name: ${display_name} security
# Image settings
  max_framerate: 5 fps
  resolution: SXGA
  jpeg_quality: 20
  contrast: 2
  brightness: 2
  saturation: 2
```

### Which version of ESPHome has the issue?

2023.04.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.04.6

### What platform are you using?

ESP32

### Board

esp32-wrover-cam

### Anything in the logs that might be useful for us?

```txt
INFO Successfully connected to 10.10.11.23
[18:12:18][I][app:102]: ESPHome version 2023.4.0 compiled on Apr 23 2023, 18:01:03
[18:12:18][C][wifi:505]: WiFi:
[18:12:18][C][wifi:363]:   Local MAC: 10:52:XX:XX:XX:XX
[18:12:18][C][wifi:364]:   SSID: [redacted]
[18:12:18][C][wifi:365]:   IP Address: 10.10.11.23
[18:12:18][C][wifi:367]:   BSSID: [redacted]
[18:12:18][C][wifi:368]:   Hostname: 'the_hostname'
[18:12:18][C][wifi:370]:   Signal strength: -78 dB ▂▄▆█
[18:12:18][C][wifi:374]:   Channel: 11
[18:12:18][C][wifi:375]:   Subnet: 255.255.255.0
[18:12:19][C][wifi:376]:   Gateway: 10.10.11.1
[18:12:19][C][wifi:377]:   DNS1: 10.10.11.1
[18:12:19][C][wifi:378]:   DNS2: 0.0.0.0
[18:12:19][C][logger:294]: Logger:
[18:12:19][C][logger:295]:   Level: DEBUG
[18:12:19][C][logger:296]:   Log Baud Rate: 115200
[18:12:19][C][logger:297]:   Hardware UART: UART0
[18:12:19][C][gpio.output:010]: GPIO Binary Output:
[18:12:19][C][gpio.output:011]:   Pin: GPIO4
[18:12:19][C][light:103]: Light 'The Name light'
[18:12:19][C][restart:075]: Restart Switch 'The Name Restart'
[18:12:19][C][restart:077]:   Icon: 'mdi:restart'
[18:12:19][C][restart:098]:   Restore Mode: always OFF
[18:12:19][C][esp32_camera:048]: ESP32 Camera:
[18:12:19][C][esp32_camera:049]:   Name: The Name security
[18:12:19][C][esp32_camera:050]:   Internal: NO
[18:12:19][C][esp32_camera:052]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[18:12:19][C][esp32_camera:053]:   VSYNC Pin: 25
[18:12:19][C][esp32_camera:054]:   HREF Pin: 23
[18:12:19][C][esp32_camera:055]:   Pixel Clock Pin: 22
[18:12:19][C][esp32_camera:056]:   External Clock: Pin:0 Frequency:20000000
[18:12:19][C][esp32_camera:060]:   I2C Pins: SDA:26 SCL:27
[18:12:19][C][esp32_camera:062]:   Reset Pin: -1
[18:12:19][C][esp32_camera:089]:   Resolution: 1280x1024 (SXGA)
[18:12:19][E][esp32_camera:123]:   Setup Failed: ESP_ERR_NOT_SUPPORTED
[18:12:19][C][psram:020]: PSRAM:
[18:12:19][C][psram:021]:   Available: YES
[18:12:19][C][psram:024]:   Size: 3 MB
[18:12:19][C][captive_portal:088]: Captive Portal:
[18:12:19][C][mdns:108]: mDNS:
[18:12:19][C][mdns:109]:   Hostname: the_name 
[18:12:19][C][ota:093]: Over-The-Air Updates:
[18:12:19][C][ota:094]:   Address: 10.10.11.23:3232
[18:12:19][C][api:138]: API Server:
[18:12:19][C][api:139]:   Address: 10.10.11.23:6053
[18:12:19][C][api:143]:   Using noise encryption: NO
[18:12:19][C][wifi_signal.sensor:009]: WiFi Signal 'The Name WiFi Signal'
[18:12:19][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[18:12:19][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[18:12:19][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[18:12:19][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
```

NODeeJay, 2023-05-01:
I get the same error with Freenove Wrover CAM. However, I noticed your configuration differs from mine and the manual, you do not have GPIO4 as data_pin: https://esphome.io/components/esp32_camera.html#configuration-for-wrover-kit-boards which seems to be the settings for the AI Thinker CAM. So I went ahead and tried your config on the Wrover board with the error Setup Failed: ESP_ERR_NOT_FOUND
Do you configure the Wrover board like a AI Thinker because Wrover config seems not to work? https://community.home-assistant.io/t/esphome-camera-setup-failed-esp-err-not-found/524903/14

I build a Tasmota and see if it works there....

Martinvdm, 2023-05-02:
Thanks for your answer. I do get ESP_ERR_NOT_SUPPORTED. 
I checked your articles, but I do use the 
board: esp-wrover-kit
And not other board.

I tried changing the data pins but no change. Still the same error.

NODeeJay, 2023-05-02:
Ahhh, this is the AI Thinker module https://esphome.io/components/esp32_camera.html#configuration-for-ai-thinker-camera and not the Wrover Kit https://esphome.io/components/esp32_camera.html#configuration-for-wrover-kit-boards. That's why your config is also different.
I have one of those modules which does not do UART logging anymore since I once activated the internal hall sensor while the cam was active. I have another one that seems to be fine still. I will try latest esphome with it later and let you know if I run into the same issue.

Do you have another of those boards and other camera modules to narrow the issue down?
And you could switch the logger
```
logger:
  level: VERBOSE
```
and see if it brings some more infos that can lead to the cause.

Martinvdm, 2023-05-02:
Thanks. I edited my comment btw. I do have another cam, I will try it later. I don’t have a spare board.
If this is AiThinker board shouldn’t I have to use 
board = esp32cam
instead of the wrover-kit?

NODeeJay, 2023-05-02:
I saw several configs with wrover as board so I did not mention anything about it, seems there are quite some configs that work that way and TBH I have no clue what really the difference in the board selection does, except allowing/disallowing pins.

I used my Ai Thinker (the same module as you have, also connected to a BME280 :-)), working fine with ESPHome from last year and flashed the update. I do not get an error with the camera and I "see" it working in the log
`[D][esp32_camera:196]: Got Image: len=99847`
but my cam webserver does not start anymore since
`[E][esp32_camera_web_server:097]:   Setup Failed`

So I tried the second module and get the same constellation, camera ok, web server failed.
As you mentioned it, the internal flashlight I also have on GPIO4, works on both. Sorry that I can't help further here. I will try to get the webserver working and see if there is really a picture

closed 2023-09-21 by github-actions[bot]