esphome/issues #4454: LuatOS ESP32C3 gpio12 and gpio13 cannot be used

Roofcatbusy, 2023-04-24:
### The problem

I've found out that currently ESPHome does not allow to use gpio12 and gpio13 on esp32c3 because of code in components/esp32/gpio_esp32_c3.py which does not allow to use gpio12-13 since normally it attached to flash/ However for this particular board gpio12 and 13 are not used by SPI  (see below in additional info) and I want to use these leds as indicators.

### Which version of ESPHome has the issue?

2023.4.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

Luatos ESP32C3 (http://luatos.com/t/esp32c3)

### Component causing the issue

esp32

### Additional information

according to [board documentation](https://wiki.luatos.com/chips/esp32c3/board.html) (google translated) "In order to increase the number of available GPIOs, **the development board chooses the DIO mode of 2-wire SPI. IO12 and IO13 are not connected to flash**. Use your own compilation When installing the software, you need to pay attention to configure the flash as DIO mode." I've successfully compiled and flashed esphome to the board in dio mode, but I cannot use gpio12 and 13 in config.

Aikhjarto, 2023-04-24:
There is already seems to be a feature request for this: https://github.com/esphome/feature-requests/issues/2018

Roofcatbusy, 2023-04-24:
I should have check feature requests beforehand. Indeed, https://github.com/esphome/feature-requests/issues/2018 covers same issue

closed 2023-04-24 by Roofcatbusy

watermarkhu, 2023-07-17:
@Roofcatbusy Can you share your config for the LuatOS ESP32C3? I'm a just getting a boot loop.

ThomasWagnerJKU, 2023-07-17:
@watermarkhu 
To avoid the boot loop, I specified the boards flash configuration manually, e.g.
```yaml
esphome:
  platformio_options:
    # flash_mode defaults to qio, but Air101/LuatOS requires dio mode
    board_build.f_flash: 40000000L
    board_build.flash_mode: dio
    board_build.flash_size: 4MB

esp32:
  board: esp32-c3-devkitm-1
```

starskyzheng, 2023-07-19:
You can modified the esphome source file:
https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/gpio_esp32_c3.py
remove line 11 and 12:
``` python
_ESP32C3_SPI_PSRAM_PINS = {
#    12: "SPIHD",
#    13: "SPIWP",
    14: "SPICS0",
    15: "SPICLK",
    16: "SPID",
    17: "SPIQ",
}
```

zwgmy, 2023-08-10:
> You can modified the esphome source file: https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/gpio_esp32_c3.py remove line 11 and 12:
> 
> ```python
> _ESP32C3_SPI_PSRAM_PINS = {
> #    12: "SPIHD",
> #    13: "SPIWP",
>     14: "SPICS0",
>     15: "SPICLK",
>     16: "SPID",
>     17: "SPIQ",
> }
> ```

how to use to compile it? I remove it on local ,when I compile the yaml,it use the net file