esphome/issues #4455: fastled_clockless won't compile after latest update

boweeble, 2023-04-25:
### The problem

Updated to the newest version of ESPHome and my LED strip required and update - YAML still validates and no changes have been made but won't compile.  Tried create a new LED strip device and same results.

### Which version of ESPHome has the issue?

2023.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini Pro

### Component causing the issue

fastled_clockless

### Example YAML snippet

```yaml
esphome:
  name: rgb-led-strip
  friendly_name: RGB-Led-Strip

  on_boot:
    priority: 600
    then:
     - light.turn_on:
        id: light_rgb
        brightness: 30%
        red: 100%
        green: 100%
        blue: 0%
    
esp8266:
  board: esp01_1m
  framework:
    version: 2.7.4
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "dKl76voORlBMAyAmPnfKEg3uEFuklUsDED5o0yuSBBk="

ota:
  password: "1d99806718454339b28acb1b26d37bb7"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rgb-Led-Strip Fallback Hotspot"
    password: "6xsbQtSbiAUj"

light:
[... 313 lines cut ...]
src/esphome/core/string_ref.h: In function 'bool esphome::operator==(const esphome::StringRef&, const string&)':
src/esphome/core/string_ref.h:87:14: error: 'const class esphome::StringRef' has no member named 'size'
   return lhs.size() == rhs.size() && std::equal(std::begin(lhs), std::end(lhs), std::begin(rhs));
              ^
src/esphome/core/string_ref.h: In function 'bool esphome::operator==(const esphome::StringRef&, const char*)':
src/esphome/core/string_ref.h:93:14: error: 'const class esphome::StringRef' has no member named 'size'
   return lhs.size() == strlen(rhs) && std::equal(std::begin(lhs), std::end(lhs), rhs);
              ^
src/esphome/core/string_ref.h: In function 'std::string& esphome::operator+=(std::string&, const esphome::StringRef&)':
src/esphome/core/string_ref.h:113:31: error: 'const class esphome::StringRef' has no member named 'size'
   lhs.append(rhs.c_str(), rhs.size());
                               ^
src/esphome/core/string_ref.h: In function 'std::string esphome::operator+(const char*, const esphome::StringRef&)':
src/esphome/core/string_ref.h:119:31: error: 'const class esphome::StringRef' has no member named 'size'
   str.append(rhs.c_str(), rhs.size());
                               ^
src/esphome/core/string_ref.h: In member function 'constexpr bool esphome::StringRef::empty() const':
src/esphome/core/string_ref.h:69:52: warning: control reaches end of non-void function [-Wreturn-type]
   constexpr bool empty() const { return len_ == 0; }
                                                    ^
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [/data/rgb-led-strip/.pioenvs/rgb-led-strip/src/esphome/core/entity_base.cpp.o] Error 1
========================= [FAILED] Took 258.49 seconds =========================
```

ssieb, 2023-04-25:
You're using an old, incompatible Arduino version.  See the big warning at the top:
`WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.`

closed 2023-04-25 by ssieb

boweeble, 2023-04-25:
If I take the version out then it won't compile either and this is what I get which is why it was in there to begin with:

[image]

ssieb, 2023-04-25:
Yes, that's true, but that's a different issue.  For now you need to use neopixel instead or try one of the PRs for fixing fastled.

boweeble, 2023-05-01:
Thank you for the comment, after digging through the documentation I see the recommendation to use the neopixelbus platform.

ssieb, 2023-05-03:
This is temporarily fixed in the latest point release that is just coming out.

ssieb, 2023-05-03:
https://github.com/esphome/esphome/pull/4777