esphome/issues #4456: Unable to compile with time component

bipsendk, 2023-04-25:
### The problem

Trying to add homeasisstant time component in order to control/schedule deep sleep - but get errors during compilation (syntax validation says OK).

### Which version of ESPHome has the issue?

2023.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
time:
  - platform: homeassistant
    id: esptime
```

### Anything in the logs that might be useful for us?

```txt
Messages after compile:

INFO Reading configuration /config/esphome/bathroom-bluetooth-proxy.yaml...
INFO Detected timezone 'Europe/Copenhagen'
INFO Generating C++ source...
INFO Compiling app...
Processing bathroom-bluetooth-proxy (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- Improv @ 1.2.3
Linking /data/bathroom-bluetooth-proxy/.pioenvs/bathroom-bluetooth-proxy/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bathroom-bluetooth-proxy/.pioenvs/bathroom-bluetooth-proxy/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection20on_get_time_responseERKNS0_15GetTimeResponseE+0x0): undefined reference to `esphome::homeassistant::global_homeassistant_time'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bathroom-bluetooth-proxy/.pioenvs/bathroom-bluetooth-proxy/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection20on_get_time_responseERKNS0_15GetTimeResponseE+0x4): undefined reference to `esphome::time::RealTimeClock::synchronize_epoch_(unsigned int)'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bathroom-bluetooth-proxy/.pioenvs/bathroom-bluetooth-proxy/src/esphome/components/api/api_connection.o: in function `esphome::homeassistant::HomeassistantTime::set_epoch_time(unsigned int)':
/config/esphome/.esphome/build/bathroom-bluetooth-proxy/src/esphome/components/homeassistant/time/homeassistant_time.h:15: undefined reference to `esphome::time::RealTimeClock::synchronize_epoch_(unsigned int)'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bathroom-bluetooth-proxy/.pioenvs/bathroom-bluetooth-proxy/src/main.o:(.literal._Z5setupv+0xbc): undefined reference to `vtable for esphome::homeassistant::HomeassistantTime'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bathroom-bluetooth-proxy/.pioenvs/bathroom-bluetooth-proxy/src/main.o:(.literal._Z5setupv+0xfc): undefined reference to `esphome::time::RealTimeClock::RealTimeClock()'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bathroom-bluetooth-proxy/.pioenvs/bathroom-bluetooth-proxy/src/main.o: in function `esphome::Component::Component()':
/config/esphome/.esphome/build/bathroom-bluetooth-proxy/src/esphome/core/component.h:67: undefined reference to `esphome::time::RealTimeClock::RealTimeClock()'
collect2: error: ld returned 1 exit status
*** [/data/bathroom-bluetooth-proxy/.pioenvs/bathroom-bluetooth-proxy/firmware.elf] Error 1
========================= [FAILED] Took 31.54 seconds =========================
```

ssieb, 2023-04-25:
Try doing a clean build.

bipsendk, 2023-04-25:
That did the trick - thanks

closed 2023-04-25 by bipsendk