esphome/issues #4458: Unable to update bluetooth proxy

Pinez83, 2023-04-25:
### The problem

hello everyone, i use 2 esp32 as bluetooth proxy, created using the site
https://esphome.github.io/bluetooth-proxies/

Everything was fine until last week when I failed to update due to the following error:

Reading CMake configuration...
-- Component directory /data/cache/platformio/packages/framework-espidf/components/openthread does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/esp32-bluetooth-proxy-84f890/.pioenvs/esp32-bluetooth-proxy-84f890/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:227 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:507 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:3 (project)

Unfortunately I'm not an expert and I have no idea what to do!

Thanks to those who will help me

### Which version of ESPHome has the issue?

2023.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2023.4.6

### What platform are you using?

ESP32

Pinez83, 2023-04-26:
> You might need to re-install the addon to fix that.

It worked perfectly.  Thank you!

closed 2023-04-26 by ssieb

jeroenkl, 2023-07-06:
> You might need to re-install the addon to fix that.

Re-install ESPhome add-on? 
Is this not affecting any other ESPhome devices which are intergrated?