esphome/issues #4460: Undocumented breaking changes in 2023.4.0 or 2023.4.1?

nagyv, 2023-04-26:
### The problem

I'm using ESPHome in HomeAssistant. I recently upgraded from 2023.3.2 to 2023.4.1. My external display stopped working without showing any issues in the logs or at compile. I can even access the display wirelessly, reinstall it or query the logs, just the coded functionality does not work.

I managed to restore a backup of 2023.3.2 and after recompiling, the external display works again.

### Which version of ESPHome has the issue?

2023.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP8266

### Board

WEMOS-D1-MINI

### Component causing the issue

?

### Example YAML snippet

```yaml
substitutions:
  name: internal-display
  friendly_name: "Benti kijelző"
  rx_pin: D1
  tx_pin: D6
  
esphome:
  name: "${name}"

esp8266:
  board: d1_mini

uart:
  rx_pin: $rx_pin
  tx_pin: $tx_pin
  baud_rate: 115200

<<: !include packages/core.yaml

packages:
  #time: !include packages/time.yaml
  api: !include packages/api.yaml
  display: !include packages/display.yaml
  #meteo: !include packages/meteo.yaml

button:
  - platform: restart
    name: "Restart display: ${friendly_name}"

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time:
    # Update time display every minute
    - seconds: 0
      minutes: /1
      then:
      - lambda:
          auto time = id(homeassistant_time).now();
          id(nextion_clock).set_state(time.strftime("%H:%M"), false, true);
    # Sleep the display for the night
    - seconds: 0
      minutes: 00
      hours: 23
      then:
      - lambda:
          #id(internal_display).set_touch_sleep_timeout(30);
          #id(internal_display).set_auto_wake_on_touch(true);
          #id(internal_display).sleep(true);
          id(internal_display)->set_backlight_brightness(0);
    # A felébresztés nem működik. Ezért Node-RED-ből újraindítom a kijelzőt reggel 6-kor.
   # - seconds: 0
    #  minutes: 0
    #  minutes: 0
    #  hours: 6
    #  then:
    #  - lambda:
          #id(internal_display).sleep(true);
    #      id(internal_display).update_all_components();

text_sensor:
  - platform: nextion
    nextion_id: internal_display
    id: nextion_clock
    component_name: Main.clock
  - platform: nextion
    nextion_id: internal_display
    id: nextion_temperature
    component_name: Main.temperature
  - platform: homeassistant
    id: weather_forecast
    entity_id: weather.otthon
    on_value:
      then:
        - lambda:
            # picture id - state mapping
            # 2 - fog
            # 3 - sunny
            # 4 - snowy-rainy
            # 5 - snowy
            # 6 - rainy
            # 7 - pouring
            # 8 - partlycloudy
            # 9 - lightning-rainy
            # 10 - cloudy
            # 11 - clear-night
            # 12 - unknown
            
            int image_id;
            
            if(x == "fog") {
              image_id = 2;
            } else if (x == "sunny") {
              image_id = 3;
            } else if (x == "snowy-rainy") {
              image_id = 4;
            } else if (x == "snowy") {
              image_id = 5;
            } else if (x == "rainy") {
              image_id = 6;
            } else if (x == "pouring") {
              image_id = 7;
            } else if (x == "partlycloudy") {
              image_id = 8;
            } else if (x == "lightning-rainy") {
              image_id = 9;
            } else if (x == "cloudy") {
              image_id = 10;
            } else if (x == "clear-night") {
              image_id = 11;
            } else {
              ESP_LOGD("main", "Could not find weather state %s", x);
              image_id = 12;
            }
            
            ESP_LOGD("main", "The weather image is %d", image_id);
            id(internal_display).set_component_pic("forecast", image_id);

binary_sensor:
  - platform: nextion  # Handle action button clicks from the display
    internal: true
    id: action_btn
    page_id: 0
    component_id: 5
    on_press:
      then:
        - homeassistant.service:
            service: input_boolean.turn_off
            data:
              entity_id: input_boolean.teregetesre_var
  - platform: homeassistant  ## Show/Hide the action button on the display
    id: toggle_action_btn_visibility
    entity_id: input_boolean.teregetesre_var
    on_state:
      then:
        - lambda:
            if(x) {
              id(internal_display).show_component("action_btn");
            } else {
              id(internal_display).hide_component("action_btn"); 
            }
  
sensor: 
  - platform: homeassistant   # Outside temperature
    id: temperature_outside    
    entity_id: sensor.trisensor_air_temperature
    on_value:
      then:
        - lambda:
            char buffer [6];
            int n;
            n=sprintf (buffer, "%2.1f", x);
            
            ESP_LOGD("display", "Temperature is %s", buffer);
            id(nextion_temperature).set_state(buffer, false, true);
  - platform: homeassistant   # Outside light
    id: lights_outside    
    entity_id: sensor.trisensor_illuminance
    on_value:
      then:
        - lambda:
            float brightness;
        
            if(x < 100) {
              brightness = .02;
            } else if (x < 1000) {
              brightness = .08;
            } else {
              brightness = .12;
            }
            
            ESP_LOGD("display", "Brightness is %.2f", brightness);
            id(internal_display)->set_backlight_brightness(brightness);

the included `packages/core.yaml`

```yaml
# Enable logging
logger:
#  baud_rate: 0  # disables logging

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #power_save_mode: light
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Internal-Display"
    password: !secret hotspot_password
```
```

ssieb, 2023-04-26:
You didn't include the `display.yaml`.  Serial logs would also be very helpful.

ssieb, 2023-04-26:
Serial logs would be better.  But also, software serial doesn't work well at that speed.  You could try adding `debug:` to the `uart:` config to see if there's any communication.

closed 2023-04-27 by nagyv