esphome/issues #4463: ERROR Could not find auto generated code begin in file, either delete the main sketch file or insert the comment again.

spitfire, 2023-04-26:
### The problem

I'm getting:
`ERROR Could not find auto generated code begin in file, either delete the main sketch file or insert the comment again.`
I used to be able to compile it with versions <= 2023.4.0, but now it is failing.

### Which version of ESPHome has the issue?

2023.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP32

### Board

esp32dev

### Example YAML snippet

```yaml
esphome:
  name: foxess-inverter
  includes:
    - foxess_t_series.h

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: VERBOSE
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key 
    
ota:
  password: !secret esphome_ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret domain

  ap:
    ssid: "Inverter Fallback Hotspot"
    password: !secret esphome_ota
    

captive_portal:

#bluetooth_proxy:
  #active: true

uart:
  id: uart_bus
  tx_pin: 16
[... 295 lines cut ...]
    accuracy_decimals: 1
    
  - name: "T-Series Inverter State"
    id: "inverter_state"
    internal: true
    on_value:
      then:
        - lambda: |-
            if(x==0) {
              id(inverter_mode_text_sensor).publish_state("Offline");
            }
            else if(x==1) {
              id(inverter_mode_text_sensor).publish_state("Online");
            }
            else if(x==2) {
              id(inverter_mode_text_sensor).publish_state("Error");
            }
            else {
              id(inverter_mode_text_sensor).publish_state("Waiting for response...");
            }
```

### Additional information

The referenced/included .h file can be found at https://github.com/assembly12/Foxess-T-series-ESPHome-Home-Assistant/blob/main/foxess_t_series.h and also has not been changed.

ssieb, 2023-04-26:
You didn't include the logs.  If you haven't modified the `main.cpp` file, then try a clean build.

spitfire, 2023-04-26:
@ssieb which logs should I include?
How do I start a clean build?

ssieb, 2023-04-26:
The compile logs, wherever you're seeing that error message.
In the menu, there's an option to clean the build files.

spitfire, 2023-04-27:
Tried cleaning build files, the issue persists.
Here's the log output for installation:

```
INFO Reading configuration /config/esphome/foxess-inverter.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
ERROR Could not find auto generated code begin in file, either delete the main sketch file or insert the comment again.
```

spitfire, 2023-04-29:
Any idea what that is? my other ESPHome devices seem to be updating fine, so it's probably a matter of some component used in this one.

huenemeca, 2023-05-28:
I have 2 devices doing this now.

ssieb, 2023-05-28:
@spitfire do you use the custom component in any other devices?  Can you attach the `main.cpp` here if you can find it?  I think the data directory is accessible from the HA base OS.
@huenemeca are you using any custom components in those devices?  Same request as well for the `main.cpp` file.

spitfire, 2023-05-28:
> @spitfire do you use the custom component in any other devices? Can you attach the `main.cpp` here if you can find it? I think the data directory is accessible from the HA base OS. @huenemeca are you using any custom components in those devices? Same request as well for the `main.cpp` file.

All that I have in that project is the yaml file with content pasted in bug description along with the header (.h) file (from https://github.com/assembly12/Foxess-T-series-ESPHome-Home-Assistant/blob/main/foxess_t_series.h) next to it, there is no .cpp file in there (the only C++ code is in the yaml, which I'm attaching here: [foxess-inverter.yaml.zip](https://github.com/esphome/issues/files/11585199/foxess-inverter.yaml.zip)). This project used to compile with ESPHome 2023.4.*

ssieb, 2023-05-28:
You have to get it from the build directory.

spitfire, 2023-05-29:
It doesn't even create one. After I've cleaned last build files (as suggested earlier) it doesn't create a new one.
All other projects' files are there:
```
homeassistant01# pwd
/usr/share/hassio/addons/data/5c53de3b_esphome
homeassistant01# ls -l
total 4
drwxr-xr-x 1 root root 20 Apr 26 16:10 cache
drwxr-xr-x 1 root root 38 May 25 13:13 esp32-gabinet
drwxr-xr-x 1 root root 38 May 25 13:13 esp32-kotlownia
-rw------- 1 root root  2 May 29 03:24 options.json
drwxr-xr-x 1 root root 38 May 25 13:14 poziomwody-test
drwxr-xr-x 1 root root 38 May 25 13:14 t-can485-foxess
drwxr-xr-x 1 root root 38 May 29 09:41 ttgo01
drwxr-xr-x 1 root root 38 May 25 13:14 ttgo-dongle01
drwxr-xr-x 1 root root 38 May 29 09:42 ttgo-tdisplay-s3-01
drwxr-xr-x 1 root root 38 Apr 27 13:29 wodomierz
```
 I guess it fails when generating source.

spitfire, 2023-06-12:
Anything else I can check?

spitfire, 2023-07-29:
Still doesn't work, and hasn't been for 3 months now. Anything else I can try?

1nterlud3, 2023-10-18:
I also got this error. Any fixes or causes? It seams that the generation crashes.

```
> INFO ESPHome 2023.10.0
> INFO Reading configuration /config/esphome/inside-garage.yaml...
> INFO Generating C++ source...
> ERROR Could not find auto generated code begin in file, either delete the main sketch file or insert the comment again.
```

spitfire, 2023-10-18:
While cleaning the build files didn't help renaming hostname in ESPHome did ¯\_(ツ)_/¯

noispe, 2024-06-06:
Now I also got this error, but I was able to fix it without changing the hostname. I think my problem started with switching different versions of ESP-IDF.

Cleaning the build files has left some remains inside the build directory. Deleting the whole subdirectory of the device inside the build directory did the magic - my device now compiles fine again.

e.g.:
build path: `esphome/.esphome/build`
device name: `foxess-inverter`

-> delete directory (and everything inside): `esphome/.esphome/build/foxess-inverter`