esphome/issues #4465: Wiegand Unknown Bit

marcolinux46, 2023-04-28:
### The problem

I installed a wiegand FSTech keypad and RFID pad but i received the error in screenshoot.

### Which version of ESPHome has the issue?

2023.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

Wiegand Keypad and RFID

### Example YAML snippet

```yaml
esphome:
  name: wiegand-gate
  friendly_name: wiegand-gate

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80
  version: 2

wifi:
  ssid: "xxxx"
  password: "xxxx"

wiegand:
  - id: wggate
    d0: GPIO14 # Pin D5 wemos
    d1: GPIO12 # Pin D6 wemos
    on_key:
      - lambda: ESP_LOGI("KEY", "received key %d", x);
    on_tag:
      - lambda: ESP_LOGI("TAG", "received tag %s", x.c_str());
    on_raw:
      - lambda: ESP_LOGI("RAW", "received raw %d bits, value %llx", bits, value);
    on_tag:
      - lambda: ESP_LOGI("TAG", "received tag %s", x.c_str());
      - homeassistant.tag_scanned: !lambda 'return x;'
      - text_sensor.template.publish:
          id: wg_tag
          state: !lambda 'return x;'
      
key_collector:
  - id: wg_pincode
    source_id: wggate
    min_length: 4
    max_length: 4
    end_keys: "#"
    end_key_required: true
    back_keys: "*"
    clear_keys: "C"
    allowed_keys: "0123456789"
    timeout: 5s
    on_result:
      - logger.log:
          format: "input result: '%s', started by '%c', ended by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)", "(end == 0 ? '~' : end)" ]

# Text sensors
text_sensor:
  - platform: template
    name: "LAST WG Tag"
    id: wg_tag
```

### Additional information

[image]

ssieb, 2023-04-28:
That looks like you're getting random data.  Did you connect the grounds?

marcolinux46, 2023-04-28:
Hi @ssieb 

i connected black and red wires to 12Vdc, green(d0) and white(d1) wires to pins D1 and D2 on my wemos D1 mini.
Have i to connect other wires?

Thanks

ssieb, 2023-04-28:
Yes, you have to connect the black wire (assuming that's the ground side) also to the ESP GND.

marcolinux46, 2023-04-29:
> Yes, you have to connect the black wire (assuming that's the ground side) also to the ESP GND.

Connected ESP GND...... Now everything is working....

Thank you @ssieb

closed 2023-04-29 by ssieb