esphome/issues #4468: Issue with IR Remote Climate and toggle based protocols like Whirlpool and remote IR receiver

the-mentor, 2023-04-29:
### The problem

When configuring "IR Remote Climate" and toggle based protocols like Whirlpool and remote IR receiver is confusing the device causing it to turn on and off right away.

Disabling the IR receiver prevents this problem from happening.
My guess is that a fix for this is to disable the IR receiver when sending IR commands for protocols that are toggle based like whirlpool etc or maybe just disabling the IR receiver all together when sending IR commands.

Below is a recording of the issue and some logs I've gathered from the device

[Here is a recording of the issue](https://user-images.githubusercontent.com/11185446/235300535-bdbc8117-aa4a-4cc2-9f03-b7167ce53ee6.mp4) 

[logs_living-room-ir_logs.txt](https://github.com/esphome/issues/files/11359041/logs_living-room-ir_logs.txt)

### Which version of ESPHome has the issue?

2023.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
remote_transmitter:
  id: remote_transmitter_id
  pin: GPIO14
  carrier_duty_percent: 50%

remote_receiver:
  id: "remote_receiver_id"
  pin:
    number: GPIO5
    inverted: True
#    mode:
#      input: true
#      pullup: true
  tolerance: 55%
  dump: all

climate:
  - platform: whirlpool       # adjust to match your AC unit!
    name: "Living Room AC"
    model: "DG11J1-91"
    sensor: room_temperature
    transmitter_id: remote_transmitter_id
    receiver_id: remote_receiver_id
```

### Anything in the logs that might be useful for us?

```txt
[23:37:47][C][api:141]:   Using noise encryption: YES
[23:37:47][C][wifi_signal.sensor:009]: WiFi Signal 'Living Room IR WiFi Signal Sensor'
[23:37:47][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[23:37:47][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[23:37:47][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[23:37:47][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[23:37:47][C][homeassistant.sensor:030]: Homeassistant Sensor 'livingroom_temperature'
[23:37:47][C][homeassistant.sensor:030]:   State Class: ''
[23:37:47][C][homeassistant.sensor:030]:   Unit of Measurement: ''
[23:37:47][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[23:37:47][C][homeassistant.sensor:031]:   Entity ID: 'sensor.bt_living_room_temperature'
[23:37:59][D][climate:010]: 'Living Room AC' - Setting
[23:37:59][D][climate:014]:   Mode: COOL
[23:37:59][D][remote_transmitter:075]: Sending remote code...
[23:37:59][D][climate:392]: 'Living Room AC' - Sending state:
[23:37:59][D][climate:395]:   Mode: COOL
[23:37:59][D][climate:400]:   Fan Mode: AUTO
[23:37:59][D][climate:412]:   Swing Mode: OFF
[23:37:59][D][climate:415]:   Current Temperature: nan°C
[23:37:59][D][climate:421]:   Target Temperature: 24.00°C
[23:37:59][D][climate:392]: 'Living Room AC' - Sending state:
[23:37:59][D][climate:395]:   Mode: OFF
[23:37:59][D][climate:400]:   Fan Mode: AUTO
[23:37:59][D][climate:412]:   Swing Mode: OFF
[23:37:59][D][climate:415]:   Current Temperature: nan°C
[23:37:59][D][climate:421]:   Target Temperature: 24.00°C
```

juapem, 2023-06-16:
I am facing the same problem. I think the problem is in the whirlpool protocol. The same was happening to me using the TASMOTA IR with MQTT and ESPHOME

the-mentor, 2023-06-16:
@juapem when using Tasmota IR I don't have this issue.

The whirlpool protocol doesn't have on and off commands the on command is used to toggle the AC so when the unit is on it will power it off and when the unit is off it will turn it on.

Not ideal to say the least.

juapem, 2023-06-19:
Ok.. but then I don't understand why this isn't a problem when the remote control is used. If I put the AC in heat-mode (when is in heat-mode) is not turning off.

I think there is a problem in the implementation of this https://esphome.io/api/namespaceesphome_1_1whirlpool.html

juapem, 2023-06-19:
https://esphome.io/api/whirlpool_8cpp_source.html

`auto powered_on = this->[mode](https://esphome.io/api/classesphome_1_1climate_1_1_climate.html#a005cde5ea5156b4876c181eabab269f5) != [climate::CLIMATE_MODE_OFF](https://esphome.io/api/namespaceesphome_1_1climate.html#a8716cd7ed01ceb7ccedfa2f7b47a39d2aad53e3902481a8a370e5cebdf00c4f2f);
   if (powered_on != this->[powered_on_assumed](https://esphome.io/api/classesphome_1_1whirlpool_1_1_whirlpool_climate.html#a8dfd58e8324f23b0637e3968e1675b4a)) {
     // Set power toggle command
     remote_state[2] = 4;
     remote_state[15] = 1;
     this->[powered_on_assumed](https://esphome.io/api/classesphome_1_1whirlpool_1_1_whirlpool_climate.html#a8dfd58e8324f23b0637e3968e1675b4a) = powered_on;`

I believe the problem is there

the-mentor, 2023-06-19:
From the little conversation I had with @jesserockz on discord a few months ago it seems that the receiver needs to be disabled for the short time the device is blasting IR.

But I might be wrong

TokarevSergey, 2023-07-02:
Добрый день, у меня есть кондиционер Galactic в корпусе Whirlpool. У меня такая же проблема как у вас, подскажите пожалуйста как решить?

rdbahm, 2023-07-02:
I think this is the same issue as #3480 - though I wrote that only about one specific receiver.

The-mentor's understanding matches my own - ideally, this is fixed by disabling the RX while TX is happening.

TokarevSergey, 2023-07-03:
> Я думаю, что это та же проблема, что и [№ 3480](https://github.com/esphome/issues/issues/3480) , хотя я написал это только об одном конкретном приемнике.
> 
> Понимание наставника совпадает с моим — в идеале это можно исправить, отключив RX во время передачи.

That is, I cannot do something on my part to resolve this issue.  Just wait until they hear us and correct us in integration?  Tell me, if they share it directly to rx tx or to the LED, will this not solve the problem?  Thank you.

rdbahm, 2023-07-07:
> That is, I cannot do something on my part to resolve this issue. Just wait until they hear us and correct us in integration? Tell me, if they share it directly to rx tx or to the LED, will this not solve the problem? Thank you.
I tried a few electrical options to work around the issue - I expect if you wired TX to the AC's RX directly, and used your own RX, it would work as expected. That didn't work for my setup.

A workaround you can do is to add a "manual" transmit that bypasses the IR Climate integration. On my Midea heat pump, I made a new Esphome service like this:
```
api:
  services:
    - service: toggle_turbo
      then:
        - script.execute: toggle_turbo
script:
  - id: toggle_turbo
    then:
      - remote_transmitter.transmit_midea:
          code: [0xA2, 0x09, 0xFF, 0xFF, 0xFF]
```
Obviously this will vary based on model, but you can capture the code by enabling "dump: all" in your remote_receiver section and enabling debug logging, then using your real IR remote to send the code you want to replicate.

Once you've done all that, you can call your custom service from Home Assistant - the IR is sent without triggering the bug.

TokarevSergey, 2023-07-07:
> > То есть я не могу что-то сделать со своей стороны для решения этого вопроса. Просто ждать, пока нас услышат и поправят в интеграции? Скажите, если они поделят его напрямую на rx tx или на LED, это не решит проблему? Спасибо.
> 
> Я попробовал несколько электрических вариантов, чтобы решить эту проблему - я ожидаю, что если вы подключите TX к RX переменного тока напрямую и используете свой собственный RX, он будет работать, как и ожидалось. Это не сработало для моей установки.
> 
> Обходной путь, который вы можете сделать, это добавить «ручную» передачу, которая обходит интеграцию IR Climate. На своем тепловом насосе Midea я сделал новый сервис Esphome, подобный этому:
> 
> ```
> api:
>   services:
>     - service: toggle_turbo
>       then:
>         - script.execute: toggle_turbo
> script:
>   - id: toggle_turbo
>     then:
>       - remote_transmitter.transmit_midea:
>           code: [0xA2, 0x09, 0xFF, 0xFF, 0xFF]
> ```
> 
> Очевидно, что это зависит от модели, но вы можете записать код, включив «dump: all» в разделе remote_receiver и включив ведение журнала отладки, а затем используя свой настоящий ИК-пульт для отправки кода, который вы хотите воспроизвести.
> 
> После того, как вы все это сделали, вы можете вызвать свою пользовательскую службу из Home Assistant — IR отправляется без срабатывания ошибки.

Thank you.  I'll try to figure it out.  For me it is difficult and incomprehensible.  Could you be a little more specific.  Thank you.

TokarevSergey, 2023-07-07:
> Я попробовал несколько электрических вариантов, чтобы решить эту проблему - я ожидаю, что если вы подключите TX к RX переменного тока напрямую и используете свой собственный RX, он будет работать, [...]

I described my problem in detail here.
https://github.com/esphome/issues/issues/4650

closed 2023-09-29 by the-mentor