esphome/issues #4469: ESP8266 - fastled_clockless compile errors with ESPHome version v2023.4.2

jsuanet, 2023-04-29:
### The problem

When compiling an ESP8266 device using the fastled_clockless component I got compilation errors.

### Which version of ESPHome has the issue?

v2023.4.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

D1_mini

### Component causing the issue

fastled_clockless in combination with entity_base

### Example YAML snippet

```yaml
esphome:
  name: test-8266

esp8266:
  board: d1_mini
  framework:
    version: 2.7.4

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
   
ota:
  password: !secret ota_password

logger:
  level: DEBUG

light:
  - platform: fastled_clockless
    chipset: WS2813
    pin: D5
    num_leds: 60
    rgb_order: GRB
    default_transition_length: 50ms
    id: my_led
    name: "Led-strip"
```

### Anything in the logs that might be useful for us?

```txt
-> esphome-yaml esphome compile test-8266.yaml
INFO Reading configuration test-8266.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
**********************************************************************************************************************************************************************************************
Obsolete PIO Core v6.1.5 is used (previous was 6.1.6a4)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
**********************************************************************************************************************************************************************************************
[... 203 lines cut ...]
   return lhs.size() == rhs.size() && std::equal(std::begin(lhs), std::end(lhs), std::begin(rhs));
              ^
src/esphome/core/string_ref.h: In function 'bool esphome::operator==(const esphome::StringRef&, const char*)':
src/esphome/core/string_ref.h:93:14: error: 'const class esphome::StringRef' has no member named 'size'
   return lhs.size() == strlen(rhs) && std::equal(std::begin(lhs), std::end(lhs), rhs);
              ^
src/esphome/core/string_ref.h: In function 'std::string& esphome::operator+=(std::string&, const esphome::StringRef&)':
src/esphome/core/string_ref.h:113:31: error: 'const class esphome::StringRef' has no member named 'size'
   lhs.append(rhs.c_str(), rhs.size());
                               ^
src/esphome/core/string_ref.h: In function 'std::string esphome::operator+(const char*, const esphome::StringRef&)':
src/esphome/core/string_ref.h:119:31: error: 'const class esphome::StringRef' has no member named 'size'
   str.append(rhs.c_str(), rhs.size());
                               ^
Compiling .pioenvs/test-8266/src/esphome/core/log.cpp.o
src/esphome/core/string_ref.h: In member function 'constexpr bool esphome::StringRef::empty() const':
src/esphome/core/string_ref.h:69:52: warning: control reaches end of non-void function [-Wreturn-type]
   constexpr bool empty() const { return len_ == 0; }
                                                    ^
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/test-8266/src/esphome/core/entity_base.cpp.o] Error 1
Compiling .pioenvs/test-8266/src/esphome/core/scheduler.cpp.o
================================================================================= [FAILED] Took 9.12 seconds =================================================================================
```

NiamoR42, 2023-05-01:
Same on my side

ssieb, 2023-05-01:
Duplicate of https://github.com/esphome/issues/issues/4449

closed 2023-05-01 by ssieb

ssieb, 2023-05-03:
This is temporarily fixed in the latest point release that is just coming out.

ssieb, 2023-05-03:
https://github.com/esphome/esphome/pull/4777

jsuanet, 2023-05-04:
👍