esphome/issues #4473: Restore_From_Flash suddenly stopped working after updating HA

rajkhan0714, 2023-05-01:
### The problem

I updated Home Assistant yesterday and all of a sudden all my ESP devices stopped restoring states after power cut.

### Which version of ESPHome has the issue?

2023.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

restore_from_flash

### Example YAML snippet

```yaml
esphome:
  name: "bed-switch-esp01"

esp8266:
  board: esp01_1m
  board_flash_mode: qio
  restore_from_flash: true
  
preferences:
  flash_write_interval: 5min
  
# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  domain: .local
  reboot_timeout: 15min
  fast_connect: true
   

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esp01BedSwitch Fallback"
    password: !secret wifi_password

captive_portal:

web_server:
  port: 80
  auth:
    username: !secret web_user
    password: !secret web_password

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600
  
switch:
  - platform: gpio
    name: "Light 1"
    pin: 3
    inverted: yes

  - platform: gpio    
    name: "Light 3"
    pin: 0
    inverted: yes
    
  - platform: gpio    
    name: "wipro power"
    pin: 1
    inverted: yes

  - platform: gpio    
    name: "fan"
    pin: 2
    inverted: yes

  - platform: restart
    id: bed_switch_esp01_restart
    name: “Restart”
```

nuk3man, 2023-05-24:
I have the same issue. Can we get some sort of update on this?

rajkhan0714, 2023-05-24:
> I have the same issue. Can we get some sort of update on this?

Hi, adding  (restore_mode: 'RESTORE_DEFAULT_ON') with every switch worked for me. Now I can again save state in flash.

new code:

esphome:
  name: "bed-switch-esp01"

esp8266:
  board: esp01_1m
  board_flash_mode: qio
  restore_from_flash: true
  
preferences:
  flash_write_interval: 30s
  
# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  domain: .local
  reboot_timeout: 15min
  fast_connect: true
   

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esp01BedSwitch Fallback"
    password: !secret wifi_password

captive_portal:

web_server:
  port: 80
  auth:
    username: !secret web_user
    password: !secret web_password

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600
  
switch:
  - platform: gpio
    name: "Light 1"
    pin: 1
    inverted: yes
    restore_mode: 'RESTORE_DEFAULT_ON'

  - platform: gpio    
    name: "Light 3"
    pin: 2
    inverted: yes
    restore_mode: 'RESTORE_DEFAULT_ON'
    
  - platform: gpio    
    name: "wipro power"
    pin: 3
    inverted: yes
    restore_mode: 'RESTORE_DEFAULT_ON'

  - platform: gpio    
    name: "fan"
    pin: 0
    inverted: yes
    restore_mode: 'RESTORE_DEFAULT_ON'

  - platform: restart
    id: bed_switch_esp01_restart
    name: “Restart”

nuk3man, 2023-05-24:
> Hi, adding (restore_mode: 'RESTORE_DEFAULT_ON') with every switch worked for me. Now I can again save state in flash.
>
> new code:
>
> esphome: name: "bed-switch-esp01"
>
> esp8266: board: esp01_1m [...]

Thanks, I'll try that as soon as I can. Problem is I don't necessarily want them as default on. Does it work for you such that they also revert to the correct state?

closed 2023-05-24 by ssieb

rajkhan0714, 2023-05-24:
Well, for me it stores the state and restores after power failure. Do try and let us know if it works for you too.

nuk3man, 2023-05-27:
> Well, for me it stores the state and restores after power failure. Do try and let us know if it works for you too.

Tried it and unfortunately, it doesn't help. However, I can partially circumvent the problem by running the script that turns them on/off as lambda instead of regular ESP yaml. Still frustrating though

ssieb, 2023-05-27:
How does that circumvent the problem?  That makes no sense at all.  Come to the discord server and paste your config and explanation of the problem in a thread.

nuk3man, 2023-05-27:
> How does that circumvent the problem? That makes no sense at all. Come to the discord server and paste your config and explanation of the problem in a thread.

I would love to know as well. However, it solves the problem that my light switches are turned on and off when they're supposed to, irrespective if they lose connection/power.