esphome/issues #4474: Can't flash a "Fastled" application

ilmisu, 2023-05-01:
### The problem

ESPHome compiler stops with an unclear error, even if the Yaml has no one.

### Which version of ESPHome has the issue?

2023.04.01

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.04

### What platform are you using?

ESP8266

### Board

Nodemcu

### Component causing the issue

Fastled with P9813 chip

### Example YAML snippet

```yaml
esphome:
  name: stripled
  friendly_name: stripled

esp8266:
  board: nodemcuv2
  framework:
    version: 2.7.4

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

wifi:
  networks:
  - ssid: ""  
    password: ""
    bssid: "D0:21:F9:67:52:46"
    hidden: yes
    manual_ip:
      static_ip: 192.168.1.69
      subnet: 255.255.255.0
      gateway: 192.168.1.1
  - ssid: "Casa Misuriello PT"
    password: "15settembre" 
    manual_ip:
      static_ip: 192.168.1.69
      subnet: 255.255.255.0
      gateway: 192.168.1.1      
  - ssid: ""
    password: "" 
    manual_ip:
      static_ip: 192.168.1.69
      subnet: 255.255.255.0
      gateway: 192.168.1.1  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Stripled Fallback Hotspot"
    password: "15settembre"
web_server:
  port: 80
captive_portal:
light:
  - platform: fastled_spi
    chipset: P9813
    data_pin: D6
    clock_pin: D7  # D1
    num_leds: 3
    rgb_order: RGB
    name: "Ledstrip"
    id: ledstrip
```

### Anything in the logs that might be useful for us?

```txt
Last bunch of lines from compiler:

Linking /data/stripled/.pioenvs/stripled/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/stripled/.pioenvs/stripled/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o:(.text._ZN7esphome6socket11LWIPRawImplD2Ev[esphome::socket::LWIPRawImpl::~LWIPRawImpl()]+0x4): undefined reference to `vtable for esphome::socket::Socket'
collect2: error: ld returned 1 exit status
*** [/data/stripled/.pioenvs/stripled/firmware.elf] Error 1
========================= [FAILED] Took 33.22 seconds =========================
```

ssieb, 2023-05-01:
Duplicate of https://github.com/esphome/issues/issues/4449

closed 2023-05-01 by ssieb

ssieb, 2023-05-03:
This is temporarily fixed in the latest point release that is just coming out.

ssieb, 2023-05-03:
https://github.com/esphome/esphome/pull/4777