esphome/issues #4475: ac_dimmer component doesn't dim correctly anymore

xanderdekok, 2023-05-01:
### The problem

I used the ac_dimmer component for a while without issues. I did not update the device until recently. After the update the dimmer does not work as it used before.
Most of the time it works perfectly fine, but every couple of minutes the lights 'turn off' for about half a second and then come back on again.

I'm wondering what causes this dip. I don't know exactly what version of ac_dimmer I used before. It might have been a version from 2021 or even 2020. Can somebody tell me how I can use an older version of the ac_dimmer component? I would like to test if this problem is caused by an update to the ac_dimmer component, or something else that might have changed due to upgrades to ESP Home.

### Which version of ESPHome has the issue?

2023.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.4

### What platform are you using?

ESP8266

### Board

ESP8266

### Component causing the issue

https://esphome.io/components/output/ac_dimmer.html?highlight=dimmer

swbouman, 2023-05-08:
Same problem here. Looks like an io problem. Everytime another sensor updates, the lights are blinking.

xanderdekok, 2023-05-08:
> Same problem here. Looks like an io problem. Everytime another sensor updates, the lights are blinking.

Did you always have this problem, or only after updating the ESP device?

swbouman, 2023-05-08:
I first noticed the problem in esphome 2023.3.X. 

I’ll downgrade to 2023.1.X to see if the problem disappears.

xanderdekok, 2023-05-08:
For me the problem was also there in a 2022 release. Not sure which version I upgraded to, after at least a year of not upgrading my device.

closed 2023-09-20 by github-actions[bot]