esphome/issues #4480: Adopting esp32-s3-devkitc-1 board fails with ESPHome 2023.4.2 and Home Assistant 2023.4.6

lboue, 2023-05-02:
### The problem

collect2: fatal error: ld terminated with signal 9 [Killed]

```
INFO Reading configuration /config/esphome/esphome-web-d16f90.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-d16f90 (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
```

### Which version of ESPHome has the issue?

2023.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP32

### Board

esp32-s3-devkitc-1

### Component causing the issue

Adopting board

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esphome-web-d16f90.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-d16f90 (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/src/esphome/components/api/api_connection.cpp.o
...
Compiling /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/FrameworkArduino/firmware_msc_fat.c.o
Compiling /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/FrameworkArduino/libb64/cdecode.c.o
Compiling /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/FrameworkArduino/libb64/cencode.c.o
Compiling /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/FrameworkArduino/main.cpp.o
Compiling /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/FrameworkArduino/stdlib_noniso.c.o
Compiling /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/FrameworkArduino/wiring_pulse.c.o
Compiling /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/FrameworkArduino/wiring_shift.c.o
Archiving /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/libFrameworkArduino.a
Linking /data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/firmware.elf
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
*** [/data/esphome-web-d16f90/.pioenvs/esphome-web-d16f90/firmware.elf] Error 1
========================= [FAILED] Took 798.40 seconds =========================
```
```

### Additional information

platformio/espressif32 @ 5.3.0

randybb, 2023-05-02:
13 minutes? On a proper machine it doesn't takes more than 1 minute... If you have RPI3, then https://esphome.io/changelog/2022.11.0.html?#running-esphome-on-lower-powered-machines or if you are running it in a VM, then it might have not enough memory.

lboue, 2023-05-02:
> 13 minutes? On a proper machine it doesn't takes more than 1 minute... If you have RPI3, then https://esphome.io/changelog/2022.11.0.html?#running-esphome-on-lower-powered-machines or if you are running it in a VM, then it might have not enough memory.

HA runs on my Raspberry Pi 3 Model B.

ssieb, 2023-05-02:
You're running out of RAM.  You'll need to compile on something else.

closed 2023-05-02 by ssieb