esphome/issues #4483: Pid climate P term calculation issue

juuzonieminen, 2023-05-02:
### The problem

Every time when error (setpoint-current value) goes to 0, P-term does not go 0 as it should, but it goes -2% or similar, depending on kP.

### Which version of ESPHome has the issue?

2023.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

9.5

### What platform are you using?

ESP8266

juuzonieminen, 2023-05-02:
[image]

lewissbaker, 2023-07-10:
I've also been experiencing something similar for a temp PID controller.
When error term small value the p term is small (<1%), but when error term goes to zero, p term jumps to 15%.

Looking at the code in `esphome/components/pid/pid_controller.cpp`, a couple of things seem odd:

The [PIDController::in_deadband()](https://github.com/esphome/esphome/blob/ddde1ee31ef96476c7a7d1c8b8fe306aa4d2e503/esphome/components/pid/pid_controller.cpp#L29-L33) function has the following:
```
  float err = -error_;
  return ((err > 0 && err < threshold_high_) || (err < 0 && err > threshold_low_));
```
Which seems like it would actually return false when the `err` term is exactly zero.

This would then cause the logic in `PIDController::calculate_proportional_term_` to then fall through into the else-branch which applies a pdm_offset (equal to `(threshold_low_ - (kp_multiplier_ * threshold_low_)) * kp_`.

Could the `in_deadband()` method be changed to the following?
```
  float err = -error_;
  return (threshold_low_ <= err && err <= threshold_high_);
```

closed 2023-07-12 by jesserockz