esphome/issues #4485: Adding substitutions with packages raise an error

cavonlee, 2023-05-04:
### The problem

get both substitutions and packages
raise error: 'dict' object has no attribute 'move_to_end'.

leaving one of only substitutions or packages, works well. but not both

### Which version of ESPHome has the issue?

2023.4.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP32

### Board

ESP32 S2

### Component causing the issue

packages, substitutions

### Example YAML snippet

```yaml
substitutions:
  factory_max_capacity: "2000"

esphome:
  name: pipower-pro
  friendly_name: PiPower-Pro

esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "wuJrEm4L6udD3LDWObzzhUzJMRAQH8GBZQljsX5AQi4="

ota:
  password: "a2f47d3677ff92ba9ef2a2ba274e68c3"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Pipower-Pro Fallback Hotspot"
    password: "IRRW7cbQo71j"

captive_portal:

packages:
  sunfounder.pipower-pro: github://sunfounder/pipower-pro/pipower-pro-template.yaml@main
  # pipower-pro: !include pipower-pro-template.yaml
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/pipower-pro.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 965, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 943, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/esphome/esphome/config.py", line 986, in read_config
    res = load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 840, in load_config
    return _load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 828, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/esphome/esphome/config.py", line 690, in validate_config
    substitutions.do_substitution_pass(config, command_line_substitutions)
  File "/esphome/esphome/components/substitutions/__init__.py", line 154, in do_substitution_pass
    config.move_to_end(CONF_SUBSTITUTIONS, False)
AttributeError: 'dict' object has no attribute 'move_to_end'
```

OstenTV, 2023-07-19:
I have found that the error goes away if you change the package format from:

`packages:
  remote_package: github://esphome/non-existant-repo/file1.yml@main`
to:
`packages:
  remote_package:
    url: https://github.com/esphome/non-existant-repo
    ref: main
    file: file.yaml`
https://esphome.io/guides/configuration-types.html#config-git-packages

This does not solve the issue of dashboard_import not working, because it is using the broken short format.
https://github.com/EverythingSmartHome/everything-presence-one/issues/79

dentra, 2023-08-11:
Most likely there was an exception when [loading](https://github.com/esphome/esphome/blob/5cb21324a1f0045a2ea4cb8c395e9d7976f2279c/esphome/components/packages/__init__.py#L137) the package, but it was swallowed and prints only "Unexpected exception while reading configuration"

closed 2023-10-25 by jesserockz