esphome/issues #4489: ILI9488 does not work in ESPHOME on lolin_d32 board but with same hardware works in arduino IDE with TFT_eSPI library.

wiigor, 2023-05-05:
### The problem

I have an ILI9488 display. I used [this tutorial](https://www.youtube.com/watch?v=rq5yPJbX_uk&t=125s) to make the display work using arduino IDE and this library: TFT_eSPI. I had to configure the User_Setup.H to set the correct board and pins

#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)

`// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)`

The display works great using all examples in arduino IDE.
Now without changing any hardware connections using the same hardware setup I try to run the display using esphome:

This is my config>
`esphome:
  name: air-quality-meter2

esp32:
  board: lolin_d32
  # board: esp32dev                                                                                                                                                                                                                                                                                                                    
  framework:
    type: arduino
    
# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
#api:
  #password: ""

ota:
  password: ""

wifi: !include includes/wifisp.yaml

# captive_portal:

mqtt:
  broker: 192.168.1.100

font:
  - file: "fonts/arialn.ttf"
    id: my_font
    size: 20
  - file: "fonts/Helvetica.ttf"
    id: helvetica_48
    size: 48
  - file: "fonts/Helvetica.ttf"
    id: helvetica_24
    size : 24
  - file: "fonts/Helvetica.ttf"
    id: helvetica_S
    size: 14

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%

i2c:
  sda: 21
  scl: 22
  scan: true

spi:
  clk_pin: 18
  mosi_pin: 23

display:
  - platform: ili9xxx
    model: ILI9488
    cs_pin: 15
    dc_pin: 2
    reset_pin: 4
    lambda: |-
      it.fill(Color::BLACK);
      it.print(0, 0, id(my_font), id(my_red), TextAlign::TOP_LEFT, "Hello World!");`

The log output:

`[16:29:17][I][app:102]: ESPHome version 2023.3.2 compiled on May  5 2023, 16:28:25
[16:29:17][C][wifi:504]: WiFi:
[16:29:17][C][wifi:363]:   SSID: 'MOVISTAR'
[16:29:17][C][wifi:364]:   IP Address: 192.168.1.132
[16:29:17][C][wifi:366]:   BSSID: 60:8D:26:BB:FA:3A
[16:29:17][C][wifi:367]:   Hostname: 'air-quality-meter2'
[16:29:18][C][wifi:369]:   Signal strength: -57 dB ▂▄▆█
[16:29:18][V][wifi:371]:   Priority: 0.0
[16:29:18][C][wifi:373]:   Channel: 6
[16:29:18][C][wifi:374]:   Subnet: 255.0.0.0
[16:29:18][C][wifi:375]:   Gateway: 192.168.1.1
[16:29:18][C][wifi:376]:   DNS1: 0.0.0.0
[16:29:18][C][wifi:377]:   DNS2: 0.0.0.0
[16:29:18][C][logger:293]: Logger:
[16:29:18][C][logger:294]:   Level: VERBOSE
[16:29:18][C][logger:295]:   Log Baud Rate: 115200
[16:29:18][C][logger:296]:   Hardware UART: UART0
[16:29:18][C][i2c.arduino:052]: I2C Bus:
[16:29:18][C][i2c.arduino:053]:   SDA Pin: GPIO21
[16:29:18][C][i2c.arduino:054]:   SCL Pin: GPIO22
[16:29:18][C][i2c.arduino:055]:   Frequency: 50000 Hz
[16:29:18][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[16:29:18][I][i2c.arduino:068]: Results from i2c bus scan:
[16:29:18][I][i2c.arduino:070]: Found no i2c devices!
[16:29:18][C][spi:101]: SPI bus:
[16:29:18][C][spi:102]:   CLK Pin: GPIO18
[16:29:18][C][spi:104]:   MOSI Pin: GPIO23
[16:29:18][C][spi:106]:   Using HW SPI: YES
[16:29:18][C][ili9xxx:047]: ili9xxx
[16:29:18][C][ili9xxx:047]:   Rotations: 0 °
[16:29:18][C][ili9xxx:047]:   Dimensions: 480px x 320px
[16:29:18][C][ili9xxx:056]:   Color mode: 8bit 332 mode
[16:29:18][C][ili9xxx:060]:   18-Bit Mode: YES
[16:29:18][C][ili9xxx:063]:   Reset Pin: GPIO4
[16:29:18][C][ili9xxx:064]:   DC Pin: GPIO2
[16:29:18][C][ili9xxx:068]:   => Failed to init Memory: YES!
[16:29:18][C][ili9xxx:070]:   Update Interval: 1.0s
[16:29:18][C][psram:020]: PSRAM:
[16:29:18][C][psram:021]:   Available: NO
[16:29:18][C][mdns:108]: mDNS:
[16:29:18][C][mdns:109]:   Hostname: air-quality-meter2
[16:29:18][V][mdns:110]:   Services:
[16:29:18][V][mdns:112]:   - _http, _tcp, 80
[16:29:18][V][mdns:114]:     TXT: version = 2023.3.2`

I notice this message: [16:29:18][C][ili9xxx:068]:   => Failed to init Memory: YES!
Could this be the reason the display does not work?

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.3.2

### What platform are you using?

ESP32

### Board

lolin_d32

### Component causing the issue

ILI9xxx TFT LCD Series

### Example YAML snippet

```yaml
esphome:
  name: air-quality-meter2

esp32:
  board: lolin_d32
  # board: esp32dev                                                                                                                                                                                                                                                                                                                    
  framework:
    type: arduino
    
# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
#api:
  #password: ""

ota:
  password: ""

wifi: !include includes/wifisp.yaml

# captive_portal:

mqtt:
  broker: 192.168.1.100

font:
  - file: "fonts/arialn.ttf"
    id: my_font
    size: 20
  - file: "fonts/Helvetica.ttf"
    id: helvetica_48
    size: 48
  - file: "fonts/Helvetica.ttf"
    id: helvetica_24
    size : 24
  - file: "fonts/Helvetica.ttf"
    id: helvetica_S
    size: 14

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%

i2c:
  sda: 21
  scl: 22
  scan: true

spi:
  clk_pin: 18
  mosi_pin: 23

display:
  - platform: ili9xxx
    model: ILI9488
    cs_pin: 15
    dc_pin: 2
    reset_pin: 4
    lambda: |-
      it.fill(Color::BLACK);
      it.print(0, 0, id(my_font), id(my_red), TextAlign::TOP_LEFT, "Hello World!");
```

### Anything in the logs that might be useful for us?

```txt
[16:29:18][C][spi:101]: SPI bus:
[16:29:18][C][spi:102]:   CLK Pin: GPIO18
[16:29:18][C][spi:104]:   MOSI Pin: GPIO23
[16:29:18][C][spi:106]:   Using HW SPI: YES
[16:29:18][C][ili9xxx:047]: ili9xxx
[16:29:18][C][ili9xxx:047]:   Rotations: 0 °
[16:29:18][C][ili9xxx:047]:   Dimensions: 480px x 320px
[16:29:18][C][ili9xxx:056]:   Color mode: 8bit 332 mode
[16:29:18][C][ili9xxx:060]:   18-Bit Mode: YES
[16:29:18][C][ili9xxx:063]:   Reset Pin: GPIO4
[16:29:18][C][ili9xxx:064]:   DC Pin: GPIO2
[16:29:18][C][ili9xxx:068]:   => Failed to init Memory: YES!
[16:29:18][C][ili9xxx:070]:   Update Interval: 1.0s
[16:29:18][C][psram:020]: PSRAM:
[16:29:18][C][psram:021]:   Available: NO
[16:29:18][C][mdns:108]: mDNS:
[16:29:18][C][mdns:109]:   Hostname: air-quality-meter2
[16:29:18][V][mdns:110]:   Services:
[16:29:18][V][mdns:112]:   - _http, _tcp, 80
[16:29:18][V][mdns:114]:     TXT: version = 2023.3.2`
```

wiigor, 2023-05-05:
Working config of the TFT_eSPI library in arduino IDE that leads the screen to work with the same hardware configuration:

#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)

`// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)`

The display with the lolin_d32 board works great using all examples in arduino IDE using TFT_eSPI .

rakoan, 2023-06-06:
i have the same issue with a ili9488 (+model: ILI9488 in the yaml).
Changing the lcd screen to an ili9341 (+ changing with **model: ILI9341** in the yaml ) with the same pin connections works perfectly.

Gyosa3, 2023-07-27:
Hi, same problem here, using a "3.5inch RPi Display" device with an ESP32S v4 and display fails to initialise.
When using ILI9341 there is no "Failed to init memory" message but it obviously does not work.

any fix would be appreciated !

nielsnl68, 2023-07-27:
@Gyosa3 could you give more details about which display you have (link to the display) and how you connecting it and what is exactly the YAML file look like.
Any logs are also welcome,

wibear-gh, 2025-05-23:
Hello, I have exact the same problem and wasted already a lot of time with it.
Any help will be highly appreciated! Thanks in advance and regards.

Here is the yaml part with a display:

`# Display configuration

spi:
  clk_pin: GPIO18 # SCK
  mosi_pin: GPIO19 # SDI
  miso_pin: GPIO23 # SDO
  
display:
  - platform: ili9xxx
    model: ILI9488
    dc_pin: GPIO2 # DC / RS
    reset_pin: GPIO4 
    cs_pin: GPIO15 # chip select
    color_palette: 8BIT
    invert_colors: false
    rotation: 0
    update_interval: 1s 
    lambda: |-
      it.print(0, 0, id(Verdana16), "Hello World!");`

And here is a message in the log:

`[16:17:31][C][wifi:441]:   Hostname: 'displaytft'
[16:17:31][C][wifi:443]:   Signal strength: -81 dB ▂▄▆█
[16:17:31][C][wifi:447]:   Channel: 11
[16:17:31][C][wifi:448]:   Subnet: 255.255.255.0
[16:17:31][C][wifi:449]:   Gateway: 192.168.178.1
[16:17:31][C][wifi:450]:   DNS1: 192.168.178.1
[16:17:31][C][wifi:451]:   DNS2: 0.0.0.0
[16:17:31][C][logger:203]: Logger:
[16:17:31][C][logger:204]:   Max Level: DEBUG
[16:17:31][C][logger:205]:   Initial Level: DEBUG
[16:17:31][C][logger:207]:   Log Baud Rate: 115200
[16:17:31][C][logger:208]:   Hardware UART: UART0
[16:17:31][C][logger:212]:   Task Log Buffer Size: 768
[16:17:31][C][spi:068]: SPI bus:
[16:17:31][C][spi:069]:   CLK Pin: GPIO18
[16:17:31][C][spi:070]:   SDI Pin: GPIO23
[16:17:31][C][spi:071]:   SDO Pin: GPIO19
[16:17:31][C][spi:076]:   Using HW SPI: SPI
[16:17:31][C][ili9xxx:091]: ili9xxx
[16:17:31][C][ili9xxx:091]:   Rotations: 0 °
[16:17:31][C][ili9xxx:091]:   Dimensions: 320px x 480px
[16:17:31][C][ili9xxx:092]:   Width Offset: 0
[16:17:31][C][ili9xxx:093]:   Height Offset: 0
[16:17:31][C][ili9xxx:102]:   Color mode: 8bit 332 mode
[16:17:31][C][ili9xxx:106]:   18-Bit Mode: YES
[16:17:31][C][ili9xxx:108]:   Data rate: 40MHz
[16:17:31][C][ili9xxx:110]:   Reset Pin: GPIO4
[16:17:31][C][ili9xxx:111]:   CS Pin: GPIO15
[16:17:31][C][ili9xxx:112]:   DC Pin: GPIO2
[16:17:31][C][ili9xxx:114]:   Color order: BGR
[16:17:31][C][ili9xxx:115]:   Swap_xy: NO
[16:17:31][C][ili9xxx:116]:   Mirror_x: NO
[16:17:31][C][ili9xxx:117]:   Mirror_y: NO
[16:17:31][C][ili9xxx:118]:   Invert colors: NO
[16:17:31][C][ili9xxx:121]:   => Failed to init Memory: YES!
[16:17:31][C][ili9xxx:123]:   Update Interval: 1.0s
[16:17:31][E][component:085]:   Component display is marked FAILED: `

wibear-gh, 2025-05-24:
I found out, that the problem is the missing psram. I'll try it with ESP32-S3 with PSRAM.

closed 2025-05-27 by clydebarrow