esphome/issues #4490: Issue/Config ?  DSMR 2.2 Gas readings not being plubished to HA

AussiSG, 2023-05-05:
### The problem

As mentioned in the title

It seems that my gas reading is not published to HA.
I do get the reading in the serial data. 

But it doesn't seem to be translated properly?

I have a DSMR 2.2(+) Landis & Gyr E350 with DSMR 2.2(+)

### Which version of ESPHome has the issue?

2023.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP8266

### Board

D1-Mini

### Component causing the issue

DSMR

### Example YAML snippet

```yaml
substitutions:
  name: p1wifigateway

esphome:
  name: ${name}
  friendly_name: p1wifigateway
  platform: esp8266
  board: d1_mini
  esp8266_restore_from_flash: true 

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "oEOXmPj7exkRbcXWS7QSrsp3j9av0g508YKLxk14qo4="

ota:
  password: "a3da4d31ad18d2c02898fb3571774c41"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ""
    password: ""

captive_portal:

dsmr:
  crc_check: false
  gas_mbus_id: 1

uart:
  id: uart_bus
  baud_rate: 9600
  data_bits: 7
  parity: NONE
  stop_bits: 1
  tx_pin:
    number: TX
  rx_pin: 
    number: RX
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

sensor:
  - platform: dsmr
    energy_delivered_tariff1:
      name: Energy Consumed Tariff 1
    energy_delivered_tariff2:
      name: Energy Consumed Tariff 2
    energy_returned_tariff1:
      name: Energy Produced Tariff 1
    energy_returned_tariff2:
      name: Energy Produced Tariff 2
    power_delivered:
      name: Power Consumed
    power_returned:
      name: Power Produced
    electricity_failures:
      name: Electricity failures
    electricity_long_failures:
      name: Long Electricity failures
    voltage_l1:
      name: Voltage Phase 1
    voltage_l2:
      name: Voltage Phase 2
    voltage_l3:
      name: Voltage Phase 3
    current_l1:
      name: Current Phase 1
    current_l2:
      name: Current Phase 2
    current_l3:
      name: Current Phase 3
    power_delivered_l1:
      name: Power Consumed Phase 1
    power_delivered_l2:
      name: Power Consumed Phase 2
    power_delivered_l3:
      name: Power Consumed Phase 3
    power_returned_l1:
      name: Power Produced Phase 1
    power_returned_l2:
      name: Power Produced Phase 2
    power_returned_l3:
      name: Power Produced Phase 3
    gas_delivered:
      name: Gas Consumed

text_sensor:
  - platform: dsmr
    identification:
      name: "dsmr_identification"
    p1_version:
      name: "dsmr_p1_version"
    gas_delivered_text:
      name: "gas delivered raw"
```

### Anything in the logs that might be useful for us?

```txt
[20:20:44][D][uart_debug:158]: <<< "/XMX5XMXABCE000069683\r\n"
[20:20:44][D][uart_debug:158]: <<< "\r\n"
[20:20:44][D][uart_debug:158]: <<< "0-0:96.1.1(31333637343139322020202020202020)\r\n"
[20:20:44][D][uart_debug:158]: <<< "1-0:1.8.1(19938.691*kWh)\r\n"
[20:20:44][D][uart_debug:158]: <<< "1-0:1.8.2(24044.600*kWh)\r\n"
[20:20:44][D][uart_debug:158]: <<< "1-0:2.8.1(01115.278*kWh)\r\n"
[20:20:44][D][uart_debug:158]: <<< "1-0:2.8.2(02646.887*kWh)\r\n"
[20:20:44][D][uart_debug:158]: <<< "0-0:96.14.0(0002)\r\n"
[20:20:44][D][uart_debug:158]: <<< "1-0:1.7.0(0003.28*kW)\r\n"
[20:20:44][D][uart_debug:158]: <<< "1-0:2.7.0(0000.00*kW)\r\n"
[20:20:44][D][uart_debug:158]: <<< "0-0:96.13.1()\r\n"
[20:20:44][D][uart_debug:158]: <<< "0-0:96.13.0()\r\n"
[20:20:44][D][uart_debug:158]: <<< "0-1:96.1.0(3238303131303038333032323537303133)\r\n"
[20:20:44][D][uart_debug:158]: <<< "0-1:24.1.0(03)\r\n"
[20:20:44][D][uart_debug:158]: <<< "0-1:24.3.0(230505200000)(08)(60)(1)(0-1:24.2.0)(m3)\r\n"
[20:20:44][D][uart_debug:158]: <<< "(04392.240)\r\n"
[20:20:44][D][uart_debug:158]: <<< "!\r\n"
[20:20:44][D][sensor:109]: 'Energy Consumed Tariff 1': Sending state 19938.69141 kWh with 3 decimals of accuracy
[20:20:44][D][sensor:109]: 'Energy Consumed Tariff 2': Sending state 24044.59961 kWh with 3 decimals of accuracy
[20:20:44][D][sensor:109]: 'Energy Produced Tariff 1': Sending state 1115.27795 kWh with 3 decimals of accuracy
[20:20:44][D][sensor:109]: 'Energy Produced Tariff 2': Sending state 2646.88696 kWh with 3 decimals of accuracy
[20:20:44][D][sensor:109]: 'Power Consumed': Sending state 3.28000 kW with 3 decimals of accuracy
[20:20:44][D][sensor:109]: 'Power Produced': Sending state 0.00000 kW with 3 decimals of accuracy
[20:20:44][D][text_sensor:064]: 'dsmr_identification': Sending state 'XMX5XMXABCE000069683'
[20:20:44][D][text_sensor:064]: 'gas delivered raw': Sending state '(230505200000)(08)(60)(1)(0-1:24.2.0)(m3)(04392.240)'
```

AussiSG, 2023-05-06:
Seems to be an existing issue
regarding the following issue #2638

AussiSG, 2023-05-10:
Alright solved it with some earlier workaround found on github

See text sensor and extra template sensor below:

```yaml
substitutions:
  name: p1wifigateway

esphome:
  name: ${name}
  friendly_name: p1wifigateway
  platform: esp8266
  board: d1_mini
  esp8266_restore_from_flash: true 

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "oEOXmPj7exkRbcXWS7QSrsp3j9av0g508YKLxk14qo4="

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "P1Wifigateway Fallback Hotspot"
    password: ""

captive_portal:

dsmr:
  crc_check: false  
  gas_mbus_id: 1
  max_telegram_length: 1000
  receive_timeout: 500ms

uart:
  id: uart_bus
  baud_rate: 9600
  data_bits: 7
  parity: EVEN
  stop_bits: 1
  tx_pin:
    number: TX
  rx_pin: 
    number: RX
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

sensor:
  - platform: dsmr
    energy_delivered_tariff1:
      name: Energy Consumed Tariff 1
    energy_delivered_tariff2:
      name: Energy Consumed Tariff 2
    energy_returned_tariff1:
      name: Energy Produced Tariff 1
    energy_returned_tariff2:
      name: Energy Produced Tariff 2
    power_delivered:
      name: Power Consumed
    power_returned:
      name: Power Produced
    electricity_failures:
      name: Electricity failures
    electricity_long_failures:
      name: Long Electricity failures
    voltage_l1:
      name: Voltage Phase 1
    voltage_l2:
      name: Voltage Phase 2
    voltage_l3:
      name: Voltage Phase 3
    current_l1:
      name: Current Phase 1
    current_l2:
      name: Current Phase 2
    current_l3:
      name: Current Phase 3
    power_delivered_l1:
      name: Power Consumed Phase 1
    power_delivered_l2:
      name: Power Consumed Phase 2
    power_delivered_l3:
      name: Power Consumed Phase 3
    power_returned_l1:
      name: Power Produced Phase 1
    power_returned_l2:
      name: Power Produced Phase 2
    power_returned_l3:
      name: Power Produced Phase 3
    gas_delivered:
      name: Gas Consumed

  - platform: template
    # extra sensor to calculate the missing gas value
    name: Gas Consumed Calculated
    id: gas_consumed_calculated
    device_class: gas
    state_class: total_increasing
    accuracy_decimals: 3
    unit_of_measurement: "m³"

text_sensor:
  - platform: dsmr
    identification:
      name: "dsmr_identification"
    p1_version:
      name: "dsmr_p1_version"

    gas_delivered_text:
      name: "Gas Consumed Text"
      id: gas_consumed_text
      # if the state of "gas_delivered_text" changes a new value for the extra gas sensor is calculated
      on_value:
        then:
          - lambda: |-
              std::string str = id(gas_consumed_text).state;
              std::vector<std::string> v;
              char * token;
              char seps[] = "()";
              token = strtok (&str[0],seps);
              while (token != NULL)
              {
                v.push_back(token);
                token = strtok (NULL, seps);
              }
              int count = 0;
              for ( std::string s : v )
              {
                 if (count == 6) {
                   id(gas_consumed_calculated).publish_state(parse_number<float>(s.c_str()).value());
                 }   
                 count++;
              } 
```

closed 2023-05-10 by AussiSG

ph-wheels, 2023-05-17:
Thanks for pointing out this workaround, sorry for not finding/reading initial issue, have implemented it and awaiting it to show up in dashboard, it does provide the value in esp logs, this value has been added to dashboard. Many Thanks