esphome/issues #4493: The prompt box keeps showing Initializing status after clicking the New Device button

onecombo, 2023-05-06:
### The problem

When I'm about to add a new device to ESPhome, I click the new device button and the Initializing prompt box pops up and it gets stuck here without any further progress.

[image]

### Which version of ESPHome has the issue?

v2023.4.4 Documentation

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.5.2 Supervisor 2023.04.1 Operating System 10.1

### What platform are you using?

ESP32

### Anything in the logs that might be useful for us?

```txt
I'm sorry, I'm a rookie and I can't provide the log information, can you tell me how to find the log file or debug method to get the information.
```

David-Cuppan, 2023-05-13:
I now have the same issue. I am on the latest releases.

closed 2023-09-25 by github-actions[bot]

Brandon1212b, 2024-01-12:
same issue

jschwalbe, 2024-02-03:
same issue, running LXC on Proxmox