esphome/issues #4499: ESP32 Network error?

Lord-Memester, 2023-05-07:
### The problem

I'm using a NodeMCU-32S to control a servo that flips a light switch. I noticed yesterday that my fan just turned on without anyone doing anything, and it just happened again while I was in the same room. I guess the issue here is that the servo moves when a specific error occurs, and I'm not sure how to prevent the error or the issue it causes.

### Which version of ESPHome has the issue?

2023.4.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.2

### What platform are you using?

ESP32

### Board

NodeMCU-32S

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-XXXXXX
  friendly_name: ESPHome Web XXXXXX

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXXXXXXXXX"
  services:
    - service: control_servo
      variables:
        level: float
      then:
        - servo.write:
            id: my_servo
            level: !lambda 'return level / 100.0;'

ota:

output:
  - platform: ledc
    id: pwm_output
    pin: GPIO12
    frequency: 50 Hz

servo:
  - id: my_servo
    output: pwm_output

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-XXXXXX"
    password: "XXXXXXXXXXX"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
Logger: aioesphomeapi.connection
Source: runner.py:179
First occurred: 2:09:07 PM (1 occurrences)
Last logged: 2:09:07 PM

esphome-web-XXXXXX @ 192.168.X.XXX: Connection error occurred: [Errno 104] Connection reset by peer
```

### Additional information

The X's are me obscuring info.

RoboMagus, 2023-05-15:
Connection errors occur frequently in most of my devices due to sub-optimal WiFi reception. Even in best case scenarios we should assume network errors can occur. 

My guess is that the network error is not directly causing the behavior you're seeing, but perhaps it causes your ESP device to reboot. Have you checked the [uptime sensor](https://esphome.io/components/sensor/uptime.html) to see if your device reboots?

My advise would be to ensure no servo motion occurs on startup. I.e. the servo state persists between restarts.

Lord-Memester, 2023-05-16:
> Connection errors occur frequently in most of my devices due to sub-optimal WiFi reception. Even in best case scenarios we should assume network errors can occur.
>
> My guess is that the network [...]

I have not checked that, actually. The device has been unplugged for several days because of the issue, so I won't be able to get back to you until it happens again, and there's a few hours until I can get back to a place to test it in.

Lord-Memester, 2023-05-17:
> Connection errors occur frequently in most of my devices due to sub-optimal WiFi reception. Even in best case scenarios we should assume network errors can occur.
>
> My guess is that the network error [...]

It does indeed appear to restart.
[image]
How should I go about ensuring this doesn't happen upon a restart? The configuration for this device is janky enough already, and I'm not sure how I would go about preventing the issue. Could you possibly help, please?

RoboMagus, 2023-05-17:
> Could you possibly help, please?

I could give it a go, though as I've not played with Servos in ESPHome myself I'm not sure about its exact behavior.

Looking at your config, specifically the [servo](https://esphome.io/components/servo.html?highlight=servo#configuration-variables) portion, I found the `restore` option. Set this to _true_ such that upon restart the servo will go to its last known position. As you're using an ESP32, i believe restore from flash is enabled by default, so just setting this value should do the trick.

I'd also add `auto_detach_time: 5s` (Or anything reasonable related to the longest travel time of your servo motion). This way the servo doesn't remain powered after its motion completes. Probably not necessary, just my personal preference 😉

RoboMagus, 2023-05-17:
Like this:
```yaml
...

servo:
  - id: my_servo
    output: pwm_output
    restore: true
    auto_detach_time: 5s

...
```

Lord-Memester, 2023-05-17:
> Like this:
> ```yaml
> ...
> 
> servo:
>   - id: my_servo
>     output: pwm_output
>     restore: true
>     auto_detach_time: 5s
> 
> ...
> ```

Thank you!

Lord-Memester, 2023-05-18:
> Like this:
> ```yaml
> ...
> 
> servo:
>   - id: my_servo
>     output: pwm_output
>     restore: true
>     auto_detach_time: 10s
> 
> ...
> ```

It seems like that did the trick! It's restarted several times, and has behaved exactly as I desired since the change!

closed 2023-05-18 by Lord-Memester