esphome/issues #4500: 'Internal Temperature' sensor crashing on S2

sbrk-r, 2023-05-08:
### The problem

I added internal temperature sensor to one of my S2 mini boards, and after 5 minutes to an hour it crashes. (it does reboot and reconnect, and will crash again after the same amount of time). 

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.4.1

### What platform are you using?

ESP32-IDF

### Board

lolin_s2_mini

### Component causing the issue

sensor/internal_temperature

### Example YAML snippet

```yaml
esphome:
  name: $devicename
  build_path: ./build/${devicename}

esp32:
  board: $board
  framework:
    type: esp-idf
    version: recommended

wifi:
  ssid: !secret wifi_bk_ssid
  password: !secret wifi_bk_pass
  fast_connect: true
  ap:
    ssid: "esphome_${devicename}"
    password: !secret fallback_wifi_pass

logger:
  hardware_uart: USB_CDC
  
mqtt:
  broker: ${mqtt_broker_ip}
  username: ${mqtt_user}
  password: !secret mqtt_pass
  log_topic:
    topic: ${devicename}/logs
    level: INFO

ota:
  safe_mode: false
  password: !secret ota_pass

text_sensor:
  - platform: version
    name: "ESPHome Version"
    id: ${dname}_fw_ver

  - platform: wifi_info
    ip_address:
      name: ${dname}_ipaddr
    mac_address:
      name: ${dname}_hwaddr

light:
  - platform: status_led
    name: ${dname}_led_s
    pin:
      number: GPIO15

sensor:
  - platform: wifi_signal
    name: ${dname}_wifi_rssi
    update_interval: 10s
    entity_category: "diagnostic"

  - platform: internal_temperature
    name: ${dname}_tcore
    update_interval: 10s  

  - platform: uptime
    name: "${dname}_uptime"
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
[00:43:41][D][esp-idf:000]: I (356187) tsens: Config temperature range [-10°C ~ 80°C], error < 1°C
[00:43:41][D][sensor:127]: 'dev0_tcore': Sending state 46.49100 °C with 1 decimals of accuracy
[00:43:44][D][sensor:127]: 'dev0_uptime': Sending state 359.47501 s with 0 decimals of accuracy
[00:43:49][D][sensor:127]: 'dev0_wifi_rssi': Sending state -63.00000 dBm with 0 decimals of accuracy
[00:43:51][D][esp-idf:000]: I (366188) tsens: Config temperature range [-10°C ~ 80°C], error < 1°C
[00:43:56]
[00:43:56]abort() was called at PC 0x400a2879 on core 0
[00:43:56]
[00:43:56]
[00:43:56]Backtrace: 0x40023e2e:0x3ffc5990 0x40029f25:0x3ffc59b0 0x4003034a:0x3ffc59d0 0x400a2879:0x3ffc5a40 0x40024432:0x3ffc5a60 0x4009ed0c:0x3ffd4bd0 0x4009ed8c:0x3ffd4c00 0x40083e32:0x3ffd4c50 0x40083ea8:0x3ffd4c90 0x4011406d:0x3ffd4cb0 0x40092c13:0x3ffd4cd0 0x40090d95:0x3ffd4d20 0x40092d6a:0x3ffd4d50 0x40083566:0x3ffd4d70
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40023e2e: panic_abort at /home/user/.platformio/packages/framework-espidf/components/esp_system/panic.c:408
WARNING Decoded 0x40029f25: esp_system_abort at /home/user/.platformio/packages/framework-espidf/components/esp_system/esp_system.c:137
WARNING Decoded 0x4003034a: abort at /home/user/.platformio/packages/framework-espidf/components/newlib/abort.c:46
WARNING Decoded 0x400a2879: task_wdt_isr at /home/user/.platformio/packages/framework-espidf/components/esp_system/task_wdt.c:176 (discriminator 3)
WARNING Decoded 0x40024432: _xt_lowint1 at xtensa_vectors.o
WARNING Decoded 0x4009ed0c: temp_sensor_read_raw at /home/user/.platformio/packages/framework-espidf/components/driver/esp32s2/rtc_tempsensor.c:141 (discriminator 2)
WARNING Decoded 0x4009ed8c: temp_sensor_read_celsius at /home/user/.platformio/packages/framework-espidf/components/driver/esp32s2/rtc_tempsensor.c:178
WARNING Decoded 0x40083e32: esphome::internal_temperature::InternalTemperatureSensor::update() at /home/user/dev/esphome/32/build/s2-6a92/src/esphome/components/internal_temperature/internal_temperature.cpp:36
WARNING Decoded 0x40083ea8: non-virtual thunk to esphome::internal_temperature::InternalTemperatureSensor::update()
WARNING Decoded 0x4011406d: std::_Function_handler<void (), esphome::PollingComponent::call_setup()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/user/dev/esphome/32/build/s2-6a92/src/esphome/core/component.cpp:192 (inlined by) _M_invoke at /home/user/.platformio/packages/toolchain-xtensa-esp32s2@8.4.0+2021r2-patch5/xtensa-esp32s2-elf/include/c++/8.4.0/bits/std_function.h:297
WARNING Decoded 0x40092c13: std::function<void ()>::operator()() const at /home/user/.platformio/packages/toolchain-xtensa-esp32s2@8.4.0+2021r2-patch5/xtensa-esp32s2-elf/include/c++/8.4.0/bits/std_function.h:687 (inlined by) esphome::Scheduler::call() at /home/user/dev/esphome/32/build/s2-6a92/src/esphome/core/scheduler.cpp:233
WARNING Decoded 0x40090d95: esphome::Application::loop() at /home/user/dev/esphome/32/build/s2-6a92/src/esphome/core/application.cpp:69
WARNING Decoded 0x40092d6a: loop() at /home/user/dev/esphome/32/build/s2-6a92/src/main.cpp:418
WARNING Decoded 0x40083566: esphome::loop_task(void*) at /home/user/dev/esphome/32/build/s2-6a92/src/esphome/components/esp32/core.cpp:74 (discriminator 1)
ERROR Serial port closed!
```

sbrk-r, 2023-05-09:
Upgraded to 2023.4.4 as I noticed I was a little out of date, and the problem persists.

Fabian-Schmidt, 2023-05-18:
Is this related to this ESP-IDF issue: https://github.com/espressif/esp-idf/issues/8088

closed 2023-09-30 by github-actions[bot]