esphome/issues #4501: [FastLED] M5 AtomS3 lite / error: static assertion failed: Invalid pin specified

lboue, 2023-05-08:
### The problem

# [FastLED] M5 AtomS3 lite / error: static assertion failed: Invalid pin specified
### Summary
Steps to reproduce
RGB LED is connected to GPIO35 but I can't build the file with that config:

### Config snippet 
```
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: 35
    num_leds: 1
    rgb_order: GRB
    id: status_led
    name: AtomS3 lite Light
    effects:
      - random:
      - flicker:
      - addressable_rainbow:
```
### Pinout

Doc : [Schematic](https://docs.m5stack.com/en/core/AtomS3%20Lite)

### Error
```
/data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/fastpin.h: In instantiation of 'class FastPin<35>':
/data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/platforms/esp/32/clockless_rmt_esp32.h:178:23:   required from 'class ClocklessController<35, 60, 150, 90, (EOrder)66, 0, false, 5>'
/data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/chipsets.h:582:7:   required from 'class WS2812Controller800Khz<35, (EOrder)66>'
/data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/FastLED.h:105:52:   required from 'class WS2812B<35, (EOrder)66>'
src/esphome/components/fastled_base/fastled_light.h:150:41:   required from 'CLEDController& esphome::fastled_base::FastLEDLightOutput::add_leds(int) [with CHIPSET = WS2812B; unsigned char DATA_PIN = 35; EOrder RGB_ORDER = (EOrder)66]'
src/main.cpp:217:64:   required from here
/data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/fastpin.h:207:24: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
                ~~~~~~~~^~
In file included from /data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/platforms.h:36,
                 from /data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/FastLED.h:52,
                 from src/esphome/components/fastled_base/fastled_light.h:16,
                 from src/esphome.h:22,
                 from src/main.cpp:3:
/data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/platforms/esp/32/clockless_rmt_esp32.h: In instantiation of 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::startOnChannel(int) [with int DATA_PIN = 35; int T1 = 60; int T2 = 150; int T3 = 90; EOrder RGB_ORDER = (EOrder)66; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 5]':
/data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/platforms/esp/32/clockless_rmt_esp32.h:381:26:   required from 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::startNext(int) [with int DATA_PIN = 35; int T1 = 60; int T2 = 150; int T3 = 90; EOrder RGB_ORDER = (EOrder)66; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 5]'
/data/m5-atoms3-c846d4/.piolibdeps/m5-atoms3-c846d4/FastLED/platforms/esp/32/clockless_rmt_esp32.h:312:26:   required from 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showPixels(PixelController<RGB_ORDER>&) [with int DATA_PIN = 35; int T1 = 60; int T2 = 150; int T3 = 90; EOrder RGB_ORDER = (EOrder)66; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 5]'
```

### Which version of ESPHome has the issue?

2023.4.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.2

### What platform are you using?

ESP32

### Board

esp32-s3-devkitc-1

### Component causing the issue

fastled_clockless

### Example YAML snippet

```yaml
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: dev

light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: 35 # LED35
    num_leds: 1
    rgb_order: GRB
    id: status_led
    name: AtomS3 lite Light
```

### Additional information

[image]

Unfocused, 2023-05-10:
ESPHome curently uses an outdated version of FastLED, which doesn't properly support ESP32-S3. One of the issues is it disallows the wrong pins (I was affected by this too).

This PR updates FastLED to a version that supports ESP32-S3:
https://github.com/esphome/esphome/pull/4257

If you want to use/test it now, it's usable via:
```yaml
external_components:
  - source: github://pr#4257
    components: [fastled]
```

Note that it also merges `fastled_clockless` and `fastled_spi`, so you just use `fastled` as the platform.

lboue, 2023-05-16:
Thans @Unfocused.
It doesn't work in my case with a WS2812C-2020 LED with PR 4257 witch updates FastLED to a version that supports ESP32-S3
```
external_components:
  - source: github://pr#4257
    components: [fastled]

# RGB	WS2812C-2020 LED
light:
  - platform: fastled #fastled_clockless 
    chipset: WS2812B
    pin: 35 # LED35
    num_leds: 1
    rgb_order: GRB
    id: status_led
    name: AtomS3 lite Light
```

But neopixelbus worksfine:

```
# RGB	WS2812C-2020 LED
light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812X
    pin: 35
    num_leds: 1
    name: "NeoPixel Light"

```

lboue, 2023-05-19:
I found a second workaround:

[ESP32 RMT LED Strip](https://esphome.io/components/light/esp32_rmt_led_strip.html?highlight=rmt#esp32-rmt-led-strip): This is a component using the ESP32 RMT peripheral to drive most addressable LED strips.
```
light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    pin: 35
    num_leds: 1
    rmt_channel: 0
    chipset: ws2812
    id: status_led
    name: AtomS3 lite Light
```

lboue, 2023-05-30:
@Unfocused Do you think yesterday [FastLED 3.6.0 release](https://github.com/FastLED/FastLED/releases/tag/3.6.0) could fix this issue?

Unfocused, 2023-05-31:
Not my area of expertise, unfortunately. But reading through the FastLED 3.6.0 release notes ([announcement](https://www.reddit.com/r/FastLED/comments/13v33l9/fastled_360_release/) and the [commit log](https://github.com/FastLED/FastLED/compare/3.5.0...3.6.0)), it seems that release does include improved support for ESP32 variants. But skimming through the commits, I didn't see any specific fixes for ESP32-S3. 

So: Maybe, maybe not.

Note that updating https://github.com/esphome/esphome/pull/4257 to use the new FastLED release may or may not be easy. And using a new major update may be considered too risky (the new release won't have much testing yet, so there could be unknown new bugs). So even if the new FastLED release does fix this issue, you might have to wait for another PR to get it.

Unfortunately, ESP32-S3 is just too new of a chipset, and it's quite different from ESP32 in a lot of ways (it's not just a small update). It hasn't had enough time yet to get mature support in libraries, and it's different enough from ESP32 that it requires a lot of code changes to support it. Getting mature support is just going to take time - not just for code changes, but also people need more time to gain experience with it. Same thing happened with ESP32 and ESP8266 before it (and any new chipset).

FWIW, I'm using ESP32-S3 in a few devices myself and I'm constantly finding issues - so you're not alone there. 

If you haven't already, it'd be worth seeing if there's any similar issues posted in the [FastLED issue tracker](https://github.com/FastLED/FastLED/issues). If there's no relevant existing issue, and the discussion in https://github.com/esphome/esphome/pull/4257 finds out that the issue isn't in either ESPHome or in that PR, then it'd be worth posting a new issue to the FastLED issue tracker.

lboue, 2023-05-31:
@Unfocused 
Thank you for the details. I will continue to use the 2 workarounds possible with the neopixelbus and esp32_rmt_led_strip platforms.

closed 2023-10-15 by github-actions[bot]