esphome/issues #4502: Sprinkler Controller is Marking Valve Complete before it's actually complete (because of valve open delay)

olympia, 2023-05-08:
### The problem

It looks like Sprinkler Controller is marking a valve completed too early if valve open delay is set. 

In the attached log file it is visible that both "Marking valve 0 complete" and "Marking valve 1 complete" are kicking in 10s earlier (10s is the valve open delay what has been set) than the actual end time of the zone ('Relay 1' Turning OFF. and 'Relay 2' Turning OFF. respectively).

@kbx81 Is this something you could have a look?

### Which version of ESPHome has the issue?

2023.4.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.05

### What platform are you using?

ESP8266

### Board

sonoff 4ch pro

### Component causing the issue

Sprinkler

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/irrigation.yaml...
WARNING 'sonoff4chpro_sprinkler': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
WARNING ESP8266: Pin 9 (9-10) might already be used by the flash interface in QUAD IO flash mode.
WARNING ESP8266: Pin 10 (9-10) might already be used by the flash interface in QUAD IO flash mode.
INFO Starting log output from sonoff4chpro_sprinkler.localdomain using esphome API
INFO Successfully connected to sonoff4chpro_sprinkler.localdomain
[13:45:19][I][app:102]: ESPHome version 2023.4.4 compiled on May  8 2023, 13:36:26
[13:45:19][C][wifi:505]: WiFi:
[13:45:19][C][wifi:363]:   Local MAC: 4C:EB:D6:CA:82:52
[13:45:19][C][wifi:364]:   SSID: 'OlympIoTwoI'[redacted]
[13:45:19][C][wifi:365]:   IP Address: 192.168.33.118
[13:45:19][C][wifi:366]:   BSSID: B6:8B:A9:1D:E3:02[redacted]
[13:45:19][C][wifi:368]:   Hostname: 'sonoff4chpro_sprinkler'
[13:45:19][C][wifi:370]:   Signal strength: -69 dB ▂▄▆█
[13:45:19][C][wifi:374]:   Channel: 1
[13:45:19][C][wifi:375]:   Subnet: 255.255.255.0
[13:45:19][C][wifi:376]:   Gateway: 192.168.33.254
[13:45:19][C][wifi:377]:   DNS1: 8.8.8.8
[13:45:19][C][wifi:378]:   DNS2: 0.0.0.0
[13:45:19][C][logger:294]: Logger:
[13:45:19][C][logger:295]:   Level: DEBUG
[13:45:19][C][logger:296]:   Log Baud Rate: 115200
[13:45:19][C][logger:297]:   Hardware UART: UART0
[13:45:19][C][esp8266_pwm:022]: ESP8266 PWM:
[13:45:19][C][esp8266_pwm:023]:   Pin: GPIO13
[13:45:19][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[13:45:19][C][esp8266_pwm:025]:   Inverted: YES
[13:45:19][C][uptime.sensor:031]: Uptime Sensor 'Uptime'
[13:45:19][C][uptime.sensor:031]:   Device Class: 'duration'
[13:45:19][C][uptime.sensor:031]:   State Class: 'total_increasing'
[13:45:19][C][uptime.sensor:031]:   Unit of Measurement: 's'
[13:45:19][C][uptime.sensor:031]:   Accuracy Decimals: 0
[13:45:19][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[13:45:19][C][template.sensor:023]: Template Sensor 'Sprinkler Zone 1 Progress Percentage'
[13:45:19][C][template.sensor:023]:   State Class: ''
[... 441 lines cut ...]
[13:46:43][D][sensor:109]: 'Sprinkler Zone 2 Progress Percentage': Sending state 96.00000  with 1 decimals of accuracy
[13:46:43][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '-1m -9s'
[13:46:43][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 215.00000  with 1 decimals of accuracy
[13:46:44][D][text_sensor:064]: 'Sprinkler Zone 2 Time Remaining': Sending state '0s'
[13:46:44][D][sensor:109]: 'Sprinkler Zone 2 Progress Percentage': Sending state 100.00000  with 1 decimals of accuracy
[13:46:44][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '-1m -10s'
[13:46:44][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 216.00000  with 1 decimals of accuracy
[13:46:44][D][switch:016]: 'Relay 2' Turning OFF.
[13:46:44][D][switch:055]: 'Relay 2': Sending state OFF
[13:46:44][D][switch:055]: 'Sprinkler Zone 2': Sending state OFF
[13:46:44][D][text_sensor:064]: 'Sprinkler Controller Status': Sending state 'Idle'
[13:46:44][D][text_sensor:064]: 'Sprinkler Zone 2 Status': Sending state 'Idle'
[13:46:44][D][text_sensor:064]: 'Sprinkler Zone 2 Time Remaining': Sending state '--'
[13:46:45][D][text_sensor:064]: 'Sprinkler Zone 2 Time Remaining': Sending state '0s'
[13:46:45][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '-1m -10s'
[13:46:45][D][sprinkler:1388]: All valves stopped, including pumps
[13:46:45][D][switch:055]: 'Sprinkler Start/Resume/Stop': Sending state OFF
[13:46:45][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '--'
[13:46:46][D][sensor:109]: 'Sprinkler Controller Status Internal': Sending state 0.00000  with 1 decimals of accuracy
[13:46:46][D][sensor:109]: 'Sprinkler Zone 2 Progress Percentage': Sending state 0.00000  with 1 decimals of accuracy
[13:46:46][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 0.00000  with 1 decimals of accuracy
[13:46:46][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 0.00000  with 1 decimals of accuracy
[13:47:06][D][sensor:109]: 'WiFi Signal': Sending state -66.00000 dBm with 0 decimals of accuracy
[13:47:06][D][sensor:109]: 'Uptime': Sending state 620.40503 s with 0 decimals of accuracy
```

olympia, 2023-05-08:
From switching perspective you are completely right -- it is switching when expected.
However, it looks like the active valve (`active_valve()`)within the cycle becomes the next as from this "marking point".

Please have a look at the below template sensors, where I use `if(id(sprinkler_ctrl).active_valve() == 0)` and `if(id(sprinkler_ctrl).active_valve() == 1)` conditions to decide into which sensor `time_remaining_active_valve()` should populate the remaining time.

If you are now looking at the esphome log I shared initially, you will see that the as from the marking point (10s earlier than the end of zone) the last 10 seconds of the remaining time gets populated into Zone 2 instead of Zone 1.
If I don't set any valve open delay, then all works fine and as expected.

I hope this makes sense to you... does it? :)

```
# Zone 1 - Define Progress percent as a sensor and publish time remaining to its time remaining text sensor
- platform: template
  id: zone1_progress_percentage
  name: "Sprinkler Zone 1 Progress Percentage"
  update_interval: $update_freq
  filters:
    - delta: 1
  lambda: |-
    if(id(sprinkler_ctrl).active_valve() == 0) {
      int seconds = round(id(sprinkler_ctrl).time_remaining_active_valve().value_or(0));
      int days = seconds / (24 * 3600);
      seconds = seconds % (24 * 3600);
      int hours = seconds / 3600;
      seconds = seconds % 3600;
      int minutes = seconds /  60;
      seconds = seconds % 60;
      id(zone_1_time_remaining).publish_state((
        (days ? String(days) + "d " : "") +
        (hours ? String(hours) + "h " : "") +
        (minutes ? String(minutes) + "m " : "") +
        (String(seconds) + "s")
        ).c_str());
      return ((id(sprinkler_ctrl).valve_run_duration_adjusted(0) - id(sprinkler_ctrl).time_remaining_active_valve().value_or(0)) * 100 / id(sprinkler_ctrl).valve_run_duration_adjusted(0));
    }
    if (id(sprinkler_ctrl).active_valve().has_value() == false && id(sprinkler_ctrl_status).state == "Idle") {
      return 0;
    }
    else if(id(zone1_progress_percentage).state > 1 && id(sprinkler_ctrl_status).state != "Paused")
    {
      return 100;
    }
      return {};

# Zone 2 - Define Progress percent as a sensor and publish time remaining to its time remaining text sensor
- platform: template
  id: zone2_progress_percentage
  name: "Sprinkler Zone 2 Progress Percentage"
  update_interval: $update_freq
  filters:
    - delta: 1
  lambda: |-
    if(id(sprinkler_ctrl).active_valve() == 1) {
      int seconds = round(id(sprinkler_ctrl).time_remaining_active_valve().value_or(0));
      int days = seconds / (24 * 3600);
      seconds = seconds % (24 * 3600);
      int hours = seconds / 3600;
      seconds = seconds % 3600;
      int minutes = seconds /  60;
      seconds = seconds % 60;
      id(zone_2_time_remaining).publish_state((
        (days ? String(days) + "d " : "") +
        (hours ? String(hours) + "h " : "") +
        (minutes ? String(minutes) + "m " : "") +
        (String(seconds) + "s")
        ).c_str());
      return ((id(sprinkler_ctrl).valve_run_duration_adjusted(1) - id(sprinkler_ctrl).time_remaining_active_valve().value_or(1)) * 100 / id(sprinkler_ctrl).valve_run_duration_adjusted(1));
    }
    if (id(sprinkler_ctrl).active_valve().has_value() == false && id(sprinkler_ctrl_status).state == "Idle")
    {
      return 0;
    }
    else if(id(zone2_progress_percentage).state > 1 && id(sprinkler_ctrl_status).state != "Paused")
    {
      return 100;
    }
      return {};
```

kbx81, 2023-05-08:
This makes perfect sense and, while I realize it's not quite as expected, there is a very specific interval during which this condition exists and it doesn't affect the behavior of the controller itself. The reason the behavior occurs is due to the architecture of the state machine and delay interval handling -- to correct this, another set of timers and logic would be required to handle this specific situation. Call it "being lazy"...but...again...I'm curious if this is causing a specific issue for you. It appears to me that you've set up a bunch of sensors to analyze/display some of the controllers internal behavior and, while I understand what you're doing, the fix for this will be a bit involved to implement.

That said, there was some discussion around implementing an additional feature set in a future version to permit things like a maximum zone/valve run time which would likely require some re-vamping of controller's internal scheduling. If we end up proceeding down this path, this behavior will likely be corrected at that time, anyway. Any thoughts?

olympia, 2023-05-08:
Indeed, I was/am about to develop a comprehensive sprinkler dashboard what is nicely displaying - as you name it - the controllers' internal behavior, providing the users a detailed view on an eye catching way.

To answer your question, this is certainly not causing any specific functional issue and I do also understand what you mean by small gain for huge efforts. I actually thought it's something easy to fix, but I obviously I was ignorantly wrong... :)

I will live with this for now and looking forward to the mentioned re-vamping whenever it happens -- many thanks for your attention!

olympia, 2023-05-08:
Can `time_remaining_current_operation()` also somewhat defective due to the same reason when valve open delay is being used?

In the below example, I am populating "Sprinkler Controller Overall Time Remaining" by `time_remaining_current_operation()` and please see that it is also misbehaving just right after the marker point.

```
[21:23:55][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '42s'
[21:23:55][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 30.00000  with 1 decimals of accuracy
[21:23:56][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '11s'
[21:23:56][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 63.00000  with 1 decimals of accuracy
[21:23:56][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '41s'
[21:23:56][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 31.00000  with 1 decimals of accuracy
[21:23:57][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '10s'
[21:23:57][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 66.00000  with 1 decimals of accuracy
[21:23:57][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '40s'
[21:23:57][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 33.00000  with 1 decimals of accuracy
[21:23:57][D][sprinkler:1242]: Marking valve 0 complete
[21:23:58][D][sensor:109]: 'Sprinkler Controller Status Internal': Sending state 1.00000  with 1 decimals of accuracy
[21:23:58][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '9s'
[21:23:58][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 70.00000  with 1 decimals of accuracy
[21:23:58][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '9s'
[21:23:58][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 85.00000  with 1 decimals of accuracy
[21:23:59][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '8s'
[21:23:59][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 73.00000  with 1 decimals of accuracy
[21:23:59][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '8s'
[21:23:59][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 86.00000  with 1 decimals of accuracy
[21:24:00][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '7s'
[21:24:00][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 76.00000  with 1 decimals of accuracy
[21:24:00][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '7s'
[21:24:00][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 88.00000  with 1 decimals of accuracy
[21:24:01][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '6s'
[21:24:01][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 80.00000  with 1 decimals of accuracy
[21:24:01][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '6s'
[21:24:01][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 90.00000  with 1 decimals of accuracy
[21:24:02][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '5s'
[21:24:02][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 83.00000  with 1 decimals of accuracy
[21:24:02][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '5s'
[21:24:02][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 91.00000  with 1 decimals of accuracy
[21:24:03][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '4s'
[21:24:03][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 86.00000  with 1 decimals of accuracy
[21:24:03][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '4s'
[21:24:03][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 93.00000  with 1 decimals of accuracy
[21:24:04][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '3s'
[21:24:04][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 90.00000  with 1 decimals of accuracy
[21:24:04][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '3s'
[21:24:04][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 95.00000  with 1 decimals of accuracy
[21:24:05][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '2s'
[21:24:05][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 93.00000  with 1 decimals of accuracy
[21:24:05][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '2s'
[21:24:05][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 96.00000  with 1 decimals of accuracy
[21:24:06][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '1s'
[21:24:06][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 96.00000  with 1 decimals of accuracy
[21:24:06][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '1s'
[... cut ...]
[21:24:17][D][text_sensor:064]: 'Sprinkler Zone 2 Status': Sending state 'Running'
[21:24:17][D][text_sensor:064]: 'Sprinkler Zone 2 Time Remaining': Sending state '29s'
[21:24:17][D][sensor:109]: 'Sprinkler Zone 2 Progress Percentage': Sending state 3.00000  with 1 decimals of accuracy
[21:24:18][D][sensor:109]: 'Sprinkler Controller Status Internal': Sending state 2.00000  with 1 decimals of accuracy
[21:24:18][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '29s'
```

kbx81, 2023-05-08:
Always happy to help. 😄 Don't mean to push back much -- I actually use and run into this myself, hence why I was like, "yup." 😆 Unfortunately yes, it did get written off as "too much code for too little gain" which I will absolutely apologize for. 😇 Regardless, good catch and please let me/us know if you run into any other issues. Before all too terribly long, we'll iterate again to bring about even more fancy things... 😉

kbx81, 2023-05-08:
> Can `time_remaining_current_operation()` also somewhat defective due to the same reason when valve open delay is being used?

It's possible but I don't think so...but maybe. I don't recall this being an issue but I'll take a closer look.

olympia, 2023-05-09:
I didn't feel any offense at all for what you would need to apologize for. I really know what does that mean, especially a case when you need to write code for something you don't use, don't feel the need or don't see the added value. Luckily - for me - in this case you seems to be at least a tiny bit impacted... :). 

Re `time_remaining_current_operation()`: unfortunately it seems to be impacted by this issue as well as `total_cycle_time_enabled_incomplete_valves()`. The outputs from both methods are getting out of order as soon as the "marking complete" point is reached. They are working perfectly fine without valve open delay.

Re other issues, please let me know where/ how to communicate them the most efficiently. I ran into a couple of smaller ones what needed some workarounds. Shall I open a ticket one-by-one on all those here?

kbx81, 2023-05-09:
Can you briefly summarize what other issues you are running into? ...From there we can decide if it warrants opening another issue on GH or if it's a simple "just do this" type of thing. No need to paste lengthy code snips for demonstrating the issue, just a one-line or so summary is fine. I'm just curious what else you are running into.

olympia, 2023-05-09:
Rather small stuff, nothing very crucial: 

1. I am not so sure if this is intentional, but it looks like it is not possible to use any cycle functions on a single valve controller. It looks like it is not even possible adding a main_switch to a single valve controller. As a consequence of all that, multiplier cannot be added to such controllers either.
2. Paused state is not reflected in controller_state() method
3. open valve delay times are not getting added to the status methods (e.g. total_cycle_time_all_valves() | time_remaining_current_operation() | total_cycle_time_enabled_valves() | total_cycle_time_enabled_incomplete_valves()) - this is related, but not the same issue (I think) what we discussed above
4. I couldn't find a way to use state_as_str_() method. I think this might be somewhat internal method what is not accessible?
5. Same goes for next_valve_number_(). I wanted to use this as well, but couldn't figure.

I appreciate 4 and 5 might be possible, it is just that I am lacking the skills using them?!

kbx81, 2023-05-09:
Cool -- I can respond to those things. 😄

1. With a single valve, a main switch would be completely redundant because there is only one valve; in other words, the main switch would do the exact same things as the valve switch, so there's simply no need for both. Having both would just tie up additional RAM on the ESP while providing no additional gain. As for things like multiplier, there's no reason you shouldn't still be able to use that.
2. The `controller_state()` simply returns the state of the controller's internal state machine; "paused" is not a state handled by this mechanism so it won't be reflected this way. If you want to know if the controller is paused, take a look at [this section](https://esphome.io/components/sprinkler.html#how-do-i) in the documentation -- specifically, the `paused_valve()` method can reveal this.
3. This might be an oversight, I can take a look.
4. and 5. Any method with a name that ends in an underscore (`_`) is a protected method; it's for internal use by the object. You can't access these methods from outside of the class. It probably would make sense to expose the next/previous valve methods, but `state_as_str_()` is there for debugging purposes only (it's not useful to those who don't want or don't speak English). I'm also weary about exposing much of the state machine as I don't want people building a bunch of dependencies on that functionality. Future updates might bring about a different implementation for scheduling which behaves differently and then we end up with a lot of unhappy people. 😇

patfelst, 2023-05-10:
Hi I'm jumping on this issue as I've just started using this integration - it's fantastic so far!

I have noticed that the function `.total_queue_time()` doesn't seem to give the correct time, it seems to miss the first valve in the queue, whereas `.time_remaining_current_operation().value()` gives the correct time.

patfelst, 2023-05-10:
> 1. A text sensor filter for text template sensors to store the last value passed through this filter and only passes incoming values through if incoming value is different from the previously passed one (similarly to delta sensor for numeric sensors, but for text/ strings). Using this one could avoid to publish text sensor state on each "update_interval" - would be very handy for example for configuring text template sensors for human readable formatted remaining time.

I also wanted to avoid having repeated "Finished" or "0s" being continuously published when using `update_interval` to publish the remaining time via a text sensor. The solution I found is to use  a script, and put a `while time_remaining_current_operation().has_value() == true`, like this:

```yaml
script:
  - id: publish_sprinkler_time
    then:
      - while:
          condition:
            lambda: 'return id(sprinkler_controller).time_remaining_current_operation().has_value();'
          then:
            - text_sensor.template.publish:
                id: text_time_remaining
                state: !lambda |-
                  return seconds_to_readable(id(sprinkler_controller).time_remaining_current_operation().value());
            - delay: 5s
      - text_sensor.template.publish:
          id: text_time_remaining
          state: "Finished"
```
and you call the script where you start your sprinkler queue or turn on the master switch or whatever, for me it's from another script where I create my sprinkler queue like so:
```yaml
script:
  #
  # Program A - GARDENS
  #
  - id: watering_program_a
    then:
      - sprinkler.clear_queued_valves:
          id: sprinkler_controller
      - sprinkler.queue_valve:
          id: sprinkler_controller
          valve_number: 3 # Zone 4 BACK GARDEN
          run_duration: 0s
      - sprinkler.queue_valve:
          id: sprinkler_controller
          valve_number: 4 # Zone 5 POOL GARDEN
          run_duration: 0s
      - sprinkler.queue_valve:
          id: sprinkler_controller
          valve_number: 5 # Zone 6 FRONT GARDEN
          run_duration: 0s
      - sprinkler.start_from_queue:
          id: sprinkler_controller
      - script.execute: publish_queue_time
      - script.execute: publish_sprinkler_time
```
@kbx81 Keith please note, those last two lines give me different results as per my previous post where the queue time looks like this:
```yaml
  - id: publish_queue_time
    then:
      - text_sensor.template.publish:
          id: text_total_queue_time
          state: !lambda |-
            return seconds_to_readable(id(sprinkler_controller).total_queue_time());
```

Also the documentation says the run_duration default will be whatever is set for the valve, except the yaml validation fails if I delete those lines with `run_duration: 0s`.

kbx81, 2023-05-11:
> Many thanks for the quick feedback!
> 
> May I have some additional inputs/ comments on them:
> 
> 1. I am not actually missing the main switch, but I thought (probably incorrectly) that the cycle functions are only working if the main switch exists. The main issue here is that using the below config for a single valve drip controller I don't even get the multiplier_number exposed to HA

Hmm, this is interesting -- I'll look into it. That definitely doesn't make any sense, as you should still be able to use the multiplier in this case. 😇

> I am also unable to start the drip controller, by using the API service I configured to `sprinkler.start_full_cycle` on the `drip_ctrl`. As soon as I add a second dummy/ unused valve with `main_switch`, `auto_advance_switch` and an `enable_switch`, the multiplier_number immediately gets exposed and the API service and the controller start functioning as expected.

I'm curious about your motivation for using `sprinkler.start_full_cycle` with a single valve. Wouldn't you just refer directly to the single valve and use `sprinkler.start_single_valve`, instead? There should be no behavioral difference, although I do agree that `sprinkler.start_full_cycle` should still work in this scenario -- another thing I'll look into. 😉

> 2. OK, I just added this, because you were asking where I needed workarounds. Besides the paused state, I am also missing a pending state here (when the controller is waiting for example during valve open times). In my workaround, I added a fuzzy logic (what I don't find super-robust) to publish states between switches and sensors back and forth to achieve to have a full comprehensive controller status sensor.

The controller's state machines only track/use the states you see [here](https://esphome.io/api/namespaceesphome_1_1sprinkler.html#a091ab4ef6a49183d6acc39fb2d249a15). If you want more specific detail, you need to code that up yourself. The state machine is intended to support the core functionality of the controller, not enumerate every possible state anyone anywhere might be interested in. 😇

Personal opinion/side note: I'm really _not_ interested in creating states for every possible controller mode/behavior. For example, do you consider "paused" with auto-advance enabled a different state than "paused" when auto-advance is disabled? This is a gray area I really do _not_ want to get into, so (imho) what states the state machine tracks & reports should remain limited to the bare essentials it needs to operate. If you are interested in tracking other states/behavior, there are already mechanisms available to allow you to do so.

> 3. Thank you very much in advance!
> 4. and 5. OK, clear, thank you for the explanation.

You're welcome -- hope this helps! 🍻

> Then I forgot to mention two non-sprinkler related feature one for ESPHome and one for HA what I am badly missing and would be super-useful to have in a sprinkler config environment. Not sure if I should mention these here, but let me try, while I am having your attention:
> 
> 1. A text sensor filter for text template sensors to store the last value passed through this filter and only passes incoming values through if incoming value is different from the previously passed one (similarly to delta sensor for numeric sensors, but for text/ strings). Using this one could avoid to publish text sensor state on each "update_interval" - would be very handy for example for configuring text template sensors for human readable formatted remaining time.
> 2. It would be nice to have an option in HA to set the format of time displayed for sensors with duration device_class. Currently this is displayed with DD:HH:MM:SS and would be great to see them more human readable, i.e. like 1d 1h 23m 47s. This way, it would be even more easier to have the time remaining display in a nice way. Looking at the HA forums, most users prefer this for sprinklers opposed to DD:HH:MM:SS.

For these, you should [ask on Discord](https://discord.gg/KhAMKrd) and/or open new [feature requests](https://github.com/esphome/feature-requests/issues) -- but be sure to carefully check the docs so you're not being redundant! 😉

kbx81, 2023-05-11:
# Issue Summary

- `time_remaining_current_operation()` and `total_cycle_time_enabled_incomplete_valves()` seem to report incorrect times when valve open delay is used
- `total_queue_time()` seems to miss the first valve in the queue
- `sprinkler.start_full_cycle` does not seem to work with a single valve controller
- `multiplier_number` component doesn't appear in front end when used with a single valve controller

Let me know if I've missed anything in this list! 😇

kbx81, 2023-05-11:
> Hi I'm jumping on this issue as I've just started using this integration - it's fantastic so far!
>
> I have noticed that the function `.total_queue_time()` doesn't seem to give the correct time, it [...]

@patfelst could you clarify what you mean by "miss the first valve in the queue"? I was just taking a quick peek at the code that does this [here](https://github.com/esphome/esphome/blob/ed024a0aa5f029b788af3398dbd382dcf5445036/esphome/components/sprinkler/sprinkler.cpp#L1126-L1133) and I don't really see how that's possible.

Did you mean that it's not including the valve that is currently running? If so, that's by design -- the valve that's running is no longer considered to be in the queue because it is the active valve. If you want the correct time, you need to add the result from `total_queue_time()` and `time_remaining_active_valve()` to get this. 😇 Perhaps this is something that could be clarified in the docs/comments.

patfelst, 2023-05-11:
yes that's what I meant, well I'm reporting `total_queue_time()` just after I start it, and I was expecting to see a few seconds under the queue time of adding up the valve times. But what you say makes sense, I'll try your solution.

Actually isn't this redundant, because via testing I've noticed that I get the expected remaining time via reporting `time_remaining_current_operation().value()`? I guess in the situation where you create a queue, but not start it, you might want to know that time.

kbx81, 2023-05-11:
> yes that's what I meant, well I'm reporting `total_queue_time()` just after I start it, and I was expecting to see a few seconds under the queue time of adding up the valve times. But what you say makes sense, I'll try your solution.

Sweet! I think that will get you the behavior you're looking for. 😄

> Actually isn't this redundant, because via testing I've noticed that I get the expected remaining time via reporting `time_remaining_current_operation().value()`? I guess in the situation where you create a queue, but not start it, you might want to know that time.

Yes, I broke the functionality out for different things in different ways because everybody seems to want to know something different. 😆 I _believe_ that with the methods available, at this point, you should be able to get any combination of things, with a couple super-common-everybody-wants-this methods built-in. In other words, if you want something other than time remaining current valve or time remaining _all inclusive,_ you should be able to pull that together using some combination of the methods already there. 😇

kbx81, 2023-05-11:
Regarding the text sensors, have you tried just using a regular `sensor` and setting `device_class: duration` to get "pretty" formatting in the front end? Seems a lot simpler than all of the extra code to make the text sensor work just so. 😇

patfelst, 2023-05-11:
Hey @kbx81 thanks for your responses, very helpful. I think your comment about pretty formatting is for @olympia but I'm interested too. From a quick search it seems formatting durations on the HA side front end is maybe not so easy, e.g.
https://community.home-assistant.io/t/understanding-device-class-duration-better/444140. Maybe there are simpler solutions I'm not aware of, but looks like some templating would need to be done, and I think it simpler in C++ personally :laughing:

This is how I approached it to make the yaml code a bit more readable, I created this function in a `.h` include file:
```C++

/* 
** convert a number of seconds to a human readable C++ string in the format:
** "dd hh mm ss"
*/
std::string seconds_to_readable(int sec) {
  int seconds = sec;
  int minutes = 0;
  int hours = 0;
  int days = 0;
  std::string txt = "";

  days = seconds / (24 * 3600);
  seconds = seconds % (24 * 3600);
  hours = seconds / 3600;
  seconds = seconds % 3600;
  minutes = seconds /  60;
  seconds = seconds % 60;

  if (days) txt = (to_string(days) + "d "); else "";
  if (hours) txt+= (to_string(hours) + "h "); else "";
  if (minutes) txt+= (to_string(minutes) + "m "); else "";
  txt+= to_string(seconds) + "s";
  // ESP_LOGW("Sprinkler debug", "Time remaining is %s", txt.c_str());
  return txt;
}
```
then you can publish the text sensor like:
```yaml
  - text_sensor.template.publish:
        id: text_time_remaining
        state: !lambda |-
          return seconds_to_readable(id(sprinkler_controller).time_remaining_current_operation().value());
```
and you can add to the text if you need, like I added the current queue name (I call it a program aka Rainbird irrigation naming)
```yaml
...
  - lambda: 'id(current_program_str) = "Prog A";'
...

      - text_sensor.template.publish:
          id: text_total_queue_time
          state: !lambda |-
            std::string prog_txt = "";
            prog_txt = id(current_program_str) + " " +
                        seconds_to_readable(id(sprinkler_controller).total_queue_time());
            return prog_txt;
```

olympia, 2023-05-11:
> Regarding the text sensors, have you tried just using a regular `sensor` and setting `device_class: duration` to get "pretty" formatting in the front end? Seems a lot simpler than all of the extra code to make the text sensor work just so. 😇

That's the approach I mentioned above what I tried and was wondering about a formatting feature in HA for. The cosmetic issue with this is that the way HA displays `device_class: duration` is still not pretty/ human readable enough. HA will display such sensors as 10:10:10 while the bulky code we sprinkler funs are usually adding (quoted by Patrick) are making it much more human readable: 10h 10m 10s. Hence I was wondering about an option to get this configurable in HA frontend, but I fully get and appreciate this is out of scope here.

olympia, 2023-05-11:
> # Issue Summary
> * `time_remaining_current_operation()` and `total_cycle_time_enabled_incomplete_valves()` seem to report incorrect times when valve open delay is used
> * `total_queue_time()` seems to miss the first valve in the queue -- **UPDATE:** [expected behavior](https://github.com/esphome/issues/issues/4502#issuecomment-1543202018)
> * `sprinkler.start_full_cycle` does not seem to work with a single valve controller
> * `multiplier_number` component doesn't appear in front end when used with a single valve controller
> * `run_duration` is not optional in `sprinkler.queue_valve` (but should be) -- **UPDATE:** see [sprinkler.queue_valve fails without run_duration #4069](https://github.com/esphome/issues/issues/4069)
> 
> Let me know if I've missed anything in this list! 😇

Thanks for the summary:

May I ask the followings to be added too:
-  'active_valve()'  seem to report incorrect valve when valve open delay is used (during the opening delay).
- It seems like none of the timer methods (either for total or remaining time methods) are adding up the open valve delay time when doing the math - this is not the same as your first bullet. I think your first one is more related to the 'active_valve()' being incorrect during the opening delay issue (I.e. my first bullet just above) -- who I hope this make sense to you :D

kbx81, 2023-05-13:
Please test https://github.com/esphome/esphome/pull/4816 and let me know if this helps. You can test it using [external components](https://next.esphome.io/components/external_components.html) like this:

```yaml
external_components:
  - source: github://pr#4816
    components: [ sprinkler ]
```

olympia, 2023-05-13:
> Please test [esphome/esphome#4816](https://github.com/esphome/esphome/pull/4816) and let me know if this helps. You can test it using [external components](https://next.esphome.io/components/external_components.html) like this:
> 
> ```yaml
> external_components:
>   - source: github://pr#4816
>     components: [ sprinkler ]
> ```

Wow, didn't expect such quick turn around time, many thanks @kbx81!

Just tested, here are the results:

- `sprinkler.start_full_cycle` does not seem to work with a single valve controller -- FIXED, thanks!
- `multiplier_number` component doesn't appear in front end when used with a single valve controller -- FIXED, thanks!
- `total_cycle_time_all_valves()` didn't calculate with the open valve delay time -- FIXED, thanks!
- `active_valve()` seem to report incorrect valve when valve open delay is used (during the opening delay) -- I am still a bit somewhat confused here, so I might came back re this, nevertheless this also seems to be FIXED, thanks. Can you confirm that while this is correct, the marker point in the log is still on the wrong place?
- total_cycle_time_enabled_incomplete_valves() seem to report incorrect times when valve open delay is used -- this still doesn't seems to report the correct value. Interesting is that the value seems to be correct after a clean boot and summing up Valve 1 + Valve 2 + valve open delay time, but gets incorrect once a cycle is initiated, then it will only calculate with Valve 2 without the valve open delay time -- does this make sense?
- `time_remaining_current_operation()` seem to report incorrect times when valve open delay is used -- this still seems to report incorrect time remaining. I still see two issue here: 1) not counting with the valve open delay time and 2) this actually reports 0 during valve open delay

kbx81, 2023-05-17:
> Wow, didn't expect such quick turn around time, many thanks @kbx81!

The first couple points were "stupid" issues that shouldn't have been an issue in the first place -- easy fixes. 😇

> * `sprinkler.start_full_cycle` does not seem to work with a single valve controller -- FIXED, thanks!
> * `multiplier_number` component doesn't appear in front end when used with a single valve [...]

Awesome, glad to hear it and thanks for confirming! 🍻

> * `active_valve()` seem to report incorrect valve when valve open delay is used (during the opening delay) -- I am still a bit somewhat confused here, so I might came back re this, nevertheless this [...]

Cool, glad this seems to be producing behavior that's a bit more in-line with what's expected. As for the logs, honestly, don't worry about that; what you're seeing in the logs is messaging generated by the state machine transitioning states/control. The behavior of the state machine is somewhat different from what we as humans might expect, but it is what's generating the messages in the logs. Regardless of these messages/behaviors, the valves will still perform as expected.

> * `total_cycle_time_enabled_incomplete_valves()` seem to report incorrect times when valve open delay is used -- this still doesn't seems to report the correct value. Interesting is that the value [...]

Yeah, I was hoping that the tweaking I did to `active_valve()` would fix it, but apparently not. I just took another look and did some work to address it -- it's still not _perfect_ but much closer I think. Please test again and let me know how it works. 😇

> * `time_remaining_current_operation()` seem to report incorrect times when valve open delay is used -- this still seems to report incorrect time remaining. I still see two issues here: 1) not counting with the valve open delay time and 2) this actually reports 0 during valve open delay

As above -- this should be (mostly) addressed, too. You'll still see the timer "hang" during the valve open delay, but it will at least report a sane value now and just pick up once the valve actually starts. I forgot how complex this whole thing is; the reason for this behavior is because there is no visibility into the timers used in the `set_timeout` functions we have in ESPHome. The sprinkler controller's timers will need to be re-worked to use a mechanism that's independent of this and that's a bit more work than I can take on right now. Hopefully this at least puts you into a better spot for the time being. 😄

In any event, please try with these latest changes and let me/us know if it's all behaving as expected now. 😇

olympia, 2023-05-17:
> As above -- this should be (mostly) addressed, too. You'll still see the timer "hang" during the valve open delay, but it will at least report a sane value now and just pick up once the valve [...]

I suppose the timer "hanging" means that `time_remaining_current_operation()` will not include the valve open delay, but it should stop counting down during valve open delay, right. I.e. If you have 2 valves, each with 1 min run duration and 10s open valve delay then `time_remaining_current_operation()` should start counting down from 2 mins (not including the 10s open valve delay time), stop counting down at 1min during 10s valve open delay time when both valves are closed and then should start counting down again as soon as valve 2 started. Is this a correct interpretation? 
If yes, then this is not what's happening. Please see `Sprinkler Controller Overall Time Remaining` in the below log. It reports 0s when valve open delay is active. 
**(one question here though: should I add anything to the esphome code to enforce re-downloading of new commits from the external component - I didn't see any evidence in the compile logs that it redownloaded your latest changes).**

Nevertheless, on top of that, if my above interpretation of what "hang" means is correct, then I suppose for the sake of consistency nor `total_cycle_time_all_valves()`, neither `total_cycle_time_enabled_incomplete_valves()` should have the valve open delay time added (opposed to what I was asking for initially).
This is because one should be able to calculate overall progress percentage using:
`((id(sprinkler_ctrl).total_cycle_time_all_valves() - id(sprinkler_ctrl).time_remaining_current_operation().value_or(0)) * 100 / id(sprinkler_ctrl).total_cycle_time_all_valves() );` 
If `time_remaining_current_operation()` doesn't have open valve delay time included, but `total_cycle_time_all_valves()` does, then this math will return an incorrect percentage.

Logs:
```
[10:27:18][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '1m 5s'
[10:27:18][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 50.00000  with 1 decimals of accuracy
[10:27:19][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '4s'
[10:27:19][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 93.00000  with 1 decimals of accuracy
[10:27:19][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '1m 4s'
[10:27:20][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '3s'
[10:27:20][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 95.00000  with 1 decimals of accuracy
[10:27:20][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '1m 3s'
[10:27:20][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 51.00000  with 1 decimals of accuracy
[10:27:21][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '2s'
[10:27:21][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 96.00000  with 1 decimals of accuracy
[10:27:21][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '1m 2s'
[10:27:21][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 52.00000  with 1 decimals of accuracy
[10:27:22][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '1s'
[10:27:22][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 98.00000  with 1 decimals of accuracy
[10:27:22][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '1m 1s'
[10:27:22][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 53.00000  with 1 decimals of accuracy
[10:27:23][D][text_sensor:064]: 'Sprinkler Zone 1 Time Remaining': Sending state '0s'
[10:27:23][D][sensor:109]: 'Sprinkler Zone 1 Progress Percentage': Sending state 100.00000  with 1 decimals of accuracy
[10:27:23][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '1m 0s'
[10:27:24][D][switch:016]: 'Relay 1' Turning OFF.
[10:27:24][D][switch:055]: 'Relay 1': Sending state OFF
[10:27:24][D][switch:055]: 'Sprinkler Zone 1': Sending state OFF
[10:27:24][D][text_sensor:064]: 'Sprinkler Controller Status': Sending state 'Pending'
[10:27:24][D][text_sensor:064]: 'Sprinkler Zone 2 Status': Sending state 'Pending'
[10:27:24][D][text_sensor:064]: 'Sprinkler Zone 1 Status': Sending state 'Completed'
[10:27:24][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '0s'
[10:27:24][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 100.00000  with 1 decimals of accuracy
[10:27:25][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '0s'
[10:27:26][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '0s'
[10:27:27][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '0s'
[10:27:28][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '0s'
[10:27:29][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '0s'
[10:27:30][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '0s'
[10:27:31][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '0s'
[... cut ...]
[10:27:38][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '55s'
[10:27:38][D][sensor:109]: 'Sprinkler Controller Overall Progress Percentage': Sending state 57.00000  with 1 decimals of accuracy
[10:27:38][D][text_sensor:064]: 'Sprinkler Zone 2 Time Remaining': Sending state '55s'
[10:27:38][D][sensor:109]: 'Sprinkler Zone 2 Progress Percentage': Sending state 8.00000  with 1 decimals of accuracy
[10:27:39][D][text_sensor:064]: 'Sprinkler Controller Overall Time Remaining': Sending state '54s'
```

kbx81, 2023-05-17:
I think something didn't refresh on your end. `time_remaining_current_operation()` no longer returns zero during the valve open delay interval -- it should simply "pause"/stop counting during this interval, and "jump" down to match the correct time remaining when the valve actually starts. Try cleaning your build cache and re-run your build.

olympia, 2023-05-17:
Yeah, that was kind of suspicious to me as well, hence my question in bold and of course you were right, cleaning the build cache was the cure.

...and I happily report that your latest changes indeed working on the way you described and with those few mentioned minor conditions, all functions we were discussing here are now having a very nice, almost perfect behavior!

Thank you very much for your attention and constructive and very fast fixes!

Are you going to merge this PR to the next ESPHome version?

kbx81, 2023-05-17:
Awesome -- happy to hear it and (again) thanks for re-testing and confirming!

Yes, I think we'll aim to get this into the 5.1 release this month. A [comment on the PR](https://github.com/esphome/esphome/pull/4816) to confirm you have acceptable behavior would also be great, if you don't mind. 😇

olympia, 2023-05-17:
Sure thing Keith, done.

closed 2023-05-17 by jesserockz