esphome/issues #4505: Can not use gpio 0 for adc

Dzarda7, 2023-05-09:
### The problem

Esp32 in esphome can not use gpio 0 for ADC. It will not let me compile it, even though it works in Arduino IDE. Same for ESP32 C3

### Which version of ESPHome has the issue?

2023.4.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Esp32 and esp 32 C3 wroom modules

### Component causing the issue

adc

### Example YAML snippet

```yaml
# Example configuration entry
sensor:
  - platform: adc
    pin: 0
    name: "Living Room Brightness"
    update_interval: 60s
```

ssieb, 2023-05-09:
You can't use GPIO0 for adc on the ESP32 because it's on ADC2 and that's not compatible with wifi.
It's supported on the C3 though.  Show your config and the log messages if you think it isn't.

Dzarda7, 2023-05-10:
Thank you, ESP32 C3 is my main problem. I was trying  ESP32 just to be sure. Here is my config. I can not show you my log right now. Will do in the evening. The error it is writing is that GPIO0 can not be used as ADC.
[test.txt](https://github.com/esphome/issues/files/11438310/test.txt)

Dzarda7, 2023-05-10:
Isn't is problem with board type? But even though I think it should be supporrted.

ssieb, 2023-05-10:
I copied your adc sensor to my C3 config that has the same board and it compiled with no errors.

closed 2023-05-10 by Dzarda7