esphome/issues #4509: ESP 8266 impossible to Flash

Andro67, 2023-05-12:
### The problem

Hello,
I hope i'm posting on the right place :)
I can't flash my Wemos D1 mini ( i have two and the same problem with all)
I tested with 6 differents cable USB
I tested 2 differents Computer and differents USB ports 
i have en error message on the screen from ESP Flasher:

__Using 'COM6' as serial port.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Connecting....

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 0066E412
 - MAC Address: 48:3F:DA:66:E4:12
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
**Unexpected error: The firmware binary is invalid (magic byte=45, should be E9)_**_

If i press View Logs:
Using 'COM6' as serial port.
Unexpected error: could not open port 'COM6': PermissionError(13, 'Accès refusé.', None, 5)

Thank you for futuring help.

PS: What type of installation are you using? 
I don't know. I start directly the EXE, i have no Home Assistant, Docker or pip

### Which version of ESPHome has the issue?

ESP HOME flasher 1.4.0 windows X86

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

WEMOS D1 mini

ssieb, 2023-05-12:
github issues are not for support requests.  Ask on the esphome discord or one of the forums.

closed 2023-05-12 by ssieb

Andro67, 2023-05-12:
I found the solution, my file was corrupted, i re download and no is going. 
Tahnk you