esphome/issues #4513: ESP32 is consuming 1.3 mA in deep sleep

noeldtz, 2023-05-15:
### The problem

hi all, 

i made a battery-powered air quality sensor, using BME680 with ESP32 (Firebeetle DFR0654). The ESP32 is powered by 2 LiFePO in parallel directly connected to the 3v pin of the board. The sensor wakes up every 90 seconds, dumps sensor values to MQTT and goes back to sleep again. During deep sleep, the board is supposed to consume power in the micro-amps range, but when measuring this myself with a multimeter, it consistently consumes about 1.3 mA. This power consumption is too high for a long-term battery powered device as the battery will be drained within a few weeks. I tried to find a solution to this and i came across one in #1668 where they suggest that the issue is related to the fact that the ADC is not powered off before the device enters deep sleep. They also suggest to somehow include the adc_power_off function into the project (either by modifying the deep sleep component of esphome, or by creating a custom component). As i have installed ESPhome and home assistant with docker-compose, i'm a little clueless on where to find the deep_sleep component in order to be able to patch it in. I'm also not experienced enough to create a custom component yet, so i would really appreciate if someone would be able to help me with this. 

Thanks!!

### Which version of ESPHome has the issue?

2023.4.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

 2023.5.2

### What platform are you using?

ESP32

### Board

Firebeetle DFR0654

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
esphome:
  name: sensor-badkamer-new
  friendly_name: sensor-badkamer-new

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

ota:
  password: "a6d5a387c8aa9405d7e6e260f358c710"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Sensor-Badkamer-New"
    password: "iCKl5JejpjZH"

captive_portal:
    
deep_sleep:
  id: deep_sleep_1
  sleep_duration: 90s
  esp32_ext1_wakeup:
    pins: 27
    mode: ANY_HIGH 
  
mqtt:
  broker: 192.168.2.19
  username: !secret mqtt_username
  password: !secret mqtt_password
  birth_message:
  will_message:
  discovery: false 
  discovery_retain: false
  reboot_timeout: 1 min 
  on_connect:
    - binary_sensor.template.publish:
        id: MQTT_connected
        state: ON
  
i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_a

binary_sensor:
  - platform: template
    name: wifi_received
    id: wifi_received
    internal: True 

  - platform: template
    name: battery_received
    id: battery_received
    internal: True 

  - platform: template
    name: temperature_received
    id: temperature_received
    internal: True

  - platform: template
    name: pressure_received
    id: pressure_received
    internal: True

  - platform: template
    name: humidity_received
    id: humidity_received
    internal: True

  - platform: template
    name: gas_resistance_received
    id: gas_resistance_received
    internal: True

  - platform: template
    name: MQTT_connected
    id: MQTT_connected
    internal: True

  - platform: template
    name: Updates from all sensors received
    id: all_updates_received
    internal: True
    #icon: mdi:database-refresh-outline
    lambda: |-
      return
      id(wifi_received).state &&
      id(battery_received).state &&
      id(temperature_received).state &&
      id(pressure_received).state &&
      id(humidity_received).state &&
      id(gas_resistance_received).state && 
      id(MQTT_connected).state;
    
    on_state:
      if:
        condition:
          binary_sensor.is_on: all_updates_received  
        then:
        - binary_sensor.template.publish:
            id: MQTT_connected
            state: OFF      
        - binary_sensor.template.publish:
            id: wifi_received
            state: OFF
        - binary_sensor.template.publish:
            id: battery_received
            state: OFF
        - binary_sensor.template.publish:
            id: temperature_received
            state: OFF
        - binary_sensor.template.publish:
            id: pressure_received
            state: OFF
        - binary_sensor.template.publish:
            id: humidity_received
            state: OFF      
        - binary_sensor.template.publish:
            id: gas_resistance_received
            state: OFF                       
        - binary_sensor.template.publish:
            id: all_updates_received
            state: OFF  
        - deep_sleep.enter: deep_sleep_1
          
sensor:
  - platform: wifi_signal
    name: "WiFi_Signal_Sensor"
    id: "WiFi_Signal_Sensor"
    on_value:
        then:
          - binary_sensor.template.publish:
              id: wifi_received
              state: ON

  - platform: adc
    pin: A0
    name: "battery_voltage"
    id: "battery_voltage"
    attenuation: auto
    on_value:
        then:
          - binary_sensor.template.publish:
              id: battery_received
              state: ON

  - platform: bme680
    temperature:
      name: "BME680 Temperature"
      on_value:
        then:
          - binary_sensor.template.publish:
              id: temperature_received
              state: ON
    pressure:
      name: "BME680 Pressure"
      on_value:
        then:
          - binary_sensor.template.publish:
              id: pressure_received
              state: ON

    humidity:
      name: "BME680 Humidity"
      on_value:
        then:
          - binary_sensor.template.publish:
              id: humidity_received
              state: ON

    gas_resistance:
      name: "BME680 Gas Resistance"
      on_value:
        then:
          - binary_sensor.template.publish:
              id: gas_resistance_received
              state: ON
    
    heater: 
      duration: 0 ms
    address: 0x77
```

tcirstea, 2023-05-18:
Not a dev on the project, and randomly stumbled upon this issue. Having said that, there's probably a couple questions worth asking:
- What multimeter are you using, and is it rated for such low power?
- Did you cut the low power pad for your DFR0654?
  - https://lucidar.me/fr/esp32/power-consumption-of-esp32-firebeetle-dfr0654/
- Another thing to try is to remove the ADC usage - since the issue you linked is specifically due to that. See if that decreases the usage down to ~10 uA, and then you will have at least found the culprit - and can track it in the other linked issue.

KoRRo, 2023-09-17:
I came here to ask a similar question and I've found yours.
For me the power consumption is even higher at 1.77mA, but I think I've found a possible cause. It looks like the wifi doesn't sleep properly. If I call "wifi.disable" before "deep_sleep.enter" the power consumption goes down to 0.2mA.
Can you check if it's the same for you?