esphome/issues #4514: HomeKit BLE Nanolear Remote via ESPHome BT Proxy

psychowood, 2023-05-15:
### The problem

I'm trying to connect a NanoLeaf Remote via ESPHome BT Proxy but I'm having some issues.

The device itself should be supported ([confirmed working without a proxy](https://github.com/home-assistant/core/issues/80455)), I was able to pair it collectly in the HomeKit integration using current dev version (ESPHome version 2023.4.4 compiled on May  8 2023, 19:19:02):

[image]

The initialization fails, tho:

`Logger: homeassistant.config_entries`
`Source: config_entries.py:1247`
`First occurred: 19:21:12 (1 occurrences)`
`Last logged: 19:21:12`
`Config entry 'Nanoleaf Remote 947D' for homekit_controller integration not ready yet: Nanoleaf Remote 947D [E3:32:40:XX:XX:XX] (id=8B:70:15:YY:YY:YY) connection failed: Bluetooth GATT Error address=E3:32:40:XX:XX:XX handle=5 error=133 description=Error; rssi=None; Retrying in background`

this looks like the same error in the proxy logs:

`[18:24:48][W][bluetooth_proxy.connection:081]: [0] [E3:32:40:XX:XX:XX] Error reading char/descriptor at handle 0x48, status=133`

Any ideas?

Attached below are the bluetooth proxy logs:

[logs_esp32-bluetooth-proxy-xxxxxx_logs.txt](https://github.com/esphome/issues/files/11481093/logs_esp32-bluetooth-proxy-xxxxxx_logs.txt)

### Which version of ESPHome has the issue?

ESPHome version 2023.4.4 compiled on May  8 2023, 19:19:02 (stable version couldn't pair it)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.5.3

### What platform are you using?

ESP32

### Board

esp32-wroom

### Component causing the issue

bluetooth-proxy

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-xxxxxx
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  enable_btm: true
  enable_rrm: true
  reboot_timeout: 5min

api:
  reboot_timeout: 5min

bluetooth_proxy: # or not
  active: true

esp32_ble_tracker: # or not
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

button:
  - platform: restart
    name: "BT Proxy Restart"
```

### Anything in the logs that might be useful for us?

```txt
[18:24:48][W][bluetooth_proxy.connection:081]: [0] [E3:32:40:XX:XX:XX] Error reading char/descriptor at handle 0x48, status=133
```

bdraco, 2023-08-21:
Proxies only supported 4 notify handles until recently. They now support 9 notify handles.

Update your ESPHome device to 2023.8.2 or later and Home Assistant to 2023.8.x or later

psychowood, 2023-09-08:
> Proxies only supported 4 notify handles until recently. They now support 9 notify handles.
> 
> Update your ESPHome device to 2023.8.2 or later and Home Assistant to 2023.8.x or later

Updated (ESPHome 2023.8.3, Home Assistant 2023.9.0, also tried with ESPHome-dev) and retried, without success:

[image]

bdraco, 2023-09-08:
Bummer 133 is a generic error that doesn't tell us much.  Hopefully a future ESP-IDF improves compatiblity