esphome/issues #4516: ESPHome Cover doesn't show "stop" in home assistant only up and down since 2023.5.0

pbvdven, 2023-05-17:
### The problem

Since the update to esphome 2023.5.0 the cover entity lost the "stop" button in home assistant.

ESPhome
[image]

Home assistant
[image]

### Which version of ESPHome has the issue?

2023.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.3

### What platform are you using?

ESP8266

### Board

nodemcu-32s

### Component causing the issue

cover

### Example YAML snippet

```yaml
esphome:
  name: somfy
  platform: ESP32
  board: nodemcu-32s
  libraries:
    - EEPROM
    - SPI
    - SmartRC-CC1101-Driver-Lib@2.5.6
    - Somfy_Remote_Lib@0.4.1
  includes:
    - somfy_secrets.h
    - somfy_cover.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ESPSomfyGateway
    password: !secret api_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key
    
ota:
  password: !secret ota_password

web_server:
  port: 80
  auth:
    username: pbvdven
    password: !secret web_server_password

sensor:
  - platform: uptime
    name: "Uptime somfy"
  
  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 60s
    entity_category: "diagnostic"

  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"

cover:
  - platform: custom
    lambda: |-
      auto somfy_remote = new SomfyESPRemote();
      somfy_remote->add_cover("somfy", "woonkamer", SOMFY_REMOTE_ZW_WOONKAMER);
      App.register_component(somfy_remote);
      return somfy_remote->covers;

    covers:
      - id: "somfy"
        name: "Zonnewering woonkamer"

switch:
  - platform: template
    name: "PROG"
    turn_on_action:
      - lambda: |-
          ((SomfyESPCover*)id(somfy))->program();
  - platform: template
    name: "Zonnewering woonkamer My"
    turn_on_action:
      - lambda: |-
          ((SomfyESPCover*)id(somfy))->make_call().set_command_stop().perform();
  - platform: template
    name: "Zonnewering woonkamer MyUp"
    turn_on_action:
      - lambda: |-
          ((SomfyESPCover*)id(somfy))->myup();
  - platform: template
    name: "Zonnewering woonkamer MyDown"
    turn_on_action:
      - lambda: |-
          ((SomfyESPCover*)id(somfy))->mydown();
  - platform: template
    name: "Zonnewering woonkamer UpDown"
    turn_on_action:
      - lambda: |-
          ((SomfyESPCover*)id(somfy))->updown();
```

### Anything in the logs that might be useful for us?

```txt
Cant find anything about the issue in the HA log
```

hynek2001, 2023-05-21:
having same issue. not sure what to try/debug. anyone have idea ?

muddyfeet, 2023-05-25:
Had the same issue with a somfy controller. Solution is to add to the somfy_cover.h file that you are using the line traits.set_supports_stop(true). 

CoverTraits get_traits() override {
    auto traits = CoverTraits();
    traits.set_is_assumed_state(true);
    traits.set_supports_position(false);
    traits.set_supports_tilt(false);
    **traits.set_supports_stop(true);**
    return traits;
}

jesserockz, 2023-05-25:
You are using a custom component/header. It is up to the autohor of those components to keep them up to date, or PR them into ESPHome where they would be kept up to date when we make internal changes like this.

closed 2023-05-25 by jesserockz

deltaphi, 2023-06-08:
I run three Somfy RTS covers off of an ESP32. I also ran into this issue. I added `traits.set_supports_stop(true);` to my trait object and re-flashed my ESP wirelessly. Unfortunately, my Home Assistant still does not show the stop button. I also tried restarting Home Assistant, but no luck.

Is there something else I need to update or to reload to convince home assistant that the stop functionality is now there?

muddyfeet, 2023-06-08:
I think that is all that is required as far as I know. My code is as above - I only added the traits.set_supports_stop(true); line to get it working again.

tracestep, 2023-09-14:
This should have definitively been marked as a breaking change!