esphome/issues #4517: Esphome 2023.5.0 pyparsing version incompatible

miGii, 2023-05-17:
### The problem

When I compile (on 2023.5.0) any device's code I got following error in logs:
`ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
Successfully installed cachecontrol-0.12.11 certifi-2023.5.7 charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.6 contextlib2-21.6.0 future-0.18.3 idf-component-manager-1.2.2 idna-3.4 lockfile-0.12.2 msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0 requests-2.30.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.65.0 urllib3-2.0.2
esphome 2023.5.0 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.`

Despite this, the code compiles and installs correctly.

### Which version of ESPHome has the issue?

2023.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.3

### What platform are you using?

ESP32

selfmade01, 2023-05-17:
I have had same Problem. (with ESP32-S2)
i get that the first time when i updated my esphome version to 2023.5.0
after that i updated HA-Version too to 2023.5.3

when I compiled the same esphome code the second time, the error was gone (after I updated my HA)
I'm not sure if that actually fixed the problem.
It's possible that if I compile another ESP32-S2 I'll get the error again.

PaleoW, 2023-05-17:
This seems to happen when using the ESP-IDF platform instead or Arduino.

```
Installing ESP-IDF's Python dependencies
Collecting pyparsing<2.4.0,>=2.0.3
  Using cached pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Installing collected packages: pyparsing
  Attempting uninstall: pyparsing
    Found existing installation: pyparsing 3.0.9
    Uninstalling pyparsing-3.0.9:
      Successfully uninstalled pyparsing-3.0.9
Successfully installed pyparsing-2.3.1
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.5.0 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
```

ssieb, 2023-05-17:
Yes, it is specific to IDF:
```
Installing ESP-IDF's Python dependencies
Collecting pyparsing<2.4.0,>=2.0.3
```

tplas, 2023-05-18:
I am seeing same issue in the log, and only occurs when updating device that uses ESP-IDF.  Which is required (or highly recommended) for Bluetooth Proxy ESPHome devices.

I saw it on my first update of a BT Proxy device, & haven't done any more since. That one is an M5Stack AtomU device.  I have not updated my other BT Proxy devices-- one more M5Stack AtomU, and two GL-S10 devices from GL-iNet.

The one unit I did update displayed the warning message, then continued with compile / install, and seems to be operational after it rebooted.  (But I'm not going to update any more, esp. the GL-S10 units, until I hear about resolution.)

twister36, 2023-05-18:
same issue for me when trying to push the latest version of ESPHome to my devices

liamlain, 2023-05-18:
i got the error when updating:

ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.5.1 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.

Pyromane, 2023-05-18:
```
rocessing esp32-bluetooth-proxy-455014 (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

.....

  Attempting uninstall: pyparsing
    Found existing installation: pyparsing 3.0.9
    Uninstalling pyparsing-3.0.9:
      Successfully uninstalled pyparsing-3.0.9
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.5.1 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
Successfully installed cachecontrol-0.12.11 certifi-2023.5.7 charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.6 contextlib2-21.6.0 future-0.18.3 idf-component-manager-1.2.2 idna-3.4 lockfile-0.12.2 msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0 requests-2.30.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.65.0 urllib3-2.0.2
```

janprive, 2023-05-18:
Same here with my bluetooth proxy!

TRusselo, 2023-05-18:
Same here for updating my bt proxy

TaylorTWBrown, 2023-05-22:
Same issue. Using ESP-IDF framework with the bluetooth-proxy functionality. Running ESPHome 2023.5.2

Tamsy, 2023-05-22:
Exactly the same here with ESP-IDF:

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.5.2 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
```

KosieKramer, 2023-05-22:
I had the same issue, but after cleaning the build files, the build worked without error.

KosieKramer, 2023-05-23:
Nope... seems the error is back this morning after ESPHome update to 2023.5.3

lysiong, 2023-05-23:
same here as well whichever there is an ESPHome update on ESPHome 2023.5.0, 2023.5.1, 2023.5.2, 2023.5.3

`ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.5.3 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.`

tplas, 2023-05-23:
I'm afraid if you just run re-install, it doesn't do full  compile (just sends already-compiled firmware to the device), so it doesn't go thru the compilation dependency checks.

I also tried the suggestion to Clean Build Files before recompile (when I updated to 2023.5.3), but still got same error during the build.  

But once again, even though it issued that warning, the compile process proceeded to completion, and successfully sent the new firmware to the device, and the device appears fully operational. So am not sure of impact of this warning message.

But I'm still not updating my other BT Proxy devices until we learn what this is all about.

Meli73, 2023-05-24:
still get this Error, too. But seams to run, anyway

rleongcs, 2023-05-24:
Same here even with 2023.5.3.

KosieKramer, 2023-05-25:
Error is still present in 2023.5.4.

sippeeey, 2023-05-25:
Same here with 2023.5.4 by Update my M5 Stack atom-bluetooth-proxy.

IamMattM, 2023-05-25:
Same here for me with ESPHOME v 2023.5.4

micky-bamboleo, 2023-05-25:
Same here (atom bt proxy, ESPHome 2023.5.4)

distante, 2023-05-27:
Same ESP32 with BT proxy, ESPHome 2023.5.4

brunopiras, 2023-05-28:
Same ESP32 BLE,  ESPHome 2023.5.4

natodd, 2023-05-29:
Same here. Just installed the latest and still getting it. ESP32 BLE, ESPHome 2023.5.5

horve000, 2023-05-29:
Same problem here.

emilianogetino, 2023-05-29:
Same problem here.

napalmz, 2023-05-30:
Same here, sadly.

tplas, 2023-05-30:
Wow, the issue seems to have been resolved with my devices.

I have installed Home Assistant Core 2023.5.4 and ESPHome 2023.5.5

I have four ESP32 BT Proxy devices with ESP-IDF framework, in two different hardware types: GL-S10 from GLiNet, and AtomU from M5Stack.

None of them issued the warning message during most recent build; they had completely clean install when I updated them for newest version (2023.5.5) of ESPHome.

I notice that the release notes for ESPHome 2023.5.5 mention "not breaking yaml parsing".  So hopefully this was it?

adekloet, 2023-05-30:
> Links:

I still see the error on esphome 5.5 on HASSOS RPI4

ERROR: pip's dependency resolver does not currently take into account 
all the packages that are installed. This behaviour is the source of the 
following dependency conflicts.
esphome 2023.5.5 requires pyparsing>=3.0, but you have pyparsing 2.3.1 
which is incompatible.
Successfully installed cachecontrol-0.12.11 certifi-2023.5.7 
charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.6 contextlib2-21.6.0 
future-0.18.3 idf-component-manager-1.2.3 idna-3.4 lockfile-0.12.2 
msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0 requests-2.31.0 
requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 
tqdm-4.65.0 urllib3-1.26.16
Reading CMake configuration...

scoulondre, 2023-06-05:
Same here, esp-idf requires pyparsing<2.4.0,>=2.0.3 and esphome requires pyparsing>=3.0
We are stuck with this conflict

adekloet, 2023-06-05:
Oke what I did is get esphome 5.5 installed. Clean the build files for the device your upgrading and then install. I had no error after that. The "clean build files" removes all previous cached files and does a full new download/cmake/compile/link and it seemed to work. Give it a try.

  (Issue #4517)

scoulondre, 2023-06-05:
Finally, cleaning everything, including all caches (under Debian: .esphome, ~/.platformio, venv), then installing again everything still shows the pyparsing problem, but it does not stop the process anymore.

tplas, 2023-06-05:
Agreed. That's what I've done.

I'm running HA Core 2023.5.4 and ESPHome 2023.5.5. 

I performed "Clean build files" --but only for one of the devices; does that operation clear them for all when you do that?

I then performed Install for each of my ESP-IDF ESPHome devices -- All are BT Proxy devices: two are GL-S10 units from GLiNet, and two are AtomU units from M5Stack. All of them compiled cleanly with no errors (other than usual pin-usage warning on the GL-S10's).

All four of my devices compiled and installed / uploaded just fine, and all are running fine.

smartmatic, 2023-06-17:
Same here, installed my first M5 Atom Lite as ESP Bluetooth Proxy and get this error in the logs.
My system is:
HA 2023.6
ESPHome 2023.5.5

BluThunder2k, 2023-06-22:
Same here on ESPHome 5.x and 6.0.    Affects my bluetooth proxies as well.

HaraldGithub, 2023-06-24:
I've had the same problem with one of my M5Stack Bluetooth proxy:
[image]
But find a solution:
KLICK "Clean Build Files" in ESPhome-main-menue for each devices.
After that the problem doesn't occur any longer during the update process.

crzykidd, 2023-06-24:
> KLICK "Clean Build Files" in ESPhome-main-menue for each devices.

This solved it for me as well.   Thanks for the share!

KitHubek, 2023-07-04:
newest update to 2023.6.4 and error still exsist (cleaned built files before update)

```
INFO ESPHome 2023.6.4
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-bluetooth-proxy (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Installing ESP-IDF's Python dependencies
Collecting future>=0.15.2
  Downloading future-0.18.3.tar.gz (840 kB)
Collecting pyparsing<2.4.0,>=2.0.3
  Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Collecting idf-component-manager~=1.0
  Downloading idf_component_manager-1.3.1-py2.py3-none-any.whl (134 kB)
Collecting tqdm<5
  Downloading tqdm-4.65.0-py3-none-any.whl (77 kB)
Collecting schema
  Downloading schema-0.7.5-py2.py3-none-any.whl (17 kB)
Collecting six
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting requests<3
  Downloading requests-2.31.0-py3-none-any.whl (62 kB)
Collecting pyyaml>5.2
  Downloading PyYAML-6.0-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_12_x86_64.manylinux2010_x86_64.whl (661 kB)
Collecting cachecontrol[filecache]>0.12.6
  Downloading cachecontrol-0.13.1-py3-none-any.whl (22 kB)
Collecting click
  Downloading click-8.1.3-py3-none-any.whl (96 kB)
Collecting packaging
  Downloading packaging-23.1-py3-none-any.whl (48 kB)
Collecting urllib3<2
  Downloading urllib3-1.26.16-py2.py3-none-any.whl (143 kB)
Collecting contextlib2>0.6.0
  Downloading contextlib2-21.6.0-py2.py3-none-any.whl (13 kB)
Collecting colorama
  Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting requests-toolbelt
  Downloading requests_toolbelt-1.0.0-py2.py3-none-any.whl (54 kB)
Collecting requests-file
  Downloading requests_file-1.5.1-py2.py3-none-any.whl (3.7 kB)
Collecting msgpack>=0.5.2
  Downloading msgpack-1.0.5-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (322 kB)
Collecting filelock>=3.8.0
  Downloading filelock-3.12.2-py3-none-any.whl (10 kB)
Collecting idna<4,>=2.5
[... 41 lines cut ...]
    Found existing installation: packaging 23.1
    Uninstalling packaging-23.1:
      Successfully uninstalled packaging-23.1
  Attempting uninstall: colorama
    Found existing installation: colorama 0.4.6
    Uninstalling colorama-0.4.6:
      Successfully uninstalled colorama-0.4.6
  Attempting uninstall: click
    Found existing installation: click 8.1.3
    Uninstalling click-8.1.3:
      Successfully uninstalled click-8.1.3
  Attempting uninstall: pyparsing
    Found existing installation: pyparsing 3.1.0
    Uninstalling pyparsing-3.1.0:
      Successfully uninstalled pyparsing-3.1.0
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.6.4 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
Successfully installed cachecontrol-0.13.1 certifi-2023.5.7 charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.6 contextlib2-21.6.0 filelock-3.12.2 future-0.18.3 idf-component-manager-1.3.1 idna-3.4 msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0 requests-2.31.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.65.0 urllib3-1.26.16
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- noise-c @ 0.1.4
|-- Improv @ 1.2.3
Compiling /data/esp32-bluetooth-proxy/.pioenvs/esp32-bluetooth-proxy/src/esphome/components/api/api_connection.o
```

HaraldGithub, 2023-07-04:
Same here.
I reinstalled my second proxy (up to this point without this error message) and now, I've the error on both proxies.
I think it's a bug on ESPHome-side, since HA is updated to python 3.11 with 2023.4.0-Update.

rleongcs, 2023-07-04:
Same here when updating to 2023.6.4 even with clean build. Earlier versions with clean build could not see the error already.

[image]

lysiong, 2023-07-05:
I notice that it only happen on framework esp-idf, but on framework arduino working fine. I mean the compiler.

HaraldGithub, 2023-07-05:
@lysiong 
How could I select between this different compilers?

regystro, 2023-07-05:
> @lysiong How could I select between this different compilers?

https://esphome.io/search.html?q=esp-idf

rleongcs, 2023-07-05:
> @lysiong How could I select between this different compilers?

@HaraldGithub You can change the compilers under "type". Having said that, we need esp-idf for bluetooth proxy. 😞 

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
```

lysiong, 2023-07-05:
> @lysiong How could I select between this different compilers?

when you set the ESP32 as below, framework type set to arduino

esp32:
  board: esp32dev
  framework:
    type: arduino

lysiong, 2023-07-05:
@rleongcs actually got different between arduino & esp-idf? previously I did use arduino, then later found up esp-idf also can compile.

rleongcs, 2023-07-05:
> @rleongcs actually got different between arduino & esp-idf? previously I did use arduino, then later found up esp-idf also can compile.

I believed there are but not totally sure the differences. At least the bluetooth proxy documention says it needs to use esp-idf.

nepozs, 2023-07-13:
I do not now why #4670 is closed, as it is new problem for me, and I'm upgrading ESPHome every month (or even every release).
More interesting I've uninstalled ESPHome and installed addon again, but error persist…
```
Successfully installed cachecontrol-0.13.1 certifi-2023.5.7 charset-normalizer-3.2.0 click-8.1.5 colorama-0.4.6 contextlib2-21.6.0 filelock-3.12.2 future-0.18.3 idf-component-manager-1.3.2 idna-3.4 msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0 requests-2.31.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.65.0 urllib3-1.26.16
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.6.5 requires click==8.1.3, but you have click 8.1.5 which is incompatible.
esphome 2023.6.5 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
Reading CMake configuration...
Generating assembly for certificate bundle...
No dependencies
```
Compilation was possible only after fresh install, BUT really imposiible after upgrade from 2023.6.4 to 2023.6.5

ssieb, 2023-07-13:
#4670 was closed as a duplicate of this one.  There's no need to have multiple copies of the same issue.
Are you saying it won't compile now?  All I've seen so far is that it complains, but everything still works.

HaraldGithub, 2023-07-14:
[quote]All I've seen so far is that it complains, but everything still works[/quote]
That's right, but boring!
Why isn't it possible, that during the next ESP-Update the right pyparsing-version (>3.0) will be installed?

ssieb, 2023-07-14:
Because the esp-idf version that is being used requires something lower.

HaraldGithub, 2023-07-14:
I don't believe in that and it's for me not logical, because I've two ESP-BT-Proxies, one with the above mentioned error message and one without.
Both are working well, one with a pyparsing-version >3 (3.0.1?) and one with the old on an the error.
So, were is the problem?

ssieb, 2023-07-14:
That message only happens once.

kixtarter, 2023-07-16:
With the ESPHome update to 2023.6.5 this problem still persists, and now there is a similar issue with click (Command Line Interface Creation Kit?).
This is an excerpt from the ESPHome install log showing the (I hope) applicable parts when updating an ESP32 device with the espidf framework:

```
INFO ESPHome 2023.6.5
INFO Reading configuration /config/esphome/t7v13-02.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing t7v13-02 (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Installing ESP-IDF's Python dependencies

Collecting pyparsing<2.4.0,>=2.0.3
  Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)

Collecting click
  Downloading click-8.1.5-py3-none-any.whl (98 kB)

Installing collected packages:  click, pyparsing, 

  Attempting uninstall: click
    Found existing installation: click 8.1.3
    Uninstalling click-8.1.3:
      Successfully uninstalled click-8.1.3
  Attempting uninstall: pyparsing
    Found existing installation: pyparsing 3.1.0
    Uninstalling pyparsing-3.1.0:
      Successfully uninstalled pyparsing-3.1.0
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.6.5 requires click==8.1.3, but you have click 8.1.5 which is incompatible.
esphome 2023.6.5 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
Successfully installed click-8.1.5, pyparsing-2.3.1 
Reading CMake configuration...
Generating assembly for certificate bundle...
No dependencies
```

Following this log it looks like:

1. ESPHome first downloads and installs pyparsing 2.3.1 and click 8.1.5
2. ESPHome then uninstalls pyparsing 3.1.0 and click 8.1.3
3. ESPHome then complains about pyparsing 2.3.1 and click 8.1.5 not being the correct versions: these should be pyparsing >=3.0 and click 8.1.3

I don’t understand how this all works in the background, but it looks like something is screwed up here.

The compilation and installation on the ESP device seems to finish normally, but I am still not sure whether this is only a cosmetic problem, or a real problem.
Anybody with more knowledge about this?

ssieb, 2023-07-16:
It's only cosmetic.

kixtarter, 2023-07-16:
> It's only cosmetic.

OK, thanks, that’s reassuring.
Still it would be great when this issue would be resolved, because such an “ERROR:” message in the logs is giving another impression.

nepozs, 2023-07-16:
@ssieb 
> Are you saying it won't compile now? 

Yes, after upgrading from 2023.6.4 to 2023.6.5 it won't compile.
Clean Build Files does not help in this situation. Downgrade and upgrade neither.

The only working way was to completely remove the ESPHome add-on and install the latest version.

Only ESP-IDF projects affected.

Maybe I've seen this error earlier, but it never blocked compilation, and compiled firmware was working OK (I think).

ssieb, 2023-07-16:
What are you calling "latest version"?  And what was the compile error?

nepozs, 2023-07-16:
Latest version was 2023.6.5 

No compilation error, it just has ended with (it is just example, I haven't saved log file…) something like this:
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.6.5 requires click==8.1.3, but you have click 8.1.5 which is incompatible.
esphome 2023.6.5 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
```
and last line was 
`it took xx minutes yy seconds`
but compilation wasn't complete, just every time was stopping the same place

I can't replicate problem now, as I have fresh ESPHome addon installation and it works (errors from current issue exist, buy without blocking compilation).

KitHubek, 2023-07-19:
the same, uptating from 2023.6.x to 2023.7.0 

**Updating device wt32-eth01 - no this errors**
**Updates made after: Clean build files hit**

**Affected**:
ESP32 wifi module
board: esp32dev; framework: espidf; platform: platformio/espressif32@5.3.0
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

[image]

**txt LOG:**

```
  Attempting uninstall: click
    Found existing installation: click 8.1.3
    Uninstalling click-8.1.3:
      Successfully uninstalled click-8.1.3
  Attempting uninstall: pyparsing
    Found existing installation: pyparsing 3.1.0
    Uninstalling pyparsing-3.1.0:
      Successfully uninstalled pyparsing-3.1.0
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.7.0 requires click==8.1.3, but you have click 8.1.6 which is incompatible.
esphome 2023.7.0 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
Successfully installed cachecontrol-0.13.1 certifi-2023.5.7 charset-normalizer-3.2.0 click-8.1.6 colorama-0.4.6 contextlib2-21.6.0 filelock-3.12.2 future-0.18.3 idf-component-manager-1.3.2 idna-3.4 msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0.1 requests-2.31.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.65.0 urllib3-1.26.16
Reading CMake configuration...
Generating assembly for certificate bundle...
```

[2 more replies]