esphome/issues #4519: fatal: detected dubious ownership in repository at '/config/esphome/.esphome/packages/f8744889'

RobSheppard1976, 2023-05-18:
### The problem

After upgrading to the latest version, none of my ESPHome devices won't update to 2023.5.0. They all report this error:

```
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-df5558.yaml...
Failed config

packages: [source /config/esphome/esp32-bluetooth-proxy-df5558.yaml:5]
  
  fatal: detected dubious ownership in repository at '/config/esphome/.esphome/packages/f8744889'
  To add an exception for this directory, call:
  
  	git config --global --add safe.directory /config/esphome/.esphome/packages/f8744889.
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
```

### Which version of ESPHome has the issue?

2023.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.3

### What platform are you using?

ESP32

jesserockz, 2023-05-18:
Are you able to get a shell into the container and paste the result of `ls -al /config/esphome/.esphome/packages/`

RobSheppard1976, 2023-05-18:
Is this what you're looking for?
```
[core-ssh ~]$ ls -al /config/esphome/.esphome/packages/
total 20
drwx------    5 911      911           4096 Mar 15 03:21 .
drwx------    9 911      911           4096 May 17 20:25 ..
drwx------    4 911      911           4096 Apr 27 21:10 0cf47246
drwx------    4 911      911           4096 Apr 27 22:25 a0579e33
drwxr-xr-x    5 911      911           4096 Apr 27 21:05 f8744889
```

jesserockz, 2023-05-18:
Interesting, mine show as:
```
drwxr-xr-x 5 root root 4096 May 18 14:15 f8744889
```

JS876, 2023-05-20:
> Are you able to get a shell into the container and paste the result of `ls -al /config/esphome/.esphome/packages/`

Running into the same issue.
```
[core-ssh ~]$ ls -al /config/esphome/.esphome/packages/
total 16
drwxr-xr-x    4 911      911           4096 Jan 30 23:01 .
drwxr-xr-x    6 911      911           4096 May 19 23:52 ..
drwxr-xr-x    5 911      911           4096 Feb  8 22:40 7b8574f6
drwxr-xr-x    5 911      911           4096 Apr 12 12:27 f8744889
```

cbuxton5200, 2023-05-21:
I run ESPHome in a separate docker container and I am experiencing the exact same problem, but my rights look a lot different:
ls -al
drwxr-xr-x 3 craig root 4096 Jan 23 11:14 .
drwxr-xr-x 6 craig root 4096 May  6 01:19 ..
drwxr-xr-x 5 craig root 4096 Apr 24 20:50 f8744889
craig@host:/opt/esphome/.esphome/packages$

juan11perez, 2023-05-24:
I report the same issue

RobSheppard1976, 2023-05-25:
I'm not sure if it is the *right* thing to do, but I connected in via SSH, and ran the following command:

`chown -R root f8744889/`

Now ESPHome is updating my devices. So far, all have completed the upgrade successfully.

d0ugal, 2023-05-25:
FWIW I fixed this by running the command git suggests in the output in the container. For example

```
docker exec esphome git config --global --add safe.directory /config/esphome/.esphome/packages/f8744889
```

I suspect this will be a temporary fix, but it unblocked me for now. Bear in mind that this assumes the container is called `esphome`, yours may have a different name.

You can confirm the change with
```
$ docker exec esphome git config --global -l
safe.directory=/config/esphome/.esphome/packages/f8744889
```

ssieb, 2023-06-08:
That directory isn't in the container anyway, so a file manager addon should let you fix it or ssh into the base OS.

jtenniswood, 2023-06-09:
Just ran into the same issue, I ran chown -R root on the config file inside the container, it fixed the issue

RobSheppard1976, 2023-06-09:
Anyone know why this would happen? I haven't seen the issue again, even with the latest updates, and I've never messed with anything on the OS level until now. I just wonder what may have messed with the permissions that necessitates running the chown command to 'fix' them.

ptr727, 2023-09-07:
I'd suggest the docker build script executes `git config --system --add safe.directory '*'` such that the git settings when used inside the container ignores path ownership enforcement by git.

randybb, 2023-09-07:
> I'd suggest the docker build script executes `git config --system --add safe.directory '*'` such that the git settings when used inside the container ignores path ownership enforcement by git.
It is already doing https://github.com/esphome/esphome/pull/5102

ptr727, 2023-09-07:
If it has been merged and deployed to docker then it is not working, else it may not have been deployed to release, or maybe it only applies to running in devcontainer not docker?

ptr727, 2023-09-08:
I confirmed there is no git config persisted in the docker image, tested with `2023.8.3`:

```console
docker exec -it esphome /bin/bash
...
root@esphome:/config# git config --list
root@esphome:/config# git config --list --global
fatal: unable to read config file '/root/.gitconfig': No such file or directory
root@esphome:/config# git config --list --system
fatal: unable to read config file '/etc/gitconfig': No such file or directory
```

Filesystem ownership belongs to `users:nobody` and grants all access to all, container runs as `root`.

Tested with YAML that uses:

```yaml
external_components:
  - source: github://tube0013/esphome-stream-server-v2@main
  - source: github://syssi/esphome-zeroconf@main
```

Error during compilation:

```console
INFO ESPHome 2023.8.3
INFO Reading configuration /config/tubezb-cc2652p2-ethusb-2022.yaml...
Failed config

external_components: [source /config/tubezb-cc2652p2-ethusb-2022.yaml:27]
  - 
    fatal: detected dubious ownership in repository at '/config/.esphome/external_components/c6095abb'
    To add an exception for this directory, call:
    
    	git config --global --add safe.directory /config/.esphome/external_components/c6095abb.
    source: github://tube0013/esphome-stream-server-v2@main
  - source: github://syssi/esphome-zeroconf@main
```

Per my understanding of [esphome/pull/5102](https://github.com/esphome/esphome/commit/899aa31df3394720db98cc49e5f920ae12e33284) it only fixes the issue for hte devcontainer, it does not fix the issue for docker.

Adding `RUN git config --system --add safe.directory '*'` to `Dockerfile` will fix the issue.  
Using `--system` will write the setting in `/etc/gitconfig` such that the behavior will be correct for any account that is used the run the container under.

lluisd, 2023-09-29:
Today I ran in the same issue when updating to the new version 2023.9.1

doe81, 2023-12-22:
This is still an issue.
Fixed by running: `docker exec esphome git config --system --add safe.directory '*'`

JohnGalt1717, 2024-05-15:
Still getting this. The defined environment variable user and group has ownership of the folder in question but it still throws this until you run the command above.

closed 2024-06-04 by jesserockz