esphome/issues #4521: Esp8266 Analog Buttons over ADC -> slow/unresponsive

armanivers, 2023-05-18:
### The problem

I'm currently running 4 buttons over a single wire to the adc A0 pin to the esp8266. 
In order to identify the voltages of every button, I had it read and display the current voltage when every button is being pressed, they are fairly constant and it different intervals so I noted them down.

Then my next step was to use a lambda function to determine which button is pressed and invoke a button press.
All worked fine but it's very unresponsive/slow. I'm not sure why. I used to implement this using the MultiAnalogButton library on github.

My guess was the button debounce (delay how long after a button is pressed it can read and activate a function) can't be implemented. Or it's the refresh rate, I'm not very sure, so I implemented the same logic using the analog_threshold option platform.

Here its more cosistent and responsive, though slower, but still better. The problem is that some thresholds aren't working, upon button click nothing happens.

So my question -> Is there a better way to implement this button keypad over adc pin so it's faster whenever I click and it increases the temperature, the display shows it faster there?

### Which version of ESPHome has the issue?

2023.5.1 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.5.3

### What platform are you using?

ESP8266

### Board

ESP8266, (on the back it says NodeMCUv3, I use the nodeMCUv2 in the yaml)

### Component causing the issue

ADC pin / Analog Threshold Binary Sensor

### Example YAML snippet

```yaml
esphome:
  name: buttons
  friendly_name: Buttons

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Gor1dpoYLu80R+daafwOt+WNj8FN+vWNWqRwaQDHQnc="

ota:
  password: "c81deef8548052cebb12a2f2fdb273b8"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Buttons Fallback Hotspot"
    password: "3VD9oTCrALw0"

captive_portal:

i2c:
  scl: D1
  sda: D2

display:
    - platform: lcd_pcf8574
      id: mydisplay
      dimensions: 16x2
      address: 0x27
      update_interval: .1s
      lambda: |-
        it.printf(0,0,"T:%.1f%c%s",id(dht22_temp).state,char(0xDF),"C");
        it.printf(9,0,"H:%.1f%s",id(dht22_hum).state,"%");
        it.printf(0,1,"Set T: %.1f  %s",id(thermostat_controller).target_temperature_low,"OFF");
        

sensor:
  - platform: dht
    pin: D0
    model: DHT22
    temperature:
      name: "Living Room Temperature"
      accuracy_decimals: 1
      id: dht22_temp
    humidity:
      name: "Living Room Humidity"
      accuracy_decimals: 1
      id: dht22_hum
    update_interval: 1s

  - platform: adc
    pin: A0 #A0 or GPIO17 on ESP8266
    id: adc_sensor
    internal: true
    update_interval: .2s
    filters:
      - multiply: 100

  - platform: template
    id: my_button
    internal: true
    lambda: |- 
      if(id(adc_sensor).state > 0 && id(adc_sensor).state < 1){
        id(temp_inc).press();
        return 1;
      } else if (id(adc_sensor).state > 1 && id(adc_sensor).state < 6){
        id(temp_dec).press();
        return 2;
      } else if (id(adc_sensor).state > 6 && id(adc_sensor).state < 15){
        return 3;
      } else if(id(adc_sensor).state > 15 && id(adc_sensor).state < 50){
        id(disp_off).press();
        return 4;
      } else {
        return 0;
      }
    update_interval: .2s

button:
  - platform: template
    name: temperature_increase
    id: temp_inc
    on_press:
      - lambda: |-
              id(thermostat_controller).target_temperature += 0.5;
    
  - platform: template
    name: temperature_decrease
    id: temp_dec
    on_press:
      - lambda: |-
              id(thermostat_controller).target_temperature -= 0.5;

switch:
  - platform: gpio
    pin: D7
    id: relay
    inverted: true
   
climate:
  - platform: thermostat
    id: thermostat_controller
    name: "Thermostat Climate Controller"
    sensor: dht22_temp
    min_heating_off_time: 2s
    min_heating_run_time: 2s
    min_idle_time: 2s
    visual:
      temperature_step: 0.5
    heat_action:
      - switch.turn_on: relay
    idle_action:
      - switch.turn_off: relay
    default_preset: Home
    preset:
      - name: Home
        default_target_temperature_low: 21 °C
```

### Anything in the logs that might be useful for us?

```txt
No log errors
```

armanivers, 2023-05-18:
The Analog Treshold Binary Sensor implementation:

`esphome:
  name: buttons
  friendly_name: Buttons

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Gor1dpoYLu80R+daafwOt+WNj8FN+vWNWqRwaQDHQnc="

ota:
  password: "c81deef8548052cebb12a2f2fdb273b8"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Buttons Fallback Hotspot"
    password: "3VD9oTCrALw0"

captive_portal:

i2c:
  scl: D1
  sda: D2

globals:
  - id: set_temp
    type: double
    restore_value: no
    initial_value: '21.0'  

display:
    - platform: lcd_pcf8574
      id: mydisplay
      dimensions: 16x2
      address: 0x27
      update_interval: .1s
      lambda: |-
        it.printf(0,0,"T:%.1f%c%s",id(dht22_temp).state,char(0xDF),"C");
        it.printf(9,0,"H:%.1f%s",id(dht22_hum).state,"%");
        it.printf(0,1,"Set T: %.1f  %s",id(set_temp),"OFF");
  
sensor:
  - platform: dht
    pin: D0
    model: DHT22
    temperature:
      name: "Living Room Temperature"
      accuracy_decimals: 1
      id: dht22_temp
    humidity:
      name: "Living Room Humidity"
      accuracy_decimals: 1
      id: dht22_hum
    update_interval: 1s

  - platform: adc
    pin: A0 #A0 or GPIO17 on ESP8266
    id: adc_sensor
    update_interval: .1s
    filters:
      - multiply: 100

button:
  - platform: template
    name: display_on
    id: disp_on
    on_press:
      - lambda: |- 
          id(mydisplay).backlight();
      - delay: .1s

  - platform: template
    name: display_off
    id: disp_off
    on_press:
      - lambda: |- 
          id(mydisplay).no_backlight();
      - delay: .1s

binary_sensor:
  - platform: analog_threshold
    name: "button1"
    sensor_id: adc_sensor
    threshold:
      upper: 1
      lower: 0
    filters:
      - delayed_on: 0s
      - delayed_off: 20ms
    on_press:
      - lambda: |-
              id(set_temp) += 0.5;
  - platform: analog_threshold
    name: "button2"
    sensor_id: adc_sensor
    threshold:
      upper: 6
      lower: 1
    filters:
      - delayed_on: 0s
      - delayed_off: 20ms
    on_press:
      - lambda: |-
              id(set_temp) += 0.5;
  - platform: analog_threshold
    name: "button3"
    sensor_id: adc_sensor
    threshold:
      upper: 6
      lower: 15
    filters:
      - delayed_on: 20s
      - delayed_off: 0ms
    on_press:
      - lambda: |-
              id(set_temp) -= 0.5;
  - platform: analog_threshold
    name: "button4"
    sensor_id: adc_sensor
    threshold:
      upper: 15
      lower: 50
    filters:
      - delayed_on: 20s
      - delayed_off: 00ms
    on_press:
      - lambda: |-
              id(set_temp) += 0.5;
  
  

switch:
  - platform: gpio
    pin: D7
    id: relay
    inverted: true

  - platform: gpio
    pin: D6
    id: buzzer
    inverted: true
`

ssieb, 2023-05-18:
github issues are not for support requests.  Come ask on the esphome discord server or one of the forums.

closed 2023-05-18 by ssieb