esphome/issues #4523: ESPHome network traffic not routed through Zerotier add-on

Debug1Bug, 2023-05-19:
### The problem

I have a remote site with some ESPHome devices. I have a Zerotier connection between the ESPHome devices on the remote network (subnet 192.168.2.X) and the home network where Home Assistant is installed (192.168.1.X). Home Assistant has direct access to the devices and can ping them in the SSH terminal, but with one of the latest updates, ESPHome can no longer contact the devices to update firmware etc. There's just no connection.
Please make ESPHome use the Zerotier connection that is installed as an add-on in Home Assistant again. It used to work.

### Which version of ESPHome has the issue?

2023.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.1

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

ESPHome firmware update

### Anything in the logs that might be useful for us?

```txt
Ping to the device 192.168.2.165 works in the Home Assistant SSH terminal, but software update from ESPHome web interface just gives a timeout.
```

### Additional information

This worked up until a few versions ago. I believe I last successfully updated the formware in the devices about a month ago.

ssieb, 2023-05-19:
esphome doesn't do anything specific about networking.  That's all up to the host system and the container.  What does the install log show?  Can you access the devices logs/

Debug1Bug, 2023-05-19:
The device log show up just fine and updates continuously.
The install log just shows (after compile):
========================= [SUCCESS] Took 7.03 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.2.128
ERROR Connecting to 192.168.2.128:8266 failed: timed out

ssieb, 2023-05-19:
If the logs work, then esphome can connect to the device.  This is something else, like a firewall rule or something.

Debug1Bug, 2023-05-19:
I think the logs are somehow received through MQTT and not directly from the device or?

ssieb, 2023-05-19:
They should be direct unless you've configured it for mqtt.  Even then, it still can be direct.

Debug1Bug, 2023-05-19:
Ok. Any ideas for how I can troubleshoot or dig deeper into why the install firmware function in ESPHome has stopped working through Zerotier?

ssieb, 2023-05-19:
This really isn't something for a github issue.  Come to the esphome discord server.  I'll be there for a little bit longer if you join right away.

closed 2023-05-19 by ssieb