esphome/issues #4524: Restore from flash is not working

taqee, 2023-05-19:
### The problem

After updating my devices to 2023.5
All my devices can't get state after power outage
Now 

### Which version of ESPHome has the issue?

2023.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.3

### What platform are you using?

ESP8266

### Board

Nodemcu sonof basic

### Component causing the issue

Restore from flash

nuk3man, 2023-05-24:
Same issue here but with ESP8266 Wemos D1 mini.

I've added:

`
esp8266:
  board: esp01_1m
  restore_from_flash: True
preferences:
  flash_write_interval: 0s`

But it doesn't help. Losing network connection also messes up the state restore.

dukrat, 2023-06-01:
If you skipped 2023.4 you are missing this breaking change https://esphome.io/changelog/2023.4.0.html?highlight=always_off#default-restore-mode-for-switches-fans-and-lights

nuk3man, 2023-06-01:
> If you skipped 2023.4 you are likely affected by this breaking change https://esphome.io/changelog/2023.4.0.html?highlight=always_off#default-restore-mode-for-switches-fans-and-lights

Thanks, I've already seen that. However, with the code I posted above I should be able to circumvent the problem by manually specifying it in the yaml.

In the link it states: "_The default restore_mode for switches, fans and lights has been changed from RESTORE_DEFAULT_OFF to ALWAYS_OFF with the intention **that restoring values from flash should be manually specified by a user if desired in their YAML configuration**._"

ssieb, 2023-06-02:
@nuk3man you didn't post anything about restoring the value.  You do need the `restore_from_flash` option enabled as well, but you also have to specify it for the items you want to save.  I'm closing this because there's no information here and it's almost certainly due to that documented change anyway.

closed 2023-06-02 by ssieb

nuk3man, 2023-06-02:
> @nuk3man you didn't post anything about restoring the value. You do need the `restore_from_flash` option enabled as well, but you also have to specify it for the items you want to save. I'm closing [...]

The restore_from_flash is written in my response above. In any case, here's the full config:

I have a temp sensor that turns on/off lights depending on the temperature. Basically, the sensor doesn't turn on/off the lights after network loss/power loss

		esphome:
		  name: t-2

		esp8266:
		  board: esp01_1m
		  restore_from_flash: True

		preferences:
		  flash_write_interval: 0s

		#--------------------------------------------------------------------------------------------------------------------- SENSORS ---------------------------------------------------------------------------------------------------------------------
		#---------------------------------------- STANDARD SENSOR: TEMP ---------------------------------------- 
		dallas:
		  - pin: GPIO4
			update_interval: 3s
		sensor:
		  - platform: dallas
			address: 0x283c37f649082a28
			name: "t_2"
			unit_of_measurement: "°C"
			on_value_range:
				- above: 8.2
				  then:
					- light.turn_on: 
						id: "red_led_light_t_2"
					- switch.turn_off: "green_led_switch_t_2"
				- below: 8
				  then:
					- light.turn_off:   
						id: "red_led_light_t_2"
					- switch.turn_on: "green_led_switch_t_2"

		#----------------------------------------------------------------------------- SWITCHES & BUTTONS
		#---------------------------------------- LED: switches red/green
		switch:
		  - platform: gpio
			id: "red_led_switch_t_2"
			name: "red led_t_2" 
			pin: GPIO14
			inverted: FALSE
			
		  - platform: gpio
			id: "green_led_switch_t_2"
			name: "green led_t_2" 
			pin: GPIO13
			inverted: FALSE

		#----------------------------------------------------------------------------- LED & LIGHT
		#---------------------------------------- RED LIGHT PWM
		output:
		  - platform: esp8266_pwm
			pin: GPIO14
			frequency: 500 Hz
			inverted: FALSE
			id: red_led_output
		#---------------------------------------- LED: turn on/off
		light:
		  - platform: monochromatic
			output: red_led_output
			name: "Red LED_t_2"
			id: red_led_light_t_2
			effects:
			  - pulse:
				  name: "Pulse"
				  transition_length: 0.5s
				  update_interval: 0.5s
			on_turn_on:
			  - light.turn_on: 
				  id: red_led_light_t_2
				  effect: "Pulse"

		captive_portal:

hiteshkher, 2023-08-31:
Restore from flash is not working 
same problem after update Firmware: 2023.8.2 (Aug 31 2023, 11:51:06)

ssieb, 2023-08-31:
@hiteshkher this issue is closed and there was no problem.  Come ask for help on the esphome discord server.

hiteshkher, 2023-08-31:
Ok , thank you

hiteshkher, 2023-09-07:
> ### The problem
> After updating my devices to 2023.5 All my devices can't get state after power outage Now
>
> ### Which version of ESPHome has the issue?
> 2023.5.1
>
> ### What type of installation are you [...]

> ### The problem
> After updating my devices to 2023.5 All my devices can't get state after power outage Now
>
> ### Which version of ESPHome has the issue?
> 2023.5.1
>
> ### What type of installation are you [...]

Try this , in my case now working 

esp8266:
  board: d1_mini
  restore_from_flash: True

switch:
  - platform: gpio  
    name: light1
    id: light1
    restore_mode: RESTORE_DEFAULT_OFF
    pin: D3