esphome/issues #4525: failed to execute 'open' on 'serialport': failed to open serial port.

osokthedevil, 2023-05-20:
### The problem

failed to execute 'open' on 'serialport': failed to open serial port. getting this on my wemos dp1 mini through the esphome webportal. connecting directly to HA and loading it that way works fine. I had no problem 2 weeks ago. Yes! I tried different cable, drivers, computers, with all the same results. please fix this issue thank you

### Which version of ESPHome has the issue?

https://web.esphome.io/?dashboard_wizard

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.5.3 Supervisor 2023.04.1 Operating System 10.1 Frontend 20230503.3 - latest

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

dashboard wizard

### Additional information

[image]

andi242, 2023-05-23:
same here, same device. 
definitely worked before (about 2 weeks ago? not sure which version)

andi242, 2023-05-23:
Just tested an ESP8266MOD 12-F and got the same error as on the D1
I am using Edge `v113.0.1774.50 (64-Bit)`, cleared all temp files, cookies etc.

I got another ESP32 WROOM 32D at hand which seems to be working fine.
The difference between those three is that the WROOM has never seen esphome before freshly taken out of the package.
Not sure if this is relevant.

nsf1977, 2023-05-26:
Same problem here with all esp8266. Either using esphome webportal or esphome flasher. No problem connecting to esp32

D8mbSniper, 2023-05-28:
Same problem with new Wesmos D1 mini ESP-12F

IT-Gipsy, 2023-05-29:
I am an absolute Newbie for ESPHome, but I am also getting the same problem with ESP8266, connected to Win11
Using either Google Chrome or EDGE, I am using a SECURE connection (through Nabu Casa) to avoid that obstacle also

brandon-daniels, 2023-05-30:
I am getting the same error with the nodemcu esp32s.

andi242, 2023-06-01:
> So must be a Windows 11 thing

tested with Fedora Linux/Chromium and still no gain.

Porky550, 2023-06-02:
Hello, i am seeing the same problem, Windows 10 and tested on 5 devices as well as RP4 
Failed to execute 'open' on 'SerialPort': Failed to open serial port.
it was all working fine last week. any ideas on why or sis this something broken in the last update 2023.5.5

remcoso, 2023-06-02:
Same issue

IT-Gipsy, 2023-06-02:
I have been able to "solve" the problem by using another (Intel NUC) pc that I only use to wath the Formula 1 (running Win10 pro). 
On the PC, the connection (via Home Assistant HTTPS access) is running without issues.

I do have another question.
I am now using a CH340 esp8266 board (with all pins sticking out)
I would like to know if I can just use a D1 chip and a temporary programmer (USB) holder.
Once programmed it will be accessible over Wifi and has a "fall back" SSID.

The "D1" is much smaller and I could integrate these in for example my Panasonic HeatPump boiler
Or hide away in a socket with a small relay.

I know there are "ready made" relay solutions and RS-485 solutions or GPI-in solutions, but I would like to develop things myself and integrate them (as small as possible)

Thanks...

osokthedevil, 2023-06-02:
> I have been able to "solve" the problem by using another (Intel NUC) pc that I only use to watch the Formula 1 (running Win10 pro). On the PC, the connection (via Home Assistant HTTPS access) is running without issues.
> 
> I do have another question. I am now using a CH340 esp8266 board (with all pins sticking out) I would like to know if I can just use a D1 chip and a temporary programmer (USB) holder. Once programmed it will be accessible over Wifi and has a "fall back" SSID.
> 
> The "D1" is much smaller and I could integrate these in for example my Panasonic HeatPump boiler Or hide away in a socket with a small relay.
> 
> I know there are "ready made" relay solutions and RS-485 solutions or GPI-in solutions, but I would like to develop things myself and integrate them (as small as possible)
> 
> Thanks...

Wrong place for that question ⁉️, ask that in the discord of Facebook.

Jooel1023, 2023-06-03:
Same problem with D1

Porky550, 2023-06-03:
So, on my w10 server it works. But on other pc that are used a lot it seems to be drivers or something using it but what. So after removing ide and uninstalling every thing that uses serial I'm still hitting a wall. But the frog board connects with usb to UART. But not esp8266  any one have any luck or a solution

Porky550, 2023-06-04:
Ok sounds silly but, this how i fixed it. 
1. device manager
2. go to the device connected comport 
3. uninstall drivers tick the box in the popup
4. disconnect the esp and reconnect 
5. esphome and connect to the comport of esp 

and this worked, now can not be that simple ill try on the other pc

compumark, 2023-06-04:
> Ok sounds silly but, this how i fixed it.
> 
> 1. device manager
> 2. go to the device connected comport
> 3. uninstall drivers tick the box in the popup
> 4. disconnect the esp and reconnect
> 5. esphome and connect to the comport of esp
> 
> and this worked, now can not be that simple ill try on the other pc

wow, genius. I just tried that on my Win11 machine, as confirmed it working.

andi242, 2023-06-04:
> this how i fixed it

can confirm this for the D1 & other ESP8266 devices I was able to test. thanks!

not sure why this did not work on my linux machine, though.

Porky550, 2023-06-04:
> > this how i fixed it
> 
> can confirm this for the D1 & other ESP8266 devices I was able to test. thanks!
> 
> not sure why this did not work on my Linux machine, though.

yes this is for the D1 Mini, not shore about Linux cant get the hang of it still after working with it for 20 years i and an old fluffy windows guy :)  how ever cant see why it wouldn't work for eap 8266 don't they use the same drivers???

Uritec, 2023-06-06:
Same problem with D1 mini on Windows 11. Driver deainstallation dosen't help

ktm1065, 2023-06-06:
I downloaded other driver to here -> https://k2me.tistory.com/145

and I can install !!! T.T

Phadeguy, 2023-06-12:
Here is how I fixed it in Windows 11 with the D1 Mini plugged in:

1. Download the driver listed above and unzip: [https://k2me.tistory.com/145](https://k2me.tistory.com/145)
2. Open "Device Manager"
3. Open "Ports (COM & LPT)"
4. Right-click the troubled device (ex. "USB-SERIAL CH340 (COM10)") and click "Update Driver"
5. Click "Browse my computer for drivers"
6.  Click "Let me pick from a list of available drivers on my computer"
7. Click "Have Disk..."
8. Select the unzip folder from Step 1 into the "Copy manufacturer's files from:" and click "OK"
9. Select the model (ex. "USB-SERIAL CH340") and click "Next"
10. Click "Close"

After that, Tasmota could be successfully installed on the target device.

Andersama, 2023-06-12:
Didn't "fix" my issue, I ended up trying the docker install but on my windows 10 machine which worked fine. I was using vivaldi as a chrome based browser. In short, try different operating systems, update or reinstall drivers, and potentially try different browsers.

If someone's particularly motivated I don't know if there's additional error data about how the serial port is failing to open, but if there isn't, might be possible to add it somewhere around here: https://github.com/chromium/chromium/blob/main/services/device/serial/serial_io_handler_win.cc
https://github.com/chromium/chromium/blob/main/services/device/serial/serial_io_handler_posix.cc
Looks like there should be some form of logging happening, haven't touched chrome development in ages can't remember if there's some verbose command line flag mode or if it's a thing in the inspect console or if that's what opens those prompts with the error messages. Best of luck sorting this out, especially if it's drivers.

Jooel1023, 2023-06-12:
> Here is how I fixed it in Windows 11 with the D1 Mini plugged in:
>
> 1. Download the driver listed above and unzip: https://k2me.tistory.com/145
> 2. Open "Device Manager"
> 3. Open "Ports (COM & LPT)"
> 4. [...]

I was hopless but this chinese website save all my D1 mini lol. Thanks

Gamalta, 2023-06-13:
> Here is how I fixed it in Windows 11 with the D1 Mini plugged in:
>
> 1. Download the driver listed above and unzip: https://k2me.tistory.com/145
> 2. Open "Device Manager"
> 3. Open "Ports (COM & LPT)"
> 4. [...]

Thanksss !

Phadeguy, 2023-06-18:
> I was hopless but this chinese website save all my D1 mini lol. Thanks

They're Korean, so not that bad...  ;-)

themarcraft, 2023-06-21:
> Here is how I fixed it in Windows 11 with the D1 Mini plugged in:
>
> 1. Download the driver listed above and unzip: https://k2me.tistory.com/145
> 2. Open "Device Manager"
> 3. Open "Ports (COM & LPT)"
> 4. [...]

It actually helped me! Thanks

brendangooden, 2023-07-08:
For me, the issue appeared to be with the windows update installed for `wch.cn - Ports - 3.8.2023.2` 

https://www.catalog.update.microsoft.com/ScopedViewInline.aspx?updateid=6faa33ef-5b78-469c-abac-10f990667c5f 

[CH341SER.ZIP (CH340 wch.cn version 3.7.2022.1) ](https://github.com/esphome/issues/files/11988825/CH341SER.ZIP)

I ended up re-installing an old version of the driver (3.7.2022.1 - see above), and it started working again fine. Not sure if this is an issue that needs to be looked into or where to fix it?

nerdybobo, 2023-07-09:
> It actually helped me! Thanks

me too

davc1, 2023-07-15:
> I downloaded other driver to here -> https://k2me.tistory.com/145
> 
> and I can install !!! T.T

I had the same issue, and this driver worked for me too (Windows 11), thanks!

mpelektro, 2023-07-19:
> Here is how I fixed it in Windows 11 with the D1 Mini plugged in:
>
> 1. Download the driver listed above and unzip: https://k2me.tistory.com/145
> 2. Open "Device Manager"
> 3. Open "Ports (COM & LPT)"
> 4. [...]

Confirmed, it works on windows 11

Goddur, 2023-07-22:
After reviewing all that has been posted I am reporting these suggestions did not work for me. I am using the same steps as the OP and getting fails. My setup is Windows 11 and all drivers have been updated along with a new cable and 5 amp hub all based on suggestions in forums and other sites suggesting solutions. My fingers are crossed that whoever is programming the serial port drivers sees these posts and fixes the problem as it seems that is the root of the failures. BTW the devices that are having these fails are the ESP32-CAM board and its accompanying ESP32-CAM motherboard. Interestingly enough the fails are fewer when I am attempting an install on ESP8266 products. This OP was in May really surprised nothing has been done to correct this failure.

Goddur, 2023-07-22:
> 

Beings nothing else worked to alleviate the problem your post is an way to go. Fingers crossed I don't hose my entire system. Will report back if this works for me.

Goddur, 2023-07-24:
A shift towards allowing links to videos for error reporting could significantly enhance the effectiveness of error analysis. Video format enables the conveyance of a wealth of information beyond what written text can provide. By incorporating videos, readers can better understand the context, observe real-time interactions, and gain deeper insights into the issues at hand. This multimedia approach would lead to more comprehensive and precise troubleshooting, ultimately benefiting the entire error-reporting process.

kinghat, 2023-08-06:
for those on linux via @ssieb on discord:
> Add your user to the `dialout` group.  (I use Fedora as well.)
You have to login again to activate the group change.

vkkeeper, 2023-08-15:
Probably just a permissions issue.
Using web installation from web.esphome.io under Windows try to start your browser as Administrator.
Under Linux - yes add yourself into "dialout" group.

DanielHabenicht, 2023-08-16:
Indeed seems to be a bad driver. I was lucky and there was an older version still available on my device: 
[image]
Version 3.5.2019.1 seems to work fine.

frekimed, 2023-08-26:
I used the above steps to manually install this driver through Device Manager: https://www.wemos.cc/en/latest/ch340_driver.html 

That fixed my issue. Also on Windows 11 with a Wemos D1 mini clone.

jeremyhagan, 2023-09-10:
> Ok sounds silly but, this how i fixed it.
> 
> 1. device manager
> 2. go to the device connected comport
> 3. uninstall drivers tick the box in the popup
> 4. disconnect the esp and reconnect
> 5. esphome and connect to the comport of esp
> 
> and this worked, now can not be that simple ill try on the other pc

This worked for me too, but mainly because it caused me to reboot. My real problem was that the Arduino IDE was open in the background. Closing it released the com port.

bkbartk, 2023-09-23:
> I used the above steps to manually install this driver through Device Manager: https://www.wemos.cc/en/latest/ch340_driver.html
> 
> That fixed my issue. Also on Windows 11 with a Wemos D1 mini clone.

So a windows driver issue then :S
you saved my day, I was already frustrated for hours after an OTA update bricked the device

dgoemans, 2023-09-29:
Driver also worked for me using a AZ Delivery D1 Mini v3 NodeMCU ESP8266-12F WLAN

lmatter, 2023-11-01:
> I downloaded other driver to here -> https://k2me.tistory.com/145
> 
> and I can install !!! T.T

This worked!  I had to click on UNINSTALL first, then INSTALL and FINALLY It's working..  THanks!

vazquezjm, 2023-11-03:
Same here. Had to roll back to v3.5 driver (Win11)

deep-barot, 2023-11-12:
> Here is how I fixed it in Windows 11 with the D1 Mini plugged in:
>
> 1. Download the driver listed above and unzip: https://k2me.tistory.com/145
> 2. Open "Device Manager"
> 3. Open "Ports (COM & LPT)"
> 4. [...]

This worked for me on windows 11 after so much frustration.

chvvkumar, 2023-11-30:
I had to use these drivers on my Win11 box. None of the other drivers mentioned in this thread worked for me:

https://github.com/nodemcu/nodemcu-devkit/blob/master/Drivers/CH341SER_WINDOWS.zip

Grey-Lancaster, 2023-12-04:
Fixed it !!!
Indeed seems to be a bad driver. I was lucky and there was an older version still available on my device:
image
Version 3.5.2019.1 seems to work fine.

hiteshkher, 2023-12-11:
> I downloaded other driver to here -> https://k2me.tistory.com/145
> 
> and I can install !!! T.T

working

Kal318, 2023-12-23:
> Ok sounds silly but, this how i fixed it.
> 
> 1. device manager
> 2. go to the device connected comport
> 3. uninstall drivers tick the box in the popup
> 4. disconnect the esp and reconnect
> 5. esphome and connect to the comport of esp
> 
> and this worked, now can not be that simple ill try on the other pc

Actually all I had to do was follow the first 2 steps. Right click on the USB serial port > properties > port setting > and change the bits per second from 9600 to 115200.
That solved it for me.

aaron-yang0327-development, 2023-12-24:
apparently drivers can get "infected" so if it used to work, reinstall the drivers of the device, reinstall as in download the drivers from the website. I have a CH340 and they had a driver installer along with uninstaller so it was pretty easy to fix mine

wolfmanvid01, 2023-12-30:
Fixed it on linux

"sudo chmod a+rw /dev/[serial port name of the esp32]"

now try again

DIY-Sensors, 2024-01-02:
[image]
I think I have found the problem for me. Soms of my D1 mini's are different from the other ones. The D1 mini's on the right, no problem and direct flashing, the D1 mini's on the left not to flash an errors like "Could not open port". Tried all the different drivers for the left ones, no result. Richt ones no problem at all.

Hope one of you has had the same problem and the same result.

randybb, 2024-01-02:
> Fixed it on linux
> 
> "sudo chmod a+rw /dev/[serial port name of the esp32]"
> 
> now try again
> 
> U have to do that everytime u connect the esp32

This is the best way how to do it. Just add your user to dialout group `sudo adduser $USER dialout`, logout, login and done.

bkbartk, 2024-01-04:
@DIY-Sensors 
that might as well be  a difference between CH9102 and CP2104

ertivo, 2024-01-04:
> Actually all I had to do was follow the first 2 steps. Right click on the USB serial port > properties > port setting > and change the bits per second from 9600 to 115200. That solved it for me.

This change from 9600 to 115200 helped me,
Thanks

seagullki, 2024-01-06:
Try another USB-Cable. Some Cable has no data wires!

Grey-Lancaster, 2024-01-07:
If it was a charging cable it would not show as a port and we would not get to the error msg

Finale0171, 2024-01-18:
> Fixed it on linux
> 
> "sudo chmod a+rw /dev/[serial port name of the esp32]"
> 
> now try again
> 
> U have to do that everytime u connect the esp32

Worked perfectly for me on Fedora 39. Thanks! Adding myself to dialout (and logging out/in/rebooting) did not work.

ModuloFS, 2024-03-05:
Additional driver location of the 3.5 Version for Windows if anybody is interested: https://www.shotech.de/de/ch340g.html

drafter0815, 2024-03-06:
Hello,
if issue happens with docker. You have to put the environment variable USBDEVICES:

 devices:
  - /dev/ttyUSB0:/dev/ttyUSB0
 environment:
   USBDEVICES: /dev/ttyUSB0

This worked for me.

Best Regards
Markus

jpmorrison, 2024-04-07:
A direct link to the binaries and some esptool instructions would be nice. Webupload is nice if it works, which is about 50/50. More reliable on fresh devices?

Trying to switch from OMG 1.7 to ESP32 bluetooth proxy. Failures (press boot button) or serial port in use. Windows and Linux. 
Ran esptool.py (Linux) and erase_flash/flash_id works.  Waiting for install....

This is all weird because I used the web uploader with OMG to get it programmed in the first place. Different versions of the webserial? Also seeing Chrome works sometimes, sometimes Edge.

jpmorrison, 2024-04-07:
> A direct link to the binaries and some esptool instructions would be nice. Webupload is nice if it works, which is about 50/50. More reliable on fresh devices?
>
> Trying to switch from OMG 1.7 to [...]

Gave up waiting for ESP home web upload

Checked with minicom 

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
␀ets Jul 29 2019 12:21:46

``

I checked again with --dtr 1 and it went into boot loader

`esptool write_flash 0 esphome-web-esp32.bin`  works but left /dev/ttyUSB1 in a weird state. Toggling DTR in minicom seems to show esphome was installed? 

Hardware is ESP32. It says Wemos D1 R32 but it might be a knockoff. It's an Aruino Uno format PCB.
Might just be quirky hardware doesn't always work with the web uploader?

```
␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀--- DTR active ---
␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
␛[0;32m[I][logger:158]: Log initialized␛[0m
␛[0;35m[C][ota:483]: There have been 1 suspected unsuccessful boot attempts.␛[0m
␛[0;36m[D][esp32.preferences:114]: Saving 1 preferences to flash...␛[0m
␛[0;36m[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed␛[0m
␛[0;32m[I][app:029]: Running through setup()...␛[0m
␛[0;35m[C][esp32_ble:027]: Setting up BLE...␛[0m
␛[0;35m[C][wifi:038]: Setting up WiFi...␛[0m
␛[0;35m[C][wifi:051]: Starting WiFi...␛[0m
␛[0;35m[C][wifi:052]:   Local MAC: A0:B7:65:61:76:D4␛[0m
␛[0;35m[C][wifi:253]: Setting up AP...␛[0m
␛[0;35m[C][wifi:255]:   AP SSID: 'esphome-web-6176d4'␛[0m
␛[0;35m[C][wifi:256]:   AP Password: ''␛[0m
␛[0;35m[C][wifi:265]:   IP Address: 192.168.4.1␛[0m
␛[0;36m[D][wifi:462]: Starting scan...␛[0m
␛[0;36m[D][esp32_improv.component:244]: Setting Improv to start␛[0m
␛[0;35m[C][web_server:124]: Setting up web server...␛[0m
␛[0;35m[C][ota:096]: Over-The-Air Updates:␛[0m
␛[0;35m[C][ota:097]:   Address: esphome-web.local:3232␛[0m
␛[0;35m[C][ota:103]:   OTA version: 2.␛[0m
␛[0;35m[C][api:025]: Setting up Home Assistant API server...␛[0m
␛[0;36m[D][wifi:462]: Starting scan...␛[0m
␛[0;32m[I][app:062]: setup() finished successfully!␛[0m
␛[0;36m[D][esp32_ble:270]: Enabling BLE...␛[0m
␛[0;33m[W][component:232]: Component esp32_ble took a long time for an operation (626 ms).␛[0m
␛[0;33m[W][component:233]: Components should block for at most 30 ms.␛[0m
␛[0;33m[W][api:152]: Warning set: unspecified␛[0m
␛[0;32m[I][app:102]: ESPHome version 2024.3.0 compiled on Mar 26 2024, 07:25:13␛[0m
␛[0;32m[I][app:104]: Project esphome.web version 1.0␛[0m
␛[0;35m[C][logger:166]: Logger:␛[0m
␛[0;35m[C][logger:167]:   Level: DEBUG␛[0m
␛[0;35m[C][logger:169]:   Log Baud Rate: 115200␛[0m
␛[0;35m[C][logger:170]:   Hardware UART: UART0␛[0m
␛[0;35m[C][esp32_ble:374]: ESP32 BLE:␛[0m
␛[0;35m[C][esp32_ble:376]:   MAC address: A0:B7:65:61:76:D6␛[0m
␛[0;35m[C][esp32_ble:377]:   IO Capability: none␛[0m
␛[0;35m[C][esp32_ble_server:200]: ESP32 BLE Server:␛[0m
␛[0;35m[C][esp32_improv.component:261]: ESP32 Improv:␛[0m
␛[0;36m[D][esp32_ble_server:077]: BLE server setup successfully␛[0m
␛[0;36m[D][esp32_improv.component:077]: Creating Improv service␛[0m
␛[0;35m[C][captive_portal:088]: Captive Portal:␛[0m
␛[0;35m[C][wifi:580]: WiFi:␛[0m
␛[0;35m[C][wifi:408]:   Local MAC: A0:B7:65:61:76:D4␛[0m
␛[0;35m[C][wifi:413]:   SSID: ␛[5m''␛[6m␛[0m
␛[0;35m[C][wifi:420]:   BSSID: ␛[5m00:00:00:00:00:00␛[6m␛[0m
␛[0;35m[C][wifi:421]:   Hostname: 'esphome-web-6176d4'␛[0m
␛[0;35m[C][wifi:423]:   Signal strength: 0 dB ␛[0;32m▂▄▆█␛[0m␛[0m
␛[0;35m[C][wifi:427]:   Channel: 6␛[0m
␛[0;35m[C][wifi:428]:   Subnet: 0.0.0.0␛[0m
␛[0;35m[C][wifi:429]:   Gateway: 0.0.0.0␛[0m
␛[0;35m[C][wifi:430]:   DNS1: 0.0.0.0␛[0m
␛[0;35m[C][wifi:431]:   DNS2: 0.0.0.0␛[0m
␛[0;35m[C][web_server:169]: Web Server:␛[0m
␛[0;35m[C][web_server:170]:   Address: esphome-web.local:80␛[0m
␛[0;35m[C][mdns:115]: mDNS:␛[0m
␛[0;35m[C][mdns:116]:   Hostname: esphome-web-6176d4␛[0m
␛[0;35m[C][ota:096]: Over-The-Air Updates:␛[0m
␛[0;35m[C][ota:097]:   Address: esphome-web.local:3232␛[0m
␛[0;35m[C][ota:103]:   OTA version: 2.␛[0m
␛[0;35m[C][api:139]: API Server:␛[0m
␛[0;35m[C][api:140]:   Address: esphome-web.local:6053␛[0m
␛[0;35m[C][api:144]:   Using noise encryption: NO␛[0m
␛[0;35m[C][improv_serial:032]: Improv Serial:␛[0m
␛[0;36m[D][esp32_improv.component:097]: Service started!␛[0m

`

JMMYC, 2024-04-08:
I used LinuxMInt and had this same problem.
The solution - in my case - was as simple as adding my user to the "dialout" group and resetting the computer.
I found this solution at:
https://stackoverflow.com/questions/63112645/webserial-api-device-cannot-open-port-on-ubuntu

[22 more replies]