esphome/issues #4526: Unable to read data from device meter Chint ddsu666

vinceg0267, 2023-05-20:
### The problem

Sensors not available in home assistant

### Which version of ESPHome has the issue?

2023.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.3

### What platform are you using?

ESP8266

### Board

Wemos D1 mini v.3

### Example YAML snippet

```yaml
>  esphome:
>   name: chint
>   friendly_name: Chint Meter
> esp8266:
>   board: esp01_1m
> # Enable logging
> logger:
>   baud_rate: 9600
> # Enable Home Assistant API
> api:
>   encryption:
>     key: "***"
> ota:
>   password: "***"
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
>   # Enable fallback hotspot (captive portal) in case wifi connection fails
>   ap:
>     ssid: "Wemos Fallback Hotspot"
>     password: "***"
> 
> captive_portal:
> external_components:
>   - source:
>       type: git
>       url: https://github.com/esphome/esphome
>       ref: dev
>     components: [ modbus, modbus_controller ]
> 
> uart:
>   id: mod_bus
>   tx_pin: 0
>   rx_pin: 0
>   baud_rate: 9600
>   stop_bits: 2
> 
> modbus:
>   id: modbus1
>   send_wait_time: 2000ms
> 
> modbus_controller:
>   - id: epever
> 
>     address: 0x24
>     modbus_id: modbus1
>     setup_priority: -10
>     update_interval: 10s
> 
> sensor:
>   - platform: modbus_controller
>     modbus_controller_id: epever
>     id: power_meter_voltage
>     name: "Voltage"
>     address: 0x2000
>     unit_of_measurement: "V"
>     register_type: holding
>     skip_updates: 30
>     value_type: S_DWORD
>     accuracy_decimals: 1
>     lambda: |-
>       union My_un {
>         char temp[4];
>         float foo;
>         } un;
>       un.temp[0] = data[3];
>       un.temp[1] = data[2];
>       un.temp[2] = data[1];
>       un.temp[3] = data[0];
>       ESP_LOGI("", "foo=%f , 0x%2x%2x%2x%2x%" ,un.foo,un.temp[0],un.temp[1],un.temp[2],un.temp[3])
>       ; return un.foo ;
> 
>   - platform: modbus_controller
>     modbus_controller_id: epever
>     id: power_meter_power
>     name: "Power"
>     address: 0x2004
>     unit_of_measurement: "W"
>     register_type: holding
>     value_type: S_DWORD
>     accuracy_decimals: 1
>     lambda: |-
>       union My_un {
>         char temp[4];
>         float foo;
>         } un;
>       un.temp[0] = data[3];
>       un.temp[1] = data[2];
>       un.temp[2] = data[1];
>       un.temp[3] = data[0];
>       ESP_LOGI("", "foo=%f , 0x%2x%2x%2x%2x%" ,un.foo,un.temp[0],un.temp[1],un.temp[2],un.temp[3])
>       ; return un.foo ;
>     filters:
>       - multiply: 1000
```

### Anything in the logs that might be useful for us?

```txt
> [17:56:07][W][modbus_controller:112]: Duplicate modbus command found: type=0x3 address=8196 count=2
> [17:56:07][D][modbus_controller:029]: Modbus command to device=36 register=0x2004 countdown=0 no response received - removed from send queue
> [17:56:27][W][modbus_controller:112]: Duplicate modbus command found: type=0x3 address=8196 count=2
> [17:56:27][D][modbus_controller:029]: Modbus command to device=36 register=0x2004 countdown=0 no response received - removed from send queue
```

andreiciocanvlad, 2023-07-04:
Aren't you supposed to add the pins here 
`> uart:
>   id: mod_bus
>   tx_pin: 0
>   rx_pin: 0
>   baud_rate: 9600
>   stop_bits: 2`

NikoKS, 2024-02-26:
is this solved? @vinceg0267

vinceg0267, 2024-03-18:
Solved

closed 2024-03-18 by vinceg0267