esphome/issues #4528: ESPHome 2023.5.1 update failed with bluetooth-proxies/m5stack-atom-lite

timoverwoest, 2023-05-21:
### The problem

Tried to update ESPHome firmware on m5stack-atom-lite bluetooth-proxy and it failed. Initially installed through https://esphome.github.io/bluetooth-proxies/

See log: 
Dashboard import requires the `esphome` -> `project` information to be provided.
  package_import_url: github://esphome/bluetooth-proxies/m5stack-atom-lite.yaml@main
  import_full_config: False

[logs_bluetooth-proxy_run.txt](https://github.com/esphome/issues/files/11523928/logs_bluetooth-proxy_run.txt)

### Which version of ESPHome has the issue?

2023.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.3

### What platform are you using?

ESP32

### Board

m5stack-atom

### Component causing the issue

dashboard_import:

### Example YAML snippet

```yaml
esphome:
  name: "bluetooth-proxy"
  friendly_name: "bluetooth-proxy"

esp32:
  board: m5stack-atom
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: [redacted]

ota:
  password: [redacted]

improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "bluetooth-proxy Fallback Hotspot"
    password: [redacted]

dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/m5stack-atom-lite.yaml@main

esp32_ble_tracker:
  scan_parameters:
    # We currently use the defaults to ensure Bluetooth
    # can co-exist with WiFi In the future we may be able to
    # enable the built-in coexistence logic in ESP-IDF
    active: true

bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic
```

### Anything in the logs that might be useful for us?

```txt
Dashboard import requires the `esphome` -> `project` information to be provided.
  package_import_url: github://esphome/bluetooth-proxies/m5stack-atom-lite.yaml@main
  import_full_config: False
```

### Additional information

[image]

Write, 2023-05-22:
Same issue with an esp32 too. Always upgraded fine and out of nowhere I have this error.

ssieb, 2023-05-22:
You either have to add the `project:` information as it says (see https://esphome.io/components/esphome.html#configuration-variables) or remove the unnecessary `dashboard_import:` section.

closed 2023-05-22 by ssieb

ssieb, 2023-05-22:
The source yaml has been updated, but that doesn't affect already created configs.
https://github.com/esphome/bluetooth-proxies/blob/main/m5stack-atom-lite.yaml

Write, 2023-05-22:
> The source yaml has been updated, but that doesn't affect already created configs.
> 
> https://github.com/esphome/bluetooth-proxies/blob/main/m5stack-atom-lite.yaml

well it definitely affected me

ssieb, 2023-05-22:
I mean that any changes to the source yaml wouldn't affect any configs that were already created.
However, I've also been informed that the `project:` section has always been in the source yaml, so how did you create your config without it?

Write, 2023-05-22:
> I mean that any changes to the source yaml wouldn't affect any configs that were already created. However, I've also been informed that the `project:` section has always been in the source yaml, so how did you create your config without it?

Absolutely no idea. I bought my ESP, I've done the setup wirelessly ever since after that, and since one or two versions of ESPHome HA Addons it throw this error. I never had project: inside the config, and it always upgraded just fine. Now it throw an error about this dashboard_import.

kazdegroot, 2023-05-22:
Personally, I modified some existing configs that were previously used another bluetooth proxy setup, to work with this method. The `project:` section seemed like boilerplate for a example config, not as an essential part of the config, and afaik it worked fine without it. 

Even now, the [relevant section](https://esphome.io/components/esphome.html#project-information) seems to suggest it's for 'creators' working on 'projects', which made me think I was in the wrong section even after following the error.

ssieb, 2023-05-22:
Remove the `dashboard_import:` section.  That's only useful for OEM firmware, not for your own personal use.
There was a recent change that added the requirement for the `project:` info if you have that section.

Write, 2023-05-22:
> Remove the `dashboard_import:` section. That's only useful for OEM firmware, not for your own personal use. There was a recent change that added the requirement for the `project:` info if you have that section.

Yes indeed, it worked without dashboard_import, it upgraded my esp no error. I still don't have any "project:" line but everything work.