esphome/issues #4529: after 2023.5.1 there are no more data readings for sonoff R3 pow origin

calas80, 2023-05-22:
### The problem

I have been using this code for a long time and it has always worked. After the update, I no longer receive current, voltage and power data. 

### Which version of ESPHome has the issue?

2023.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.3

### What platform are you using?

ESP32

### Board

Sonoff POW ORIGIN (R3)

### Example YAML snippet

```yaml
uart:
  rx_pin: GPIO16
  baud_rate: 4800

################################################
#        Sensor
#########################
sensor:
  - platform: uptime
    name: "${esp_devicename} Uptime"

  - platform: cse7766
    update_interval: 5s
    current:
      name: "${esp_devicename} Corrente"
    voltage:
      name: "${esp_devicename} Voltaggio"
    power:
      name: "${esp_devicename} Power"
      id: power
    energy:
      name: "${esp_devicename} Energy"
      id: energy
      icon: mdi:chart-line
```

branda92, 2023-05-23:
I too have the usual problem, which continues to be present even with 2023.5.2

itssimon, 2023-05-28:
Same problem here with a Sonoff POW Elite

ModuCZ, 2023-05-28:
Same here

All of my Sonoff POW Origin (esp32) stopped working after update. Platform cse7766 doesn't provide any readings.
Suprisingly Sonoff POW R2 (old version of ORIGIN) is still working (althought it's cse7766 too).

ModuCZ, 2023-05-28:
ESPHome 2023.5.4 didn't fix this issue for me.

ModuCZ, 2023-05-28:
This issue is present since ESPHome 2023.5.0 - tested by downgrading .

Temporarily fixed by downgrading to ESPHome 2023.4.4 and updating all of my devices without code change.

mspiridakis, 2023-06-03:
Same here with sonoff powr320d

CarlosGS, 2023-06-03:
Hi, in the example from the first post there is GPIO16 (!!) for UART.
Just to confirm, can you try this configuration:
```yaml
logger:
  baud_rate: 0 # Disable serial logger to prevent conflict

uart:
  rx_pin: RX # cse7766 is connected to regular serial port
  baud_rate: 4800
```
And for cse7766 make sure `update_interval:` is always greater than 3 seconds.

CarlosGS, 2023-06-03:
The only related change I can see is https://github.com/esphome/esphome/pull/4762

Maybe the problem is that the UART is not being referenced with an ID, and that PR caused it to fall back to the standard UART port (?)

Can you try assigning a specific ID to the uart? like this:
```yaml
uart:
  rx_pin: GPIO16
  baud_rate: 4800
  id: pm_uart # Assign ID to UART for power meter

################################################
#        Sensor
#########################
sensor:
  - platform: uptime
    name: "${esp_devicename} Uptime"

  - platform: cse7766
    uart_id: pm_uart # Select power meter's UART
    update_interval: 5s
    current:
      name: "${esp_devicename} Corrente"
    voltage:
      name: "${esp_devicename} Voltaggio"
    power:
      name: "${esp_devicename} Power"
      id: power
    energy:
      name: "${esp_devicename} Energy"
      id: energy
      icon: mdi:chart-line
```

Also, since the change only was implemented for Arduino, you can try using esp-idf platform and see if the issue is present.
I don't have a newer Pow R3 (ESP32) so can't help much more.

mspiridakis, 2023-06-03:
Thanx for the time and help @CarlosGS.
First suggestion was a good bet, but unfortunately it didn't work.
The second one needs a lot of changes in my yaml file since I have a lot of bells and whistles there.
Do you think that there is going to be a fix sometime soon?

calas80, 2023-06-05:
anyone have any news? it is really disastrous not to receive the data

mspiridakis, 2023-06-05:
> anyone have any news? it is really disastrous not to receive the data

Try to use the workaround above.  In the meantime lets hope @spectrumjade will have a look at it again.

calas80, 2023-06-06:
> 

I didn't see your previous reply. Thank you

mspiridakis, 2023-06-06:
Just found out that it also works with a simpler manner commenting out the "baud_rate" line amd without using ids, so:
```
logger:
  level: DEBUG
  #  baud_rate: 0
uart:
  - rx_pin: GPIO16
    baud_rate: 4800

sensor:
  - platform: cse7766
```

ModuCZ, 2023-07-10:
Issue still present in ESPHome 2023.6.5.

I'm giving up. Updating code on all my devices with `#baud_rate: 0` as this fixed issue. Thank you **mspiridakis**!

jongdesteven, 2023-08-10:
I believe it has to do with following PR #4762

paulignari, 2023-08-25:
My new POWR320D was working with Tasmota but with ESPHome 2023.8.2 it wasn't able to read and display any power-related value (A, V, W, ecc.).

After a lot of tries I found that Tasmota configure the UART at 4800 baud 8E1, while ESPHOME use 4800 baud 8N1.

After setting

```
uart:
  rx_pin: GPIO16
  baud_rate: 4800
  data_bits: 8
  parity: EVEN
  stop_bits: 1
```

the values are read from CSE7766, despite this warning:

`[E][uart:029]:   Invalid parity: Integration requested parity NONE but you have EVEN!`

I hope this ca be useful.

jongdesteven, 2023-08-26:
I see the datasheet also describes the even parity bit. So this should be correct.

paulalexandrow, 2023-11-22:
I can confirm this also happens on the Athom v3 EU Plug (ESP32 C3 based), where @mspiridakis's fix using a dummy uart works (thanks!).
However, the other suggested method to enable logging does not. Strangely, in that case the logger _and_ the custom uart are still assigned to UART0, which obviously results in an error.
Just leaving this comment so others might find this for Athom C3 hardware also.

pwoldan23, 2024-08-14:
> I can confirm this also happens on the Athom v3 EU Plug (ESP32 C3 based), where @mspiridakis's fix using a dummy uart works (thanks!). However, the other suggested method to enable logging does not. [...]

I’m having the same problem. Can you show me the exact code you used to solve this? I’ve tried various combinations from the code you referenced but can’t get anything to work.

paulalexandrow, 2024-08-14:
> I’m having the same problem. Can you show me the exact code you used to solve this? I’ve tried various combinations from the code you referenced but can’t get anything to work.

I believe this issue should be solved by now, but for whatever reason you still need my solution, this is the uart configuration that did work on my Athom v3 EU Plugs:

```
uart:
  - id: dummy
    rx_pin: GPIO18
    baud_rate: 4800
  - id: cse7766_uart_bus
    rx_pin: GPIO20
    baud_rate: 4800
```

The logger is disabled like this:

```
logger:
  baud_rate: 0
```

Then just use the cse7766 sensor as normal:

```
sensor:
  - platform: cse7766
    uart_id: cse7766_uart_bus
    ...

```

hth

pwoldan23, 2024-08-17:
> I believe this issue should be solved by now, but for whatever reason you still need my solution, this is the uart configuration that did work on my Athom v3 EU Plugs:
>
> ```
> uart:
>   - id: dummy [...]

Thanks for the quick reply!

RiX012, 2024-09-25:
I have the same issue so this looks not solved out-of-the-box. But if i add the lines of code you mention i get this error:

> Too many candidates found for 'uart_id' type 'uart::UARTComponent' Some are 'dummy', 'cse7766_uart_bus'.

I added this in the yaml:

```yaml
uart:
  - id: dummy
    rx_pin: GPIO18
    baud_rate: 4800
  - id: cse7766_uart_bus
    rx_pin: GPIO20
    baud_rate: 4800

logger:
  baud_rate: 0

sensor:
  - platform: cse7766
    uart_id: cse7766_uart_bus
    current:
      name: "Current"
      filters:
        - throttle_average: ${sensor_update_interval}
        - lambda: if (x < 0.060) return 0.0; else return x;   #For the chip will report less than 3w power when no load is connected
      on_value_range:
        - above: ${current_limit}
          then:
            - switch.turn_off: relay
```

@pwoldan23 @paulalexandrow  did i overlook something? Thanks for you help!

pwoldan23, 2024-09-25:
> I have the same issue so this looks not solved out-of-the-box. But if i add the lines of code you mention i get this error:
>
>
> I added this in the yaml:
>
> ```yaml
> uart:
>   - id: dummy
>     rx_pin: [...]

I tried the same thing at first and had the same error.  You can't copy it directly into your config file as it's duplicating what is in the Athom config file:
`packages:
  Athom_Technology.Smart_Plug_V3: github://athom-tech/esp32-configs/athom-smart-plug.yaml
`

You need to copy the referenced file completely into your config file and then edit it to suit your needs as suggested by @paulalexandrow along with your own config (e.g. SSID etc).

For me this made it compile without error, but I was getting some weird readings (e.g power factor >1) so I decided to just go back to my original config.  I still get the error when compiling and logging doesn't work but I get sensible values in Home Assistant so I'm living with that for now.

I posted about this and got a response from Athom Support but it still makes no sense to me so if you figure it out let me know!
https://www.athom.tech/forum/esphome-related-1/esp32-c3-au-plug-for-esphome-update-error

closed 2024-10-06 by jesserockz