esphome/issues #4530: 400 error "Missing argument configuration" while downloading binary from Home Assistant UI

yaleman, 2023-05-22:
### The problem

Clicking "download" or following the "install" workflow trying to download a compiled .bin file results in a 400 error, and this showing up in the addon logs:

```
2023-05-23 09:23:42,993 WARNING 400 GET /download.bin (0.0.0.0): Missing argument configuration
2023-05-23 09:23:42,993 WARNING 400 GET /download.bin (0.0.0.0) 0.54ms
```

### Which version of ESPHome has the issue?

2023.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.04.1

### What platform are you using?

ESP8266

### Board

Sonoff Basic

### Component causing the issue

n/a

### Anything in the logs that might be useful for us?

```txt
2023-05-23 09:23:42,993 WARNING 400 GET /download.bin (0.0.0.0): Missing argument configuration
2023-05-23 09:23:42,993 WARNING 400 GET /download.bin (0.0.0.0) 0.54ms
```
```

ssieb, 2023-05-23:
Try updating HA.

yaleman, 2023-05-23:
Sorry, seem to have copied the wrong version from the about page - I have updated the ticket - Home Assistant 2023.5.3 is running.

DN0000, 2023-06-18:
I appear to be having a similar issue - @yaleman did you ever find a cause?

Which version of ESPHome has the issue?
2023.5.5

What type of installation are you using?
Home Assistant Add-on

Which version of Home Assistant has the issue?
Home Assistant 2023.6.2
Supervisor 2023.06.2

What platform are you using?
ESP32

Board
Sonoff NSPanel

Component causing the issue
n/a

Example YAML snippet
Default initial config

Anything in the logs that might be useful for us?
2023-06-18 15:21:32,279 INFO 200 POST /wizard (0.0.0.0) 4.47ms
2023-06-18 15:21:32,966 INFO 200 GET /json-config?configuration=lrpanel.yaml (0.0.0.0) 578.29ms
2023-06-18 15:21:32,969 INFO 200 GET /devices (0.0.0.0) 1.80ms
2023-06-18 15:21:33,249 INFO 101 GET /compile (0.0.0.0) 1.45ms
2023-06-18 15:21:33,301 INFO Running command 'esphome --dashboard compile --only-generate /config/esphome/lrpanel.yaml'
2023-06-18 15:21:34,014 INFO Process exited with return code 0
2023-06-18 15:21:37,386 INFO 200 GET /devices (0.0.0.0) 3.39ms
2023-06-18 15:21:42,447 INFO 304 GET /devices (0.0.0.0) 2.68ms
2023-06-18 15:21:42,519 INFO 200 GET /info?configuration=lrpanel.yaml (0.0.0.0) 2.89ms
2023-06-18 15:21:47,089 INFO 200 GET /static/js/esphome/c.e1319253.js (0.0.0.0) 23.67ms
2023-06-18 15:21:47,347 INFO 101 GET /compile (0.0.0.0) 1.78ms
2023-06-18 15:21:47,406 INFO Running command 'esphome --dashboard compile /config/esphome/lrpanel.yaml'
2023-06-18 15:21:47,465 INFO 304 GET /devices (0.0.0.0) 9.43ms
---snip---
2023-06-18 15:23:22,561 INFO 304 GET /devices (0.0.0.0) 1.72ms
2023-06-18 15:23:27,342 INFO Process exited with return code 0
2023-06-18 15:23:27,460 WARNING 400 GET /download.bin (0.0.0.0): Missing argument configuration
2023-06-18 15:23:27,461 WARNING 400 GET /download.bin (0.0.0.0) 1.78ms

yaleman, 2023-06-19:
Nope, I've resorted to provisioning devices by running `esphome` on my laptop then importing it into hass, which is painful.

DN0000, 2023-06-24:
@yaleman - Found the issue for myself at least. Looks like one of the machines I was having issues with was set to use external DNS servers, so was returning my cloudflare IP information instead of the local addresses.

The other problematic machine was tunnelling all DNS back to the office over VPN, causing a similar issue. Hope you get it cracked!

yaleman, 2023-06-25:
@DN0000 what was it trying to resolve and failing? It shouldn't be trying to resolve *anything* while providing a file to download?

yaleman, 2023-06-25:
It looks to relate to this function in tornadoweb: https://github.com/tornadoweb/tornado/blob/6e3521da44c349197cf8048c8a6c69d3f4ccd971/tornado/web.py#L2531 triggered by the lack of a config element... I'm guessing the HASS ingress redirector is messing with it somehow

yaleman, 2023-10-30:
Hey, this isn't stale, it's still an issue!

bdraco, 2023-11-13:
I'm not having any luck replicating this.

Can you post a video so I can try to match your steps exactly?

yaleman, 2023-11-13:
1. Click the ... on a device
2. click "install"
3. click "manual download"
4. it builds, then you get the "which format"
[image]
5. get the error regardless of what you click
[image]

Logs:

[image]

ESPHome: Current version: 2023.10.6

HASS:
Core 2023.11.1
Supervisor 2023.11.0
Frontend 20231030.1

bdraco, 2023-11-13:
Please try ha core 2023.12.x when it comes out.

bdraco, 2023-11-13:
We also have a pending upgrade to aiohttp 3.9.0 which has some more fixes in this area which might also be a contributing factor. I'm not sure that will make 2023.12.x ha core though

yaleman, 2024-05-24:
Issue's still occurring, 

- esphome version: 2024.5.2 
- home assistant:
  - Core 2024.5.4
  - Supervisor 2024.05.1
  - Frontend 20240501.1
Logs:

```
21:43:56,608 INFO 200 GET /info?configuration=m5dial1.yaml (0.0.0.0) 0.58ms
21:43:58,864 INFO 101 GET /compile (0.0.0.0) 0.33ms
21:43:58,897 INFO Running command 'esphome --dashboard compile /config/esphome/m5dial1.yaml'
21:44:07,114 INFO Process exited with return code 0
21:44:07,170 INFO 304 GET /downloads?configuration=m5dial1.yaml (0.0.0.0) 0.47ms
21:44:16,645 WARNING 400 GET /download.bin (0.0.0.0): Missing argument configuration
21:44:16,646 WARNING 400 GET /download.bin (0.0.0.0) 1.05ms
```

hippich, 2024-10-17:
Experiencing similar issue with "Missing argument configuration" error. In my case it only happens when requesting download through Cloudflared tunnel. If I used lan hostname - it works as expected.

DN0000, 2024-10-17:
Might be on to something, I use cloud flare for my external zone.

I was able to resolve this for myself when I disabled my corporate VPN
(i.e. using my local resolvers)

yaleman, 2024-10-17:
Mine goes through cloudflare but also happens locally/bypassing it, which is weird

yaleman, 2026-05-16:
Finally figured this out - looks to be a Cloudflare configuration issue - the cache setting had "ignore query string" set and somehow that was screwing with it. Literally hundreds of applications running through this tunnel/caching platform and this ONE endpoint was causing it 😡

closed 2026-05-16 by yaleman