esphome/issues #4533: name_add_mac_suffix: true breaks update entity in Home Assistant

craigwitter, 2023-05-23:
### The problem

Having name_add_mac_suffix: true in configuration for Esphome device breaks update entity (not available) in Home Assistant.

### Which version of ESPHome has the issue?

2023.5.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.3

### What platform are you using?

ESP32

### Board

esp32dev

### Example YAML snippet

```yaml
esphome:
  name: ${name}
  name_add_mac_suffix: true
```

### Anything in the logs that might be useful for us?

```txt
I've seen this happen when using both frameworks (esp-idf and default arduino). Removing the line name_add_mac_suffix: true or setting it false causes the update entity to immediately become available (after OTA update).
```

ssieb, 2023-05-23:
You should avoid having that option turned on.  It breaks things.  You also can't do OTA updates or view logs, so there's not much point in the update notification anyway, because the dashboard can't find the device.

craigwitter, 2023-05-23:
@ssieb Thanks. Unfortunately, it's part of the BT Proxy default config.

ssieb, 2023-05-23:
What do you mean by "default config"?  How did you install the config?

craigwitter, 2023-05-23:
https://esphome.io/components/bluetooth_proxy.html

and 

https://github.com/esphome/bluetooth-proxies/blob/main/esp32-generic.yaml

ssieb, 2023-05-23:
> https://esphome.io/components/bluetooth_proxy.html

That needs to be fixed...  I wonder why that was put there originally.

> https://github.com/esphome/bluetooth-proxies/blob/main/esp32-generic.yaml

That is a generic config that is intended to be flashed to devices without use of the dashboard.  It enables adoption and if you do adopt the device, the mac address option is automatically set to false for you by the dashboard.

mateuszdrab, 2023-05-25:
It's also part of the the esp muse default config, it would be handy if this worked and ESPhome could keep track of multiple devices based on the same config. For example, if I have 5 speakers.

closed 2023-10-14 by github-actions[bot]