esphome/issues #4534: MH-Z19 results in "Reading from UART timed out at byte 0!" and sometimes in "Invalid preamble"

moritzschaefer, 2023-05-23:
### The problem

Similarly to #488, I get "Reading from UART timed out at byte 0!" error messages with the MH-Z19C (https://www.ebay.de/itm/225181851233).

As shown in the logs and yaml below, I am using UART2 and disable uart-logging, so my issue seems to be different from #488.

Pin-wise, I only connect GND and 5V on one side and RX and TX on the other side. 

Can you see anything in the logs that hints towards what's wrong?

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

nodemcu v2/3

### Component causing the issue

uart/mh-z19

### Example YAML snippet

```yaml
esphome:
  name: mowater
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "Monet"
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Mowater Fallback Hotspot"
    password: "VynwNZN3glml"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.0.13
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.0.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

uart:
  id: mhz19_uart
  rx_pin: GPIO13
  tx_pin: GPIO15
  # rx_pin: GPIO3
  # tx_pin: GPIO1
  baud_rate: 9600
  debug:

captive_portal:

# Enable logging
logger:
# Disable UART Logging
  baud_rate: 0
  level: VERY_VERBOSE
[... 102 lines cut ...]
[22:47:12][VV][uart.arduino_esp8266:180]:     Flushing...
[22:47:12][E][uart:015]: Reading from UART timed out at byte 0!
[22:47:12][W][mhz19:034]: Reading data from MHZ19 failed!
[22:47:12][V][component:199]: Component mhz19.sensor took a long time for an operation (0.13 s).
[22:47:12][V][component:200]: Components should block for at most 20-30ms.
[22:47:12][D][uart_debug:114]: >>> FF:01:86:00:00:00:00:00:79
[22:47:15][VV][api.service:470]: on_ping_request: PingRequest {}
[22:47:15][VV][api.service:043]: send_ping_response: PingResponse {}
[22:47:16][VV][api.service:470]: on_ping_request: PingRequest {}
[22:47:16][VV][api.service:043]: send_ping_response: PingResponse {}
[22:47:22][VV][scheduler:195]: Running interval 'update' with interval=10000 last_execution=41966 (now=51972)
[22:47:22][D][uart_debug:114]: <<< 42:4D:08:D5:08:92:01:A7:01:2E:00:00:E2:11:02:D2:42:4D:08:D5:08:92:01:A7:01:2E:00:00:E2:11:02:D2:42:4D:08:D4:08:91:01:A4:01:2E:00:00:E2:11:02:CD:42:4D:08:D4:08:91:01:A4:01:2E:00:00:E2:11:02:CD:42:4D:08:D4:08:91:01:A4:01:2E:00:00:E2:11:02:CD:42:4D:08:D4:08:91:01:A4:01:2E:00:00:E2:11:02:CD:42:4D:08:D4:08:92:01:A0:01:2E:00:00:E2:11:02:CA:42:4D:08:D4:08:92:01:A0:01:2E:00:00:E2:11:02:CA:42:4D:08:D4:08:92:01:A0:01:2E:00:00:E2:11:02:CA:42:4D:08:D4:08:92
[22:47:22][D][uart_debug:114]: <<< 01:9D:01:2E:00:00:E2:11:02:C7
[22:47:22][VV][uart.arduino_esp8266:180]:     Flushing...
[22:47:22][E][uart:015]: Reading from UART timed out at byte 0!
[22:47:22][W][mhz19:034]: Reading data from MHZ19 failed!
[22:47:22][V][component:199]: Component mhz19.sensor took a long time for an operation (0.13 s).
[22:47:22][V][component:200]: Components should block for at most 20-30ms.
[22:47:22][D][uart_debug:114]: >>> FF:01:86:00:00:00:00:00:79
[22:47:30][VV][api.service:470]: on_ping_request: PingRequest {}
[22:47:30][VV][api.service:043]: send_ping_response: PingResponse {}
[22:47:31][VV][api.service:470]: on_ping_request: PingRequest {}
[22:47:31][VV][api.service:043]: send_ping_response: PingResponse {}
[22:47:32][VV][scheduler:195]: Running interval 'update' with interval=10000 last_execution=51966 (now=61969)
```

moritzschaefer, 2023-06-27:
I read thorugh all the comments from all the threads I found again and esphome claims it's a 'hardware issue' with the NodeMCUs. Apparently, other ESP-based board designs are working fine

74ls04, 2023-07-10:
You accidentally ended up with an HC8 sensor, another Chinese one -- I did the same thing. There are multiple eBay listings selling it as an MH-Z19 sensor but it has a completely different protocol. [See this link ](https://spezifisch.codeberg.page/posts/2022-08-23/co2-sensor-reverse-engineering/).

moritzschaefer, 2023-08-09:
Kind of sad that I gaslighted myself by lengthy threads on connectivity issues between MH-Z19 and NodeMCU. 
Would be amazing if someone builds an esphome plugin for this decide. I can't commit to it unfortunately at the moment

closed 2023-08-09 by moritzschaefer