esphome/issues #4535: ESP32 i2s_audio

mmatus1112, 2023-05-24:
### The problem

Hi all, I am having issues with my i2s_audio.

I am using an ESP32 with an external DAC → UDA1334A and my wiring diagram is as follows:

[image]

My ESPHOME configuration is:

`
esphome:

  name: audio-notifier-1

  friendly_name: audio_notifier_1

esp32:

  board: esp32dev

  framework:

    type: arduino

# Enable logging

logger:

# Enable Home Assistant API

api:

  encryption:

    key: "zEBCe37EmV8ZPmTFlEaWd8BE4YkT45DUVMV0OksMbRU="

ota:

  password: "14a502f63520e30fc336ef966b401601"

wifi:

  ssid: !secret wifi_ssid

  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails

  ap:

    ssid: "Audio-Notifier-1"

    password: "10zAAiH3qVA2"

captive_portal:

i2s_audio:

  i2s_lrclk_pin: GPIO33

  i2s_bclk_pin: GPIO19

media_player:

  - platform: i2s_audio

    name: ESPHome I2S Media Player

    dac_type: external

    i2s_dout_pin: GPIO22

    mode: stereo
`

I am able to see my media player on my Home Assistant and I can play TTS without any issues. The sound is really clear and really nice. However, when I try to play a 5 seconds mp3 sound (an alarm), I can only hear a "white" noise (like an old radio playing noise). I try playing the same audio in my PC speakers (Browser) and I can hear it without issues. 

I am unsure of what the issue is but I have exhausted my ideas on what the problem is. I have tried multiple devices, but I get the same problem. I would appreciate some help!!!

Thank you!

### Which version of ESPHome has the issue?

2023.5.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.5.3

### What platform are you using?

ESP32

### Component causing the issue

i2s_audio

### Example YAML snippet

```yaml
esphome:

  name: audio-notifier-1

  friendly_name: audio_notifier_1

esp32:

  board: esp32dev

  framework:

    type: arduino

# Enable logging

logger:

# Enable Home Assistant API

api:

  encryption:

    key: "zEBCe37EmV8ZPmTFlEaWd8BE4YkT45DUVMV0OksMbRU="

ota:

  password: "14a502f63520e30fc336ef966b401601"

wifi:

  ssid: !secret wifi_ssid

  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails

  ap:

    ssid: "Audio-Notifier-1"

    password: "10zAAiH3qVA2"

captive_portal:

i2s_audio:

  i2s_lrclk_pin: GPIO33

  i2s_bclk_pin: GPIO19

media_player:

  - platform: i2s_audio

    name: ESPHome I2S Media Player

    dac_type: external

    i2s_dout_pin: GPIO22

    mode: stereo
```

sehraf, 2023-05-25:
Can this be related to i2s configuration, e.g. bits per sample, sample rate, number of channels, and so on.
The issue is that the external DAC must be reconfigured which isn't possible right now with esphome.

Maybe tts is 16kHz mono but the mp3 is 44.1kHz stereo?

blakadder, 2023-05-26:
I'm also trying to configure I2S for Sonoff TX Ultimate switch but get a lot of white noise covering the actual sound. I've tried different mp3's with varied sample and bit rates as well as TTS from Home Assistant

Here's a video with the sound sample https://twitter.com/blakadder_/status/1661863695788433409?s=20

NightDav, 2023-05-29:
@blakadder: from your twitter video, your can hear the music, so it could be that you are mixing one channel with the music (let say the left one) and one without data or random data (the right one), when rendered to a mono speaker. have you tried to change stereo to mono ?

blakadder, 2023-05-29:
Yes, that was the first thing I tried. I've tested further and I'm getting the same issue in Tasmota when playing an mp3 but not when using it's built in text to speech function and lowering the gain to 0.3

mmatus1112, 2023-05-30:
After some updates by Home Assistant and ESPHome, now the problem has gone even worse. After all the updates, even the TTS is sounding terrible. Is anybody actually working on this problem?

blakadder, 2023-05-31:
I've been testing various formats and managed to get somewhat decent sound out of the TX Ultimate by lowering the volume to 5% (lowest it can go to produce sound) and using 8bit PCM WAV files

blakadder, 2023-06-02:
Well, i figured out what's wrong in my case. my TM8211 DAC (and its other xx8211 versions) digital input timing format follows the Least significant bit justified (LSBJ), referred to as the Japanese input format. Same seems to be the case with the OP DAC when checking its datasheet
I don't know how to fix it in esphome but it solves the noise issue in Tasmota so there should be a config option to switch between MSB, LSB, PCM and other standards used in I2S

ccorderor, 2023-06-04:
> Well, i figured out what's wrong in my case. my TM8211 DAC (and its other xx8211 versions) digital input timing format follows the Least significant bit justified (LSBJ), referred to as the Japanese input format. Same seems to be the case with the OP DAC when checking its datasheet
> 
> update: the answer lies in this function https://github.com/esphome/ESP32-audioI2S/blob/07cb6eb71fbc47d45185270b5c84c762a126bbc3/src/Audio.cpp#L4408 but no clue on how to set it from yaml

I reached the same conclusion for the TX Ultimate. It's not currently possible to specifiy the timing format from yaml, but I've forked the project, changed it and... it works, but volume is a little too low for the moment

ccorderor, 2023-06-04:
If you want to test, I've pushed a PR: https://github.com/esphome/esphome/pull/4918

mmatus1112, 2023-06-22:
update: Update 2023.6.0 broke the working I2S_Audio. It WAS working without any issues for TTS, Radio and mp3. I see there is a change on I2S on the update. Is there any more direct and responsive place to post the issues?

ccorderor, 2023-06-22:
> update: Update 2023.6.0 broke the working I2S_Audio. It WAS working without any issues for TTS, Radio and mp3. I see there is a change on I2S on the update. Is there any more direct and responsive place to post the issues?

Have you checked the changelog? There are a few changes in 2023.6

sehraf, 2023-06-22:
This comes to my mind: https://github.com/esphome/esphome/pull/4884

h4de5, 2023-10-29:
this is still an issue.