esphome/issues #4537: Restore state failed

IzeroCs, 2023-05-24:
### The problem

I'm currently using esphome 2023.3.2 because later versions can't restore the state after reboot. I tried to wait for the new versions to see if there was any improvement, but to this day it still happens.

I checked the log and the 2023.3.2 version shows "Saving 1 preferences to flash" successfully. In later versions, only some states can be saved, others cannot.

### Which version of ESPHome has the issue?

2023.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.4

### What platform are you using?

ESP32

### Board

Nodemcu

ssieb, 2023-05-24:
You didn't include config or anything.  You probably missed the `restore_mode` default change that was listed in the changelog under breaking changes.

closed 2023-05-24 by ssieb

IzeroCs, 2023-05-24:
Thank you I tried it and it worked. I noticed the changes had the restore_mode section but I didn't read it carefully.

IzeroCs, 2023-05-24:
> You didn't include config or anything. You probably missed the `restore_mode` default change that was listed in the changelog under breaking changes.

Thank you I tried it and it worked. I noticed the changes had the restore_mode section but I didn't read it carefully.