esphome/issues #4539: Unable to automatically extract API key. It may not be set. Open the configuration and look for api:.

AndyRockall, 2023-05-27:
### The problem

When I try to view the api key on any of my esphome devices, from the esphome dashboard, I get the error 'Unable to automatically extract API key. It may not be set. Open the configuration and look for api:'
This has only happened since updating to the latest version 2023.5.4
Also Home Assistant doesn't seem to be picking up any new esphome devices.
There are no errors in the log files

### Which version of ESPHome has the issue?

2023.5.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.4

### What platform are you using?

ESP8266

### Board

All Boards

### Component causing the issue

api key

### Example YAML snippet

```yaml
As sample below all my yaml files have api key declared

esphome:
  name: bed-sensor
  friendly_name: bed-sensor

esp8266:
  board: d1_mini

globals:
  - id: initial_zero
    type: float
    restore_value: yes
    # NOTE: make sure to align this value to the one used in "calibrate_linear" below!
    initial_value: '-481989'
    
  - id: auto_tare_enabled
    type: bool
    restore_value: yes
    initial_value: 'true'

  - id: auto_tare_difference
    type: float
    restore_value: yes
    initial_value: '0'

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "8xDbP9oX8T2zFEkq0lNrO9J4s7vxZxAkCsbAyr2FXD8="

ota:
  password: "78339699c57aa3dd48498747386a9d04"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bed-Sensor Fallback Hotspot"
    password: !secret wifi_password

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
Nothing in the logs to indicate an error
```

mnneely, 2023-05-27:
I’m having exactly the same problem. Was working fine prior to today. Same versions as above.

giancape, 2023-05-28:
Same problem here

mnneely, 2023-05-29:
Seems to be fixed in ESPHome 2023.5.5, at least for me.

AndyRockall, 2023-05-29:
Just got 2023.5.5 update and it seems to have fixed the problem for me too.

I'll leave it open for a few days just to make sure, as I have a few more devices to set up.

AndyRockall, 2023-06-02:
2023.5.5 update fixed the problem.

closed 2023-06-02 by AndyRockall