esphome/issues #4541: Unable to download the ESPHome Project for Pico W

Staqya98, 2023-05-27:
### The problem

I am trying to connect Pico W to ESP home by manually installing it following this video: https://www.youtube.com/watch?v=-nJsHa-pTFw 
The same method is shared in thois video as well: https://www.youtube.com/watch?v=oQpEWwXqnbM&t=199s
However I am at a loss as to why I am getting a preparation failed error. 
I am facing the same error if i try to download wirelessly as well.

### Which version of ESPHome has the issue?

https://github.com/maxgerhardt/platform-raspberrypi.git

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.5.4

### What platform are you using?

ESP32

### Board

Pico W

### Component causing the issue

yaml

### Example YAML snippet

```yaml
esphome:
  name: pico1

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "qNpxftAXS4xNUoxdVq7Go4AoFU0WMxx5yhAMfimQV78="

ota:
  password: "384bdc8ad31802946b03502b80450e25"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.190
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Pico1 Fallback Hotspot"
    password: "yM71L899ToUz"
```

### Anything in the logs that might be useful for us?

```txt
ESPHome 2023.6.0-dev
INFO Reading configuration /config/esphome/pico1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing pico1 (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
Tool Manager: Installing earlephilhower/tool-pioasm-rp2040-earlephilhower @ *
INFO Installing earlephilhower/tool-pioasm-rp2040-earlephilhower @ *
Error: Could not find the package with 'earlephilhower/tool-pioasm-rp2040-earlephilhower @ *' requirements for your system 'linux_aarch64'
```

### Additional information

[image]

ssieb, 2023-05-27:
platformio has deliberately broken it.

Staqya98, 2023-05-28:
How can I connect the pico to HA then? Spent whole of yesterday on this

Staqya98, 2023-05-28:
> platformio has deliberately broken it.
How can I connect the pico to HA then? Spent whole of yesterday on this

ssieb, 2023-05-28:
Sorry, at this point the pico platform is dead for esphome at this point and I don't see it getting fixed in the near future.

ssieb, 2023-05-28:
Actually, it might be able to be working with slightly less functionality.  The missing library is currently only used for addressable LEDs, so if you don't use that, it will probably work.  But the automatic download of that library needs to be removed for that.

Staqya98, 2023-05-28:
Thank you I will be looking for alternates.

closed 2023-05-28 by Staqya98

ssieb, 2023-05-28:
You should be able to use this again with the next point release which will be out soon.
(As long as you don't use addressable LEDs.)

ssieb, 2023-05-28:
Actually, you didn't say which esphome version you were using.  This should only have affected dev, so that is fixed now.

Staqya98, 2023-05-29:
> Actually, you didn't say which esphome version you were using. This should only have affected dev, so that is fixed now.
Am using dev. thanks for the update.