esphome/issues #4543: Why does no one care that Dallas DS18B20 sensors have not been working with the ESP32 since 6 months?

selfmade01, 2023-05-28:
### The problem

Hello ESPhome or Dallas library programmer.

I can not understand that a problem as big as here is not solved promptly by you.
Perhaps several thousand users have installed sensors with an ESP32 in their DS18B20 system and have been having massive problems for about 6 months, half a year. Perhaps they are looking elsewhere, with the cable, with the sensor, ordering new sensors and and and, but the problem is quite simply the program code.

A lot of discussion can read about that problems in home-assistant or directly her.
**With every little fart in the program, an update comes a few days later and the problem is solved**. **But when it comes to the DS18B20 sensor and maybe half the automation system in the house doesn't work anymore because they don't get values ​​from the sensors anymore, nobody seems to care,**
Whats going on here?

If you don't feel like solving the problem, please help the many users to get their sensors working again and write a tip here at which point in the program code you need to make a change.

Thanks and best regards
Achim

### Which version of ESPHome has the issue?

ESPHome 2023.5.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.4

### What platform are you using?

ESP32

### Board

ESP32-S2-Saola-1R

### Component causing the issue

1-wire/dallas/DS18B20

### Example YAML snippet

```yaml
Not need, everybody know about this problem
```

### Additional information

**### ESP32 and DS18B20 Users need help**

RoelHermus, 2023-05-28:
I did just a trial for you with an ESP32 and a 18B20. Power 3.3 volt and the data line op GPIO23. May be you forget a resistor between the data line and the plus. In the datasheet you can found a value of 4k7. In my trial I used a 2k7 one. Everthing works perfect in my setup.

ssieb, 2023-05-28:
@selfmade01 it's not that we don't care.  The problem is that anyone that *could* fix it can't reproduce the problem.  It's not "several thousand users", where did you come up with that?  It still works for most people.

closed 2023-05-28 by ssieb

selfmade01, 2023-05-29:
@RoelHermus 
the Problems with the ESP32 can find there
https://github.com/esphome/issues/issues/3980
i get the same problems on my ESP32-S2-Saola_1R, on my ESP8266 works fine.
And shure, i changed before the cable to a special shielded Data-Cable...
but on a way of 30, 50 or 100cm can try anything and this not work and i changed also the resistors from 4.7K up 3.3 and 2.2K. So long i used only 1 DS18B20 in the line this only one works fine.
If i add only a second ds18B20 this brokes everytime the second.
He find first after booting and initialize both sensors, and then he get the errors 90% ... maybe from 10 querys i get sure 9 errors
in arduino and ESP8266 works fine. i think thats the same timing problem what get the users in the link above

RoelHermus, 2023-05-29:
@selfmade01 
Today I have made a setup with 4 chinese DS18B20 sensors and a cat5 cable of 30 meters long. It works perfect on 3.3 volt and on 5 volt. I use GPIO23 as data pin and a resistor of 2k7. The esp32 is a  ESP32-S version on a Nodemcu version 1.1 board. (An old one). When I remove 1 sensor it give an error: Scratch pad checksum invalid!

Can you discrip your setup, maybe I can reproduce your error.

1N4148, 2023-05-29:
Please check the GPIO pins your onewire bus is connected to. GPIOs with a possible hardware UART are prone to fail.

selfmade01, 2023-05-29:
I have two one-Wire Bus, hub_1, hub_2 
there is the log-output
`<html>
<body>

21:19:17 | [D] | [sensor:094] | 'BigBox Panel-2 4000K Power': Sending state 8.22000 W with 3 decimals of accuracy
-- | -- | -- | --
21:19:21 | [D] | [dallas.sensor:143] | 'bigbox/ds18b20/panel_1_temp': Got Temperature=26.1°C
21:19:21 | [D] | [sensor:094] | 'bigbox/ds18b20/panel_1_temp': Sending state 25.06250 °C with 1 decimals of accuracy
21:19:37 | [D] | [dallas.sensor:143] | 'bigbox/ds18b20/uc_box_stepdown_temp': Got Temperature=25.8°C
21:19:37 | [D] | [sensor:094] | 'bigbox/ds18b20/uc_box_stepdown_temp': Sending state 24.81250 °C with 1 decimals of accuracy
21:19:37 | [W] | [dallas.sensor:261] | 'bigbox/ds18b20/uc_box_temp' - Scratch pad checksum invalid!
21:19:37 | [D] | [sensor:094] | 'bigbox/ds18b20/uc_box_temp': Sending state nan °C with 1 decimals of accuracy
21:19:51 | [D] | [dallas.sensor:143] | 'bigbox/ds18b20/panel_1_temp': Got Temperature=26.0°C
21:19:51 | [D] | [sensor:094] | 'bigbox/ds18b20/panel_1_temp': Sending state 25.00000 °C with 1 decimals of accuracy

</body>
</html>`

and this is my code
`dallas:
  - pin: GPIO3
    id: hub_1
    update_interval: 30s
  - pin: GPIO4
    id: hub_2
    update_interval: 30s
`

`- platform: dallas
    address: 0x3100000df746fa28
    dallas_id: hub_1
    name: "bigbox/ds18b20/uc_box_stepdown_temp"
    resolution: 12
    filters:
      - offset: -1.0
  - platform: dallas
    address: 0x3f00000df6546028 
    dallas_id: hub_1
    name: "bigbox/ds18b20/uc_box_temp"
    resolution: 12
    filters:
      - offset: -1.0
  - platform: dallas
    address: 0x1b00000df723ec28
    dallas_id: hub_2
    name: "bigbox/ds18b20/panel_1_temp"
    resolution: 12
    filters:
      - offset: -1.0`

on hub_1, GPIO3 i have the cable with 2 Sensors in 1 line, and one sensor only works...

@1N4148 
Hardware UART i think i can add every GPIO on this board as UART.
i use now GPIO3 and GPIO4 for the 1.wire-bus hub_1 and hub_2
all what i finf don't told me GPIO3 or 4 is reserved or can special to use for UART

selfmade01, 2023-05-29:
@RoelHermus 
yes, on a ESP32 NodeMCU i think works too... because my ESP8266 D1-mini works also with 2 sensors in a line
But i will test it tonight and i will tell you the result

My configuration

My Hardware:
- ESP32-S2-Saola-1R
- Level-Shifter (because i try the 1-wire with 5v too (on ESP8266 works both fine only on the ESP32-s2-saola-1R works only with 1 sensor in the line

you can be happy with your china DS18B20, because i tested 10 from some years ago... all 10 not works sable... after 10 times change the cable, the resistors, the solder joints i buy some new but original... all works fine, i never will buy any sensors from china-copy-company, i read to much (about problems from users) about BMExxx, SHTxx and DS18B20 fake sensors from china... toooo much on the market

My configuration code-section is like that

`substitutions: 
  devicename: bigbox-lightcontroller
  upper_devicename: ESP32_S2_BigBox_LightController
  device_location: Kueche
  device_description:  BigBox Lichtsteuerung platform using ESPhome firmware
  ipaddress: 192.168.2.102
  last_update: "2023-05-15 18:54:12"

esphome:
  name: $devicename
  friendly_name: BigBox_LightController
 # libraries:
 #   - https://github.com/RobTillaart/ACS712.git
 # includes:
 #   - acs712_component.h

esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino
    #type: esp-idf
    version: recommended
# test wegen checksumfehler bei den Dallas-sensoren
#external_components:
#  - source: github://edenhaus/esphome@test-dallas
#    components: [ dallas ]

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret ESPHOME_API_KEY

ota:
  safe_mode: true
  password: !secret HA_OTA_PASSWORD

wifi:
  power_save_mode: none # none (default for esp8266), light (default for esp32), high
  ssid: !secret WLAN_ASUS_SSID
  password: !secret WLAN_ASUS_PASSWORD
  manual_ip:
    static_ip: !secret ESP32_S2_BIGBOX_LIGHT_IP
    gateway: !secret WLAN_ASUS_GATEWAY_IP
    subnet: !secret WLAN_ASUS_SUBNET
    dns1: !secret WLAN_ASUS_DNS1
    #fast_connect: on
  ap: # Enable fallback hotspot (captive portal) in case wifi connection fails
    ssid: "Bigbox-Lightcontrol"
    password: !secret WIFI_AP_PASSWORD
  # Enable fallback hotspot (captive portal) in case wifi connection fails`

ssieb, 2023-05-29:
@selfmade01 show the logs for the detected sensors

selfmade01, 2023-05-30:
@ssieb  and @RoelHermus 

I'm currently running a fault exclusion system.
When I checked all the voltages on my PWM channels (I use 7 PWM channels), I noticed that I only had 1.6V on one channel instead of 3.29V.

I don't think it's the LevelShifter that's causing the errors on the DS18B20 sensors, but I've disassembled everything now.
I even checked a second ESP32-S2-Saoal-1R board and I had the voltage error there too.
The code looked ok because I also put the specific part of the code on other pins and then had the voltage problem with one of the 7 PWM channels there as well. Normally, the ESP-S2 can handle 8 channels.

But I'll check everything now until the error occurs again. Then I know what the problem is and I hope that this is also the error in the 1-wire bus

I'll get back to you, maybe I have to apologize to all of you

Greetings
Achim

selfmade01, 2023-05-31:
ok,
i changed all now to a breadboard and check all again with differnt configurations

i add now the log-output you can find there the Sensor 
bigbox/ds18b20/uc_box_temp
and the error-message
scratch pad invalid checksum
`[02:22:43][I][app:102]: ESPHome version 2023.5.5 compiled on May 31 2023, 02:10:35
[02:22:43][C][wifi:505]: WiFi:
[02:22:43][C][wifi:363]:   Local MAC: A0:76:4E:92:51:D4
[02:22:43][C][wifi:364]:   SSID: [redacted]
[02:22:43][C][wifi:365]:   IP Address: 192.168.2.102
[02:22:43][C][wifi:367]:   BSSID: [redacted]
[02:22:43][C][wifi:368]:   Hostname: 'bigbox-lightcontroller'
[02:22:43][C][wifi:370]:   Signal strength: -37 dB ▂▄▆█
[02:22:43][C][wifi:374]:   Channel: 13
[02:22:43][C][wifi:375]:   Subnet: 255.255.255.0
[02:22:43][C][wifi:376]:   Gateway: 192.168.2.1
[02:22:43][C][wifi:377]:   DNS1: 192.168.2.1
[02:22:43][C][wifi:378]:   DNS2: 0.0.0.0
[02:22:43][C][logger:301]: Logger:
[02:22:43][C][logger:302]:   Level: DEBUG
[02:22:43][C][logger:303]:   Log Baud Rate: 115200
[02:22:43][C][logger:305]:   Hardware UART: UART0
[02:22:43][C][i2c.arduino:053]: I2C Bus:
[02:22:43][C][i2c.arduino:054]:   SDA Pin: GPIO8
[02:22:43][C][i2c.arduino:055]:   SCL Pin: GPIO9
[02:22:43][C][i2c.arduino:056]:   Frequency: 50000 Hz
[02:22:43][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[02:22:43][I][i2c.arduino:069]: Results from i2c bus scan:
[02:22:43][I][i2c.arduino:075]: Found i2c device at address 0x23
[02:22:43][I][i2c.arduino:075]: Found i2c device at address 0x40
[02:22:43][I][i2c.arduino:075]: Found i2c device at address 0x70
[02:22:43][I][i2c.arduino:075]: Found i2c device at address 0x76
[02:22:44][C][ledc.output:160]: LEDC Output:
[02:22:44][C][ledc.output:161]:   Pin GPIO16
[02:22:44][C][ledc.output:162]:   LEDC Channel: 0
[02:22:44][C][ledc.output:163]:   PWM Frequency: 4000.0 Hz
[02:22:44][C][ledc.output:164]:   Bit depth: 13
[02:22:44][C][light:103]: Light 'Atest an GPIO16'
[02:22:44][C][light:105]:   Default Transition Length: 1.0s
[02:22:44][C][light:106]:   Gamma Correct: 2.80
[02:22:44][C][dallas.sensor:075]: DallasComponent:
[02:22:44][C][dallas.sensor:076]:   Pin: GPIO7
[02:22:44][C][dallas.sensor:077]:   Update Interval: 30.0s
[02:22:44][D][dallas.sensor:082]:   Found sensors:
[02:22:44][D][dallas.sensor:084]:     0x3f00000df6546028
[02:22:44][D][dallas.sensor:084]:     0x3100000df746fa28
[02:22:44][C][dallas.sensor:089]:   Device 'bigbox/ds18b20/uc_box_stepdown_temp'
[02:22:44][C][dallas.sensor:089]:     Device Class: 'temperature'
[02:22:44][C][dallas.sensor:089]:     State Class: 'measurement'
[02:22:44][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[02:22:44][C][dallas.sensor:089]:     Accuracy Decimals: 1
[02:22:44][C][dallas.sensor:097]:     Address: 0x3100000df746fa28
[02:22:44][C][dallas.sensor:098]:     Resolution: 12
[02:22:44][C][dallas.sensor:089]:   Device 'bigbox/ds18b20/uc_box_temp'
[02:22:44][C][dallas.sensor:089]:     Device Class: 'temperature'
[02:22:44][C][dallas.sensor:089]:     State Class: 'measurement'
[02:22:44][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[02:22:44][C][dallas.sensor:089]:     Accuracy Decimals: 1
[02:22:44][C][dallas.sensor:097]:     Address: 0x3f00000df6546028
[02:22:44][C][dallas.sensor:098]:     Resolution: 12
[02:22:44][C][cd74hc4067:025]: CD74HC4067 Multiplexer:
[02:22:44][C][cd74hc4067:026]:   S0 Pin: GPIO2
[02:22:44][C][cd74hc4067:027]:   S1 Pin: GPIO3
[02:22:44][C][cd74hc4067:028]:   S2 Pin: GPIO4
[02:22:44][C][cd74hc4067:029]:   S3 Pin: GPIO5
[02:22:44][C][cd74hc4067:030]: switch delay: 2
[02:22:44][C][bme680.sensor:215]: BME680:
[02:22:44][C][bme680.sensor:216]:   Address: 0x76
[... 240 lines cut ...]
    sensor: adc_sensor
    update_interval: 60s
    unit_of_measurement: "W"
    device_class: "power"
    filters:
      - lambda: return (x * 10);

#[02:17:10][D][dallas.sensor:084]:     0x3f00000df6546028
#[02:17:10][D][dallas.sensor:084]:     0x3100000df746fa28
  - platform: dallas
    address: 0x3100000df746fa28
    dallas_id: hub_1
    name: "bigbox/ds18b20/uc_box_stepdown_temp"
    resolution: 12
    filters:
      - offset: -1.0
  - platform: dallas
    address: 0x3f00000df6546028 
    dallas_id: hub_1
    name: "bigbox/ds18b20/uc_box_temp"
    resolution: 12
    filters:
      - offset: -1.0`
@ssieb 
if is possible to get a log-output from then sensors in more detail then tell me please what i have to do f

selfmade01, 2023-05-31:
Okay,

now i tested again and again... 
i use now on a breadboard same same ESP32-saola-1 (i have two of this boards)

i add my code 
`esphome:
    name: box
    comment: lightcontroller
esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino
    #type: esp-idf
    #version: recommended
    #sdkconfig_options:
    #  CONFIG_COMPILER_OPTIMIZATION_SIZE: y

logger:
  level: VERY_VERBOSE

api:
  encryption:
    key: !secret ESPHOME_API_KEY

ota:
  safe_mode: true
  password: !secret HA_OTA_PASSWORD

wifi:
  power_save_mode: none # none (default for esp8266), light (default for esp32), high
  ssid: !secret WLAN_ASUS_SSID
  password: !secret WLAN_ASUS_PASSWORD
  manual_ip:
    static_ip: !secret ESP32_S2_BIGBOX_LIGHT_IP
    gateway: !secret WLAN_ASUS_GATEWAY_IP
    subnet: !secret WLAN_ASUS_SUBNET
    dns1: !secret WLAN_ASUS_DNS1

  ap:
    ssid: "Bigbox-Lightcontrol"
    password: !secret WIFI_AP_PASSWORD
 
web_server:
  port: 80
  auth:
    username: admin
    password: admin

dallas:
  - pin: GPIO11

#[21:46:43][D][dallas.sensor:084]:     0x3f00000df6546028
#[21:46:43][D][dallas.sensor:084]:     0x3100000df746fa28
sensor:
  - platform: dallas
    address: 0x3f00000df6546028
    id: uc_box_ds18b20_temp
    name: "Uc Box"
  - platform: dallas
    address: 0x3100000df746fa28
    name: "Uc Box StepDown"
    id: uc_box_stepdown_ds18b20_temp`

thats my error-verbose-log
`[21:52:35][D][dallas.sensor:082]:   Found sensors:
[21:52:35][D][dallas.sensor:084]:     0x3f00000df6546028
[21:52:35][D][dallas.sensor:084]:     0x3100000df746fa28
[21:52:35][C][dallas.sensor:089]:   Device 'Uc Box'
[21:52:35][C][dallas.sensor:089]:     Device Class: 'temperature'
[... 193 lines cut ...]
}
[21:53:57][V][component:204]: Component dallas took a long time for an operation (0.07 s).
[21:53:57][V][component:205]: Components should block for at most 20-30ms.
[21:53:57][VV][scheduler:226]: Running timeout '0x3100000df746fa28' with interval=750 last_execution=91162 (now=92010)
[21:53:57][VV][dallas.sensor:258]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
[21:53:57][W][dallas.sensor:261]: 'Uc Box StepDown' - Scratch pad checksum invalid!
[21:53:57][V][sensor:043]: 'Uc Box StepDown': Received new state nan
[21:53:57][D][sensor:094]: 'Uc Box StepDown': Sending state nan °C with 1 decimals of accuracy
[21:53:57][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[21:53:57][V][json:056]: Size after shrink 76 bytes
[21:53:57][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 3083992812
  state: nan
  missing_state: NO
}`

and i get this **Uc Box StepDown' - Scratch pad checksum invalid!** problem same same like bevore
and all looks same same like this [https://github.com/esphome/issues/issues/3909#issuecomment-1535132129](url)

i think this ESP32-S2-saola-1 use also a little differnt timings or any other
because use i same same konfiguration on a ESP8266 Board all works fine without any errors

what i can do?
if is possible... i pay for you a ESP32-S2-Saola-1 board ... no problem this 10 or 15 Euros
but this problem i think i can't clear without go deep in the library to change there the waiting-timings

ssieb, 2023-06-01:
For some reason, you're using a single backquote for your big pastes, which doesn't work properly.  Use 3 backquotes on the line before and the line after the pasted chunk.  See the edits I've made.

Have you tried with just one of the sensors at a time?  Try each of them separately.  It's rather strange that one works and one doesn't, when it's the exact same code for both of them and the same GPIO pin.

selfmade01, 2023-06-01:
@ssieb
one sensor on one 1-wire hub works  without any problems.
i tryed too already... i try also with 1K, 2.2K, 3.3K and 4.7K Resistors, also i tryed with 5V and LevelShifter and without LevelShifter and 3.3V

If is not possible that i have her the same timing problem like in this other thread?
Because, everytime when the ESP start, he find all sensors,,,only later by reading the values  he get the problems

RoelHermus, 2023-06-01:
I guess there is an issue with the power source or the cable from power supply to esp32 board. It looks that the start up works fine, but when it switch to running mode, it needs more current for the wifi connection. Maybe thats the problem.

selfmade01, 2023-06-01:
@RoelHermus 
I#m not sure if this possible. If is possible then only by this kind of ESP32 Board because:
1) i get the Problem and the ESP32-S2-saola get the power from a AC/DC 220v/24V and i use a stepDown Concerter LM2596 for the 5V (but i use this for both ESP8266 and ESP32-S2-Saola and the first works without any problems... but maybe need this ESP32 a much more stable power-connection

2) i use all Boards with a USB-Bale-5V Connector from my PC-Workstation and also only that get the ESP32-S2-Saola problems

if is not possible to ask expressiff self? maybe they know it what is the different her 
I tell that already... if the  specialists her need this board.. i buy 1,2 or 3 then you can check it much better and see in what a section have the library and the board a problem , or any other like the power-connector.

selfmade01, 2023-06-01:
other idea:
everytime the first DS18B20 reading is ok.
only the second makes the problem.
If is possible to set a break between this readings?
i'm not a electronic spcialist but maybe the line for the second reading need some uS more time to get enough power

selfmade01, 2023-06-02:
@RoelHermus 
yes, thats a good idea...you mean change the addresses in the ESPHome Code?
in the sensor in think i can't change it because i think that's hard-coded

RoelHermus, 2023-06-02:
@selfmade01 , change the scan cyclus from:

  - platform: dallas
    address: 0x3f00000df6546028
    id: uc_box_ds18b20_temp
    name: "Uc Box"
  - platform: dallas
    address: 0x3100000df746fa28
    name: "Uc Box StepDown"
    id: uc_box_stepdown_ds18b20_temp
to
  - platform: dallas
    address: 0x3100000df746fa28
    name: "Uc Box StepDown"
    id: uc_box_stepdown_ds18b20_temp
  - platform: dallas
    address: 0x3f00000df6546028
    id: uc_box_ds18b20_temp
    name: "Uc Box"

The board you  use have ESP32-S2 microcontrollers installed. I do not have that kind in my collection, but I ordered 2 pieces boards with the ESP32-S2 microcontroller. I can do a little try. I let you know the results.

JamieR007, 2023-06-02:
I upgraded today to esphome 2023.5.5 and still seeing the "scratch pad checksum invalid!" errors every few seconds. I'm using the following config:

Olimex ESP32-POE
x4 DS18B20 sensors in a star configuration at the end of @6m of CAT5e, using a 3.3kOhm pullup.

For me, the only viable workaround is to modify some lock code in dallas_component.cpp from:

```
bool IRAM_ATTR DallasTemperatureSensor::read_scratch_pad() {
  auto *wire = this->parent_->one_wire_;
  {
    InterruptLock lock;

    if (!wire->reset()) {
      return false;
    }
  }

  {
    InterruptLock lock;

    wire->select(this->address_);
    wire->write8(DALLAS_COMMAND_READ_SCRATCH_PAD);

    for (unsigned char &i : this->scratch_pad_) {
      i = wire->read8();
    }
  }

  return true;
}
```

to:

```
bool IRAM_ATTR DallasTemperatureSensor::read_scratch_pad() {
  auto *wire = this->parent_->one_wire_;

  {
    InterruptLock lock;

    if (!wire->reset()) {
      return false;
    }
  //}

  //{
    //InterruptLock lock;

    wire->select(this->address_);
    wire->write8(DALLAS_COMMAND_READ_SCRATCH_PAD);

    for (unsigned char &i : this->scratch_pad_) {
      i = wire->read8();
    }
  }

  return true;
}
```

Maintaining the interrupt lock throughout the method call (as it was originally implemented before this issue started occurring) appears to resolve the issue.

Further discussion of this workaround is described in https://github.com/esphome/issues/issues/3980.

Happy to assist with further debug and testing if requried.

1N4148, 2023-06-02:
You can also copy the code into a subdir named 'external_components' in the dir where your YAML files reside and include it with

`external_components:
  - source: 
      type: local
      path: external_components`

selfmade01, 2023-06-03:
@RoelHermus 
Thank you
what you think?
now the error 
``<html>
<body>

12:54:59 | [W] | [dallas.sensor:261] | 'Uc Box' - Scratch pad checksum invalid!
-- | -- | -- | --

</body>
</html>``
the second 
i think(no, i'm sure now)all DS18B20 comes after the first sensor in the code, get the problems
bevore i change the sequence, i get only problems with " uc_box_stepdown_ds18b20_temp" because he staay on the first

@JamieR007 
thanks a lot for this code...
I've tried before but all I got after compiling was an original file because it was overwritten again.

I'll try again today or tomorrow and make sure I use an include

@1N4148 
that is a very good way... 
ok i will do that on this weekend
Thank's a lot

Thank's a lot all her!!!

RoelHermus, 2023-06-03:
@selfmade01 
Today I get a Wemos S2 Mini ESP32-S2 and, ofcourse, I did a test. You are right, the ESP32-S2 gives errors with more DS18B20 sensors in a string.

RoelHermus, 2023-06-03:
@selfmade01 

I found an anther solution without modification of the code:

```typescript
dallas:
  - pin: GPIO11
    update_interval: 10s
    id: S1
  - pin: GPIO11
    update_interval: 10s
    id: S2   
  - pin: GPIO11
    update_interval: 10s
    id: S3 
  - pin: GPIO11
    update_interval: 10s
    id: S4

#     0xe43c01d6072f1828
#     0x1d3c01d60736ce28
#     0xdd000002db578128
#     0xa03c01d60701f928
sensor:
  - platform: dallas
    dallas_id: S1
    address: 0xe43c01d6072f1828
    name: "Temp 1"

  - platform: dallas
    dallas_id: S2
    address: 0x1d3c01d60736ce28
    name: "Temp 2"

  - platform: dallas
    dallas_id: S3
    address: 0xdd000002db578128
    name: "Temp 3"

  - platform: dallas
    dallas_id: S4
    address:  0xa03c01d60701f928
    name: "Temp 4"
```

ssieb, 2023-06-03:
@RoelHermus that works when it wasn't working before?

1N4148, 2023-06-03:
@RoelHermus 
How did you get them running? Also bought some Wemos S2 Mini ESP32-S2 clones but they are quite instable regarding the wifi connection

RoelHermus, 2023-06-03:
@1N4148 
I did nothing special. I connected the ESP32-S2 to a power supply of 3.3 volt and connect it to  the Raspberry Pi 4 with an usb cable. I flash the image to the esp32-S2. I did not a long duration test. Maybe the ESP32-S2 is sometimes instable.

1N4148, 2023-06-03:
Ok, got it running, changed the Arduino framework from recommended to latest. No instable Wifi connection anymore.

Test setup with 10 sensors (cable length from 1m to 10m, star, 1.8kOhm pullup to 5V, 5V for sensor Vcc) each on a single dallas hub works even without the interrupt patch. Strange.

selfmade01, 2023-06-11:
@RoelHermus
thank's for this kind of solution with the different id's (like multiple bus-lines)
but i'm not sure how much lines dallas lib can manage

I thought before that you only use bus lines if you have different wiring and you want to separate them so that one doesn't disturb the other.
If that works then I can leave it like that for now, but it would be super great if the whole library would also be adapted later

JamieR007, 2023-06-13:
I suspect the different IDs work because the sensors will not be competing for the same lock in DallasTemperatureSensor::read_scratch_pad(). Each one will have it's own unique lock, so they cannot collide.

This seems to me to be the root of the issue. The question is why this change was implemented in the first place as it doesn't seem to work with multiple sensors (on certain devices).

JamieR007, 2023-07-06:
Can I ask why this issue has been closed when there is obviously still an reproducable issue here?

The workaround of using separate dallas hubs requires a I/O pin for each sensor which isn't scalable.

ssieb, 2023-07-06:
There are multiple other open issues about this and the title of this one is unnecessary.

Arkoxs, 2023-10-30:
> You can also copy the code into a subdir named 'external_components' in the dir where your YAML files reside and include it with
> 
> ```
> external_components:
>   - source: 
>       type: local
>       path: external_components
> ```

Hi @1N4148, This is driving me nuts.

I've created in my device directory a my_components directory and added the dallas files. There I edited the dallas_component.cpp file and added the external_components: section to the yaml file.

When removing the build files and start compiling again I don't see that my altered dallas component file is compiled and linked. Just the original file is mentioned.
....
Compiling .pioenvs/floorheating/src/esphome/components/dallas/dallas_component.cpp.o
Compiling .pioenvs/floorheating/src/esphome/components/dallas/esp_one_wire.cpp.o
....

I've put the dallas files in:
\config\esphome\floorheating.yaml
\config\esphome\my_components\dallas_component.cpp
\config\esphome\my_components\dallas_component.h
\config\esphome\my_components\esp_one_wire.cpp
\config\esphome\my_components\esp_one_wire.cpp

```
substitutions:
  devicename: floorheating
  upper_devicename: Floorheating
  device_friendly_name: "Floorheating temperature sensors"
  device_comment: "Warmtesensoren voor de aan- en afvoer van warmwater in de vloerverwarming"
  api_encryption_key: !secret api_encryption
  ota_password: !secret ota_password
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  wifi_fallback_password: !secret fallback_password

esp8266:
  board: d1_mini

<<: !include common/common.yaml
<<: !include common/homeassistant.yaml
<<: !include common/default-sensors.yaml

external_components:
  - source:
      type: local
      path: my_components

# Exposed switches.
switch:
  # Switch to restart the sensor.
  - platform: restart
    name: $upper_devicename Restart

# Initialize Dallas Temperature sensors on pin D4
dallas:
  - pin: D4

sensor:
  - platform: dallas
    address: 0x28B7A5AC0E000032
    resolution: 10
    name: $upper_devicename Zone
  - platform: dallas
    address: 0x282AB1AC0E0000AA
    name: $upper_devicename Thermostaat
  - platform: dallas
    address: 0x2883A1AC0E00001C
    name: $upper_devicename Pomp
  - platform: dallas
    address: 0x285CB8AC0E000073
    name: $upper_devicename Groep 1
  - platform: dallas
    address: 0x2852A2AC0E0000A1
    name: $upper_devicename Groep 2
  - platform: dallas
    address: 0x28F098AC0E0000AA
    name: $upper_devicename Groep 3
  - platform: dallas
    address: 0x28EDAAAC0E0000BC
    name: $upper_devicename Groep 4
  - platform: dallas
    address: 0x288AB4AC0E000024
    name: $upper_devicename Groep 5
  - platform: dallas
    address: 0x2875A4AC0E00000E
    name: $upper_devicename Groep 6 
  - platform: dallas
    address: 0x2847A4AC0E00008D
    name: $upper_devicename Afvoer
  - platform: dallas
    address: 0x28FF640E7779E44A
    name: $upper_devicename Toevoer
```

I'm quite lost here.

1N4148, 2023-10-30:
@Arkoxs Create a sub-directory named _dallas_ unter _external_components_ and put the source files there

Arkoxs, 2023-10-30:
@1N4148 Done, but now the building fails :-(

```
INFO ESPHome 2023.10.3
INFO Reading configuration /config/esphome/floorheating.yaml...
INFO Detected timezone 'Europe/Amsterdam'
Failed config

dallas: [source /config/esphome/floorheating.yaml:33]
  
  Component dallas cannot be loaded via YAML (no CONFIG_SCHEMA).
  - pin: D4
sensor.dallas: [source /config/esphome/floorheating.yaml:51]
```

Arkoxs, 2023-11-03:
OK, just to get the complete picture. I've created a directory: \config\esphome\my_components\dallas and downloaded the Dallas component files from [esphome:dallas](https://github.com/esphome/esphome/tree/dev/esphome/components/dallas) 

\config\esphome\my_components\dallas\dallas_component.cpp
\config\esphome\my_components\dallas\dallas_component.h
\config\esphome\my_components\dallas\esp_one_wire.cpp
\config\esphome\my_components\dallas\esp_one_wire.h
\config\esphome\my_components\dallas\__init__.py
\config\esphome\my_components\dallas\sensor.py

I've altered esp_one_wire.cpp and changed the `uint32_t timing_constant` to `20`
I've altered dallas_component.cpp and commented the `InterruptLock lock;` from the `read_scratch_pad()` function.

Then I added the following lines to my esphome yaml file:
```
external_components:
  - source:
      type: local
      path: my_components
    components: [ dallas ]
```
Unfortunately nothing changed, still the same error in the logging.

I've also tried a different GPIO (D4 to D2) but no luck.
Just to be sure I've switched from Wemos D1 mini to an ESP12 on a dev board but the same error occurs.
Next step is cutting down on # of sensors in my string to see if the total number of 11 sensors is just to much.

Arkoxs, 2023-11-05:
Now I've tried:
- Moving the clone sensor from D2 to D4
- Replacing the USB power source with a 5V stabilized power source
- Altering the sensors in the code so reading is not in order #1, #2, #3 but #1, #8, #2, #9
- Reading only one sensor
- Changing the hardcoded timing constant in esp_one_wire.cpp

The only option left that did work (without changes to the hardware) was to switch to Tasmota. Flashing it and defining GPIO02 and GPIO04 as two ds18b20 devices immediately resulted in a list of 8 temperature sensors. That's without changing the source code, the default maximum sensors.

So it does not seem to be an hardware issue.