esphome/issues #4549: ESPhome configuration

LYL14, 2023-05-31:
### The problem

After using HA, I want to configure bin file and yaml file in other ways. Is there any other way?

### Which version of ESPHome has the issue?

2023

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

ssieb, 2023-05-31:
github issues are not for support requests.  Come ask on the discord server or one of the forums.

LYL14, 2023-05-31:
Can ESPhome configure ESPnow and ESPmesh functions?

closed 2023-10-13 by github-actions[bot]