esphome/issues #4553: Failure to compile due to climate.h

786534786, 2023-06-01:
### The problem

Since version 2023.4.0 of ESPHome it is no longer possible to compile the code below.

Here is the full log:

```

INFO ESPHome 2023.5.5
INFO Reading configuration /config/wtw-whr930.yaml...
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing wtw-whr930 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/wtw-whr930/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/button/button.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/climate/climate.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/climate/climate_mode.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/climate/climate_traits.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/esp8266/core.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/esp8266/gpio.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/esp8266/preferences.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/restart/button/restart_button.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/wtw-whr930/src/esphome/components/socket/socket.cpp.o
[... 1114 lines cut ...]
  sensor::Sensor *fan_speed_exhaust{nullptr};
  binary_sensor::BinarySensor *is_bypass_valve_open{nullptr};
  binary_sensor::BinarySensor *is_preheating{nullptr};
  sensor::Sensor *outside_air_temperature{nullptr};
  sensor::Sensor *supply_air_temperature{nullptr};
  sensor::Sensor *return_air_temperature{nullptr};
  sensor::Sensor *exhaust_air_temperature{nullptr};
  sensor::Sensor *enthalpy_temperature{nullptr};
  sensor::Sensor *ewt_temperature{nullptr};
  sensor::Sensor *reheating_temperature{nullptr};
  sensor::Sensor *kitchen_hood_temperature{nullptr};
  sensor::Sensor *return_air_level{nullptr};
  sensor::Sensor *supply_air_level{nullptr};
  binary_sensor::BinarySensor *is_supply_fan_active{nullptr};
  binary_sensor::BinarySensor *is_filter_full{nullptr};
  sensor::Sensor *bypass_factor{nullptr};
  sensor::Sensor *bypass_step{nullptr};
  sensor::Sensor *bypass_correction{nullptr};
  binary_sensor::BinarySensor *is_summer_mode{nullptr};
};

}  // namespace comfoair
}  // namespace esphome

```

paveldn, 2023-06-20:
@786534786 
You are trying to use constructors that don't exist anymore. Replace in comfoair.h line:
`  ComfoAirComponent(UARTComponent *parent) : Climate("comfoair"), PollingComponent(600), UARTDevice(parent) { }`
with:
`  ComfoAirComponent(UARTComponent *parent) : Climate(), PollingComponent(600), UARTDevice(parent) { }`
and all lines like:
`... = new Sensor("<sensor_name>")`
with
`... = new Sensor();`

786534786, 2023-06-20:
Thanks. Also needed to add in yaml file all lines with above and add to get an unique id in Home Assistant and a name:

```    
ca->return_air_level->set_object_id("Comfoair Return Air Level");
ca->return_air_level->set_name("Comfoair Return Air Level");
```

Replace "return_air_level"  with actual sensor settings

closed 2023-06-20 by 786534786