esphome/issues #4556: Issue with pin_reset function of Rotary Encoder

MauiR, 2023-06-02:
### The problem

Description:
The pin_reset function of the Rotary Encoder is not working as expected. When using the pin_reset function, it does not have any effect on the behavior of the encoder.

I am using the following Rotary Encoder:
[Rotary Encoder on AliExpress](https://de.aliexpress.com/item/1005002845325794.html?spm=a2g0o.cart.0.0.28914ae4Xqidg8&mp=1&gatewayAdapt=glo2deu)

However, I am able to read and use the key pin using binary_sensors without any issues. Therefore, I believe that the problem is not with the circuit or the Rotary Encoder itself.

```c++
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
    name: "Encoder Button"
    on_press:
      then:
        - light.toggle:
            id: Kugel_Licht
``` 

Steps to Reproduce:

Connect the Rotary Encoder according to the provided documentation.
Configure ESPHome with the pin_reset function for the Rotary Encoder.
Observe the lack of effect of the pin_reset function.
Expected Behavior:
The pin_reset function should reset the Rotary Encoder and have an impact on its behavior.

Actual Behavior:
The pin_reset function does not reset the Rotary Encoder or have any effect on its behavior.

Please let me know if there are any further details or troubleshooting steps that I can provide to assist in resolving this issue.

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.4

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

Rotary Encoder Sensor

ssieb, 2023-06-02:
You didn't include your config or other useful information.  What are you expecting it to do?  What do you see in the logs?

MauiR, 2023-06-05:
I expect the pin_reset to reset the steps that the rotary encoder counts. This is not the case. Also nothing is shown in the logs when the pin_reset is executed. 

The config looks like this. 

```c++
light:
  - platform: monochromatic
    id: Kugel_Licht
    name: "Lampe"
    output: Kugel_Licht_PWM
    default_transition_length: 0.3s
 

sensor:
  - platform: rotary_encoder
    id: Encoder
    name: "Rotary Encoder"
    pin_a: GPIO5
    pin_b: GPIO4
    pin_reset: GPIO14
    resolution: 4
    min_value: 1
    max_value: 50
    on_value:
      - light.turn_on:
          id: Kugel_Licht
          brightness: !lambda 'return (id(Encoder).state)*2 / 100;'

/**
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
    name: "Encoder Button"
    on_press:
      then:
        - light.toggle:
            id: Kugel_Licht
**/

output:
  - platform: esp8266_pwm
    id: Kugel_Licht_PWM
    pin: GPIO13
```

ssieb, 2023-06-05:
Do you see the difference between the pin config for the encoder and the config for the binary_sensor that you tested?

MauiR, 2023-06-09:
Yes the binary sensor configuration toggles the light as expected and also outputs the button press in the command line. The pin config does nothing.

closed 2023-10-22 by github-actions[bot]