esphome/issues #4557: friendly_name is not used for MQTT discovery

AndreKR, 2023-06-03:
### The problem

**tl;dr:** `name` is used instead of `friendly_name` for MQTT discovery, which is an issue because `name` is restricted to lowercase letters.

-----

In the ESPhome docs [it says](https://esphome.io/components/esphome.html#configuration-variables):

> * **friendly_name** (*Optional*, string): This is the name sent to the frontend. It is used by Home Assistant as the integration name, device name, and is automatically prefixed to entities where necessary.

But this isn't what I'm seeing. In my device config I have:

```yaml
esphome:
  name: gosund1
  friendly_name: Gosund 1
```

And in Home Assisstant it looks like this:

[image]

[image]

So it looks like `name` is used instead of `friendly_name`?

I'm using MQTT, not the native integration, maybe this is a bug in the MQTT Client Component?

The discovery message looks like this:
```json
{
  "name": "gosund1 - Switch 1",
  "ic": "mdi:power-socket-eu",
  "stat_t": "gosund1/switch/gosund1_-_switch_1/state",
  "cmd_t": "gosund1/switch/gosund1_-_switch_1/command",
  "avty_t": "gosund1/status",
  "uniq_id": "ESPswitchgosund1_-_switch_1",
  "dev": {
    "ids": "24a16016d1be",
    "name": "gosund1",
    "sw": "esphome v2023.5.5 Jun  3 2023, 02:19:40",
    "mdl": "esp8285",
    "mf": "espressif"
  }
}
```

As you can see the `friendly_name` isn't in there at all.

I tried putting `"Gosund 1"` into `name` but that isn't a valid name:

> `'G' is an invalid character for names. Valid characters are: abcdefghijklmnopqrstuvwxyz0123456789-_ (lowercase, no spaces).`

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.5.4

### What platform are you using?

ESP8266

### Board

Gosund SP211 / `esp8285`

### Component causing the issue

mqtt

dotdoom, 2023-08-04:
+1 and this becomes especially problematic now that HA is forcing entity names to not contain a device name.  In such case, setting name to None doesn’t achieve the desired effect.

[Hh](https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099)

AndreKR, 2023-08-04:
> and this becomes especially problematic now that HA is forcing entity names to not contain a device name

Yeah, what's up with that? Just saw it this morning in the "Repairs". So "Living room sensor Temperature" is no longer a valid entity name when the device is called "Living room sensor" but "Sensor Living room Temperature" is? Is there any documentation about this change?

dotdoom, 2023-08-05:
@AndreKR it's the latter.  I think the entity list (`Settings > Devices & Services > Entities`) is populated with `friendly_name` which will already have entity `name` prefixed with device name.

The problem currently is that esphome `friendly_name` is not populated to HA in any way that I know, so HA will have to use what it has, resulting in unavoidably "unfriendly" names for entities.  Example:

```yaml
# esphome config
substitutions:
  room: "Bathroom"

esphome:
  # https://esphome.io/components/esphome.html#configuration-variables
  # May only contain lowercase, digits and hyphens, up to 24chars.
  name: heater
  # If you have more than one such device per network & HA deployment.
  name_add_mac_suffix: true
  friendly_name: "${room} heater"

climate:
  # https://esphome.io/components/sensor/index.html#base-sensor-configuration
  # We can use "None" to indicate that it's the main entity for the device.
  # This is the only place where "friendly_name" is used, AFAICT. It's used by
  # esphome itself, and will be replaced by friendly_name from section above.
  # Setting to "null" is not allowed by esphome configuration check.
  - name: None
    platform: xxx

switch:
  - name: "status led"
    platform: yyy
```

This will result in following in HA:

entity_id | friendly_name
--- | ---
`climate.heater5d8c0e_bathroom_heater` | heater5d8c0e Bathroom heater
`switch.heater5d8c0e_status_led` | heater5d8c0e status led

While desired is

entity_id | friendly_name
--- | ---
`climate.bathroom_heater` | Bathroom heater
`switch.bathroom_heater_status_led` | Bathroom heater status led

Trying to work around this by adding device name into entity name as before:

```yaml
switch:
  - name: "${room} heater status led"
```

Shall stop working since it will have an even worse effect:

entity_id | friendly_name
--- | ---
`climate.heater5d8c0e_bathroom_heater` | heater5d8c0e Bathroom heater status led

See also https://github.com/home-assistant/core/issues/97483

cc @Petro31 or @jbouwh if they would want to take a look and correct me or provide more context.  It would be most welcome.  Otherwise sorry for the ping.

AndreKR, 2023-09-11:
I believe this was fixed by https://github.com/esphome/esphome/pull/5205

closed 2023-09-11 by AndreKR