esphome/issues #4559: The neopixelbus and fastled_clockless is incompatible with RP2040

MarcoCITA, 2023-06-04:
### The problem

Hi, how can i control WS2812B led strip with Rasperry Pi W?.
When i try to write the code i get an error: The neopixelbus is incompatible with RP2040.
[image]

### Which version of ESPHome has the issue?

2023.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.4

### What platform are you using?

RP2040

### Board

Pico W

### Component causing the issue

fastled_clockless and neopixelbus

ssieb, 2023-06-04:
There is a PR for an addressable led component for the pico, but it's currently broken because of platformio.

ssieb, 2023-10-21:
The pico issue was fixed so use the https://esphome.io/components/light/rp2040_pio_led_strip component.

closed 2023-10-21 by ssieb